coolsnowwolf/lede #7043: 把package过了一遍，发现几处编译失败

EnnawYang, 2021-06-17:
target: x86_64
编译系统: Ubuntu 20.04
日志有点长，所以折叠处理了  
<details>
<summary>package/feeds/packages/jose failed to build.</summary>
<pre><code>
ninja: Entering directory `/home/xw/openwrt/lede/build_dir/target-x86_64_musl/jose-11/openwrt-build'
[1/17] Generating jose-b64-dec.1 with a custom command
FAILED: jose-b64-dec.1 
/usr/bin/a2x -f manpage -D /home/xw/openwrt/lede/build_dir/target-x86_64_musl/jose-11/openwrt-build /home/xw/openwrt/lede/build_dir/target-x86_64_musl/jose-11/doc/man/jose-b64-dec.1.adoc
a2x: ERROR: "xmllint" --nonet --noout --valid "/home/xw/openwrt/lede/build_dir/target-x86_64_musl/jose-11/openwrt-build/jose-b64-dec.1.xml" returned non-zero exit status 4

[2/17] Generating jose-b64-enc.1 with a custom command
FAILED: jose-b64-enc.1 
/usr/bin/a2x -f manpage -D /home/xw/openwrt/lede/build_dir/target-x86_64_musl/jose-11/openwrt-build /home/xw/openwrt/lede/build_dir/target-x86_64_musl/jose-11/doc/man/jose-b64-enc.1.adoc
a2x: ERROR: "xmllint" --nonet --noout --valid "/home/xw/openwrt/lede/build_dir/target-x86_64_musl/jose-11/openwrt-build/jose-b64-enc.1.xml" returned non-zero exit status 4

[3/17] Generating jose.1 with a custom command
FAILED: jose.1 
/usr/bin/a2x -f manpage -D /home/xw/openwrt/lede/build_dir/target-x86_64_musl/jose-11/openwrt-build /home/xw/openwrt/lede/build_dir/target-x86_64_musl/jose-11/doc/man/jose.1.adoc
a2x: ERROR: "xmllint" --nonet --noout --valid "/home/xw/openwrt/lede/build_dir/target-x86_64_musl/jose-11/openwrt-build/jose.1.xml" returned non-zero exit status 4

[4/17] Generating jose-jwe-enc.1 with a custom command
FAILED: jose-jwe-enc.1 
/usr/bin/a2x -f manpage -D /home/xw/openwrt/lede/build_dir/target-x86_64_musl/jose-11/openwrt-build /home/xw/openwrt/lede/build_dir/target-x86_64_musl/jose-11/doc/man/jose-jwe-enc.1.adoc
a2x: ERROR: "xmllint" --nonet --noout --valid "/home/xw/openwrt/lede/build_dir/target-x86_64_musl/jose-11/openwrt-build/jose-jwe-enc.1.xml" returned non-zero exit status 4

[5/17] Generating jose-alg.1 with a custom command
FAILED: jose-alg.1 
/usr/bin/a2x -f manpage -D /home/xw/openwrt/lede/build_dir/target-x86_64_musl/jose-11/openwrt-build /home/xw/openwrt/lede/build_dir/target-x86_64_musl/jose-11/doc/man/jose-alg.1.adoc
a2x: ERROR: "xmllint" --nonet --noout --valid "/home/xw/openwrt/lede/build_dir/target-x86_64_musl/jose-11/openwrt-build/jose-alg.1.xml" returned non-zero exit status 4

[6/17] Generating jose-jwk-eql.1 with a custom command
FAILED: jose-jwk-eql.1 
/usr/bin/a2x -f manpage -D /home/xw/openwrt/lede/build_dir/target-x86_64_musl/jose-11/openwrt-build /home/xw/openwrt/lede/build_dir/target-x86_64_musl/jose-11/doc/man/jose-jwk-eql.1.adoc
a2x: ERROR: "xmllint" --nonet --noout --valid "/home/xw/openwrt/lede/build_dir/target-x86_64_musl/jose-11/openwrt-build/jose-jwk-eql.1.xml" returned non-zero exit status 4

[7/17] Generating jose-jwe-dec.1 with a custom command
FAILED: jose-jwe-dec.1 
/usr/bin/a2x -f manpage -D /home/xw/openwrt/lede/build_dir/target-x86_64_musl/jose-11/openwrt-build /home/xw/openwrt/lede/build_dir/target-x86_64_musl/jose-11/doc/man/jose-jwe-dec.1.adoc
a2x: ERROR: "xmllint" --nonet --noout --valid "/home/xw/openwrt/lede/build_dir/target-x86_64_musl/jose-11/openwrt-build/jose-jwe-dec.1.xml" returned non-zero exit status 4

[8/17] Generating jose-fmt.1 with a custom command
FAILED: jose-fmt.1 
/usr/bin/a2x -f manpage -D /home/xw/openwrt/lede/build_dir/target-x86_64_musl/jose-11/openwrt-build /home/xw/openwrt/lede/build_dir/target-x86_64_musl/jose-11/doc/man/jose-fmt.1.adoc
a2x: ERROR: "xmllint" --nonet --noout --valid "/home/xw/openwrt/lede/build_dir/target-x86_64_musl/jose-11/openwrt-build/jose-fmt.1.xml" returned non-zero exit status 4

[9/17] Generating jose-jwe-fmt.1 with a custom command
FAILED: jose-jwe-fmt.1 
/usr/bin/a2x -f manpage -D /home/xw/openwrt/lede/build_dir/target-x86_64_musl/jose-11/openwrt-build /home/xw/openwrt/lede/build_dir/target-x86_64_musl/jose-11/doc/man/jose-jwe-fmt.1.adoc
a2x: ERROR: "xmllint" --nonet --noout --valid "/home/xw/openwrt/lede/build_dir/target-x86_64_musl/jose-11/openwrt-build/jose-jwe-fmt.1.xml" returned non-zero exit status 4

ninja: build stopped: subcommand failed.
make[2]: *** [Makefile:76: /home/xw/openwrt/lede/build_dir/target-x86_64_musl/jose-11/.built] Error 1
time: package/feeds/packages/jose/compile#2.39#0.37#1.00
</code></pre>
</details>

<details>
<summary>package/feeds/packages/tang failed to build.</summary>
<pre><code>
The Meson build system
Version: 0.57.2
Source dir: /home/xw/openwrt/lede/build_dir/target-x86_64_musl/tang-10
Build dir: /home/xw/openwrt/lede/build_dir/target-x86_64_musl/tang-10/openwrt-build
Build type: cross build
WARNING: Unknown options: "build.libdir, build.prefix, build.sbindir"
The value of new options can be set with:
meson setup <builddir> --reconfigure -Dnew_option=new_value ...
Project name: tang
[... 86 lines cut ...]
make[2]: *** [Makefile:54: /home/xw/openwrt/lede/build_dir/target-x86_64_musl/qttools-everywhere-src-5.15.1/.built] Error 2
time: package/lean/qttools/compile#1.91#0.25#2.05
</code></pre>
</details>

<details>
<summary>package/lean/qBittorrent failed to build.</summary>
<pre><code>
make[3]: Entering directory '/home/xw/openwrt/lede/build_dir/target-x86_64_musl/qBittorrent-release-4.3.5'
/home/xw/openwrt/lede/staging_dir/hostpkg/bin/qmake -o Makefile qbittorrent.pro QMAKE_LRELEASE=
cd src/ && ( test -e Makefile || /home/xw/openwrt/lede/staging_dir/hostpkg/bin/qmake -o Makefile /home/xw/openwrt/lede/build_dir/target-x86_64_musl/qBittorrent-release-4.3.5/src/src.pro QMAKE_LRELEASE= ) && make -f Makefile 
make[4]: Entering directory '/home/xw/openwrt/lede/build_dir/target-x86_64_musl/qBittorrent-release-4.3.5/src'
/home/xw/openwrt/lede/staging_dir/hostpkg/bin/qmake -o Makefile src.pro QMAKE_LRELEASE=
Project MESSAGE: Project is built in RELEASE mode.
Project MESSAGE: Disabling debug output.
lrelease webui/www/translations/webui_ar.ts
/bin/sh: 1: lrelease-qt5: not found
make[4]: *** [Makefile:777: webui/www/translations/webui_ar.qm] Error 127
make[4]: Leaving directory '/home/xw/openwrt/lede/build_dir/target-x86_64_musl/qBittorrent-release-4.3.5/src'
make[3]: *** [Makefile:47: sub-src-make_first] Error 2
make[3]: Leaving directory '/home/xw/openwrt/lede/build_dir/target-x86_64_musl/qBittorrent-release-4.3.5'
make[2]: *** [Makefile:78: /home/xw/openwrt/lede/build_dir/target-x86_64_musl/qBittorrent-release-4.3.5/.built] Error 2
time: package/lean/qBittorrent/compile#0.80#0.18#1.30
</code></pre>
</details>

coolsnowwolf, 2021-06-17:
qttols 和 qbt 是没问题的

EnnawYang, 2021-06-17:
> qttols 和 qbt 是没问题的

那可能是我没清理直接pull了吧，毕竟跑一遍时间有点长（

coolsnowwolf, 2021-06-17:
测试所有都通过，没问题

coolsnowwolf, 2021-06-17:
[image]

coolsnowwolf, 2021-06-17:
[image]

EnnawYang, 2021-06-17:
🌚尴尬

closed 2021-06-17 by EnnawYang