coolsnowwolf/lede #7049: 请整个 lede 目录删除干净后重新拉去代码编译可避免老版本代码缓存和索引带来的问题

coolsnowwolf, 2021-06-17:
请整个 lede 目录删除干净后重新拉去代码编译可避免老版本代码缓存和索引带来的问题

1. 本次同步了最新的最新 OpenWrt trunk 代码，同时做了大量的 backport 兼容以前的软件，包括兼容 WIF6 AX，和 LUA 版本 luci
2. 至于为何不直接用 js 的 luci 版本，你可以自由选择，自行体会
3. 除了 x86/x64 ， 其他设备都不推荐保留配置升级可避免配置混乱带来的问题
4. 至于修复了官方的多少bug，这里就不列出了，只能说官方代码根本不考虑兼容性，就只是毛胚房的性质

为了面向未来，支持更多的特性例如无线 MESH/ WIFI6 等，做的必要更新， 例如ARM 内核新特性，10Gbps 以上网络支持。。。。

目前为止，已经修复了所有用户报告的编译和界面上的Bug （48小时内）

请理解。

不理解也请别谩骂，谢谢

[image]

starxu1971, 2021-06-17:
Thu Jun 17 20:47:58 2021 daemon.err uhttpd[3515]: sh: /etc/init.d/dropbear: line 13: extra_command: not found
Thu Jun 17 20:47:59 2021 daemon.err uhttpd[3515]: sh: /etc/init.d/haproxy: line 21: extra_command: not found
Thu Jun 17 20:47:59 2021 daemon.err uhttpd[3515]: sh: /etc/init.d/nginx: line 69: extra_command: not found
这三个不知道是原因造成的

adamgonglj, 2021-06-17:
感谢Lean大。

adamgonglj, 2021-06-17:
我听过一个说法，在程序员的世界里面，是0.1%的开发者在写代码，剩下的99.9%开发者在围观，lean大就是那0.1%

chenxuan5858, 2021-06-17:
感谢Lean大

minlang112, 2021-06-17:
感谢各位大佬的贡献

ZeaKyX, 2021-06-17:
mesh可以看看开源的easymesh项目[prplMesh](https://github.com/prplfoundation/prplMesh) 
新bug这么多，看来l大说的homekit插件又要鸽了（手动滑稽

adamgonglj, 2021-06-17:
> mesh可以看看开源的easymesh项目[prplMesh](https://github.com/prplfoundation/prplMesh)
> 新bug这么多，看来l大说的homekit插件又要鸽了（手动滑稽

看到你是Contributor，好羡慕。我不会开发，最近才会写点shell代码自动化编译。

ZeaKyX, 2021-06-17:
> > mesh可以看看开源的easymesh项目[prplMesh](https://github.com/prplfoundation/prplMesh)
> > 新bug这么多，看来l大说的homekit插件又要鸽了（手动滑稽
> 
> 看到你是Contributor，好羡慕。我不会开发，最近才会写点shell代码自动化编译。

别别别，我的pr我自己都不好意思说了，我就是个依葫芦画瓢写点云编译代码的菜😂还要多学习大佬

rshi0212, 2021-06-17:
make distclean也不行吗？还是要全部删掉吗？

xiangfeidexiaohuo, 2021-06-17:
感谢，x86 5.10内核 已经ok，正常使用，非常nice

ihuaer, 2021-06-18:
感谢大佬更新说明，忍痛把整个项目删除重新下载

adamgonglj, 2021-06-18:
R4S和R2S都已经编译成功，感谢L大。

adamgonglj, 2021-06-18:
@coolsnowwolf L大，最新代码编译的R2S估计的squashfs和ext4格式的固件都无法启动，刷入老固件后，系统可以正常启动。

adamgonglj, 2021-06-18:
@coolsnowwolf L大，最新代码编译的固件和ext4和squashfs格式的都无法启动，请及时解决呗。用的是balenaEtcher刷的系统。
京东和Netac联名的内存卡。刷入老固件，系统正常龚。

shu0802, 2021-06-19:
X86固件编译入dockerman的根目录上下不一样。不知是升级docker到最新版的原因，还是固件升级5.10后与dockerman的原因。（https://github.com/lisaac/luci-app-dockerman）
[image]

coolsnowwolf, 2021-06-19:
fixed docker

xtkong, 2021-06-19:
你怎么这么优秀呢

closed 2021-06-20 by coolsnowwolf