coolsnowwolf/lede #7050: 编译phicomm_k3出错

281677160, 2021-06-17:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：
/workdir/openwrt/staging_dir/host/bin/otrx create /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/tmp/openwrt-bcm53xx-generic-smartrg_sr400ac-squashfs.trx.new -f /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/smartrg_sr400ac-kernel.bin -a 1024 -f /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/tmp/openwrt-bcm53xx-generic-smartrg_sr400ac-squashfs.trx -a 0x10000 -A /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/fs_mark
mv /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/tmp/openwrt-bcm53xx-generic-smartrg_sr400ac-squashfs.trx.new /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/tmp/openwrt-bcm53xx-generic-smartrg_sr400ac-squashfs.trx
cp /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/tmp/openwrt-bcm53xx-generic-smartrg_sr400ac-squashfs.trx /workdir/openwrt/bin/targets/bcm53xx/generic/openwrt-bcm53xx-generic-smartrg_sr400ac-squashfs.trx
rm -f /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/phicomm_k3-kernel.bin
cp /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/zImage /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/phicomm_k3-kernel.bin
cat /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/image-bcm47094-phicomm-k3.dtb >> /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/phicomm_k3-kernel.bin
/workdir/openwrt/staging_dir/host/bin/lzma e /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/phicomm_k3-kernel.bin -d16  /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/phicomm_k3-kernel.bin.new
mv /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/phicomm_k3-kernel.bin.new /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/phicomm_k3-kernel.bin
[ -f /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/phicomm_k3-kernel.bin -a -f /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/root.squashfs ]
sh /workdir/openwrt/scripts/ubinize-image.sh    /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/root.squashfs /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/tmp/openwrt-bcm53xx-generic-phicomm_k3-squashfs.trx.tmp -p 128KiB -m 2048   
[rootfs]
mode=ubi
vol_id=0
vol_type=dynamic
vol_name=rootfs
image=/workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/root.squashfs
vol_size=13814784
[rootfs_data]
mode=ubi
vol_id=1
vol_type=dynamic
vol_name=rootfs_data
vol_size=1MiB
vol_flags=autoresize
cat /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/tmp/openwrt-bcm53xx-generic-phicomm_k3-squashfs.trx.tmp >> /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/tmp/openwrt-bcm53xx-generic-phicomm_k3-squashfs.trx
rm /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/tmp/openwrt-bcm53xx-generic-phicomm_k3-squashfs.trx.tmp
# kernel: always use 4 MiB (-28 B or TRX header) to allow upgrades even
#	  if it grows up between releases
# root: UBI with one extra block containing UBI mark to trigger erasing
#	rest of partition
/workdir/openwrt/staging_dir/host/bin/otrx create /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/tmp/openwrt-bcm53xx-generic-phicomm_k3-squashfs.trx.new -f /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/phicomm_k3-kernel.bin -a 0x20000 -b 0x400000 -f /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/tmp/openwrt-bcm53xx-generic-phicomm_k3-squashfs.trx -A /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/ubi_mark -a 0x20000
mv /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/tmp/openwrt-bcm53xx-generic-phicomm_k3-squashfs.trx.new /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/tmp/openwrt-bcm53xx-generic-phicomm_k3-squashfs.trx
cp /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/tmp/openwrt-bcm53xx-generic-phicomm_k3-squashfs.trx /workdir/openwrt/bin/targets/bcm53xx/generic/openwrt-bcm53xx-generic-phicomm_k3-squashfs.trx
rm -f /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/tenda_ac9-kernel.bin
cp /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/zImage /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/tenda_ac9-kernel.bin
cat /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/image-bcm47189-tenda-ac9.dtb >> /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/tenda_ac9-kernel.bin
/workdir/openwrt/staging_dir/host/bin/lzma e /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/tenda_ac9-kernel.bin -d16  /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/tenda_ac9-kernel.bin.new
mv /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/tenda_ac9-kernel.bin.new /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/tenda_ac9-kernel.bin
[ -f /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/tenda_ac9-kernel.bin -a -f /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/root.squashfs ]
dd if=/workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/root.squashfs >> /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/tmp/openwrt-bcm53xx-generic-tenda_ac9-squashfs.trx
26981+1 records in
26981+1 records out
13814732 bytes (14 MB, 13 MiB) copied, 0.0716232 s, 193 MB/s
/workdir/openwrt/staging_dir/host/bin/otrx create /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/tmp/openwrt-bcm53xx-generic-tenda_ac9-squashfs.trx.new -f /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/tenda_ac9-kernel.bin -a 1024 -f /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/tmp/openwrt-bcm53xx-generic-tenda_ac9-squashfs.trx -a 0x10000 -A /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/fs_mark
mv /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/tmp/openwrt-bcm53xx-generic-tenda_ac9-squashfs.trx.new /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/tmp/openwrt-bcm53xx-generic-tenda_ac9-squashfs.trx
cp /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/tmp/openwrt-bcm53xx-generic-tenda_ac9-squashfs.trx /workdir/openwrt/bin/targets/bcm53xx/generic/openwrt-bcm53xx-generic-tenda_ac9-squashfs.trx
rm -f /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/tplink_archer-c5-v2-kernel.bin
cp /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/zImage /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/tplink_archer-c5-v2-kernel.bin
cat /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/image-bcm47081-tplink-archer-c5-v2.dtb >> /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/tplink_archer-c5-v2-kernel.bin
/workdir/openwrt/staging_dir/host/bin/lzma e /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/tplink_archer-c5-v2-kernel.bin -d16  /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/tplink_archer-c5-v2-kernel.bin.new
[... 1 lines cut ...]
[ -f /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/tplink_archer-c5-v2-kernel.bin -a -f /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/root.squashfs ]
dd if=/workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/root.squashfs >> /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/tmp/openwrt-bcm53xx-generic-tp [... cut ...]
make[3]: Leaving directory '/workdir/openwrt/target/linux'
time: target/linux/install#37.00#6.83#38.34
make[2]: *** [target/Makefile:25: target/linux/install] Error 1
make[1]: *** [target/Makefile:19: /workdir/openwrt/staging_dir/target-arm_cortex-a9_musl_eabi/stamp/.target_install] Error 2
    ERROR: target/linux failed to build.
make[2]: Leaving directory '/workdir/openwrt'
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2

云编译phicomm_k3出错，好像是说固件过大？不应该啊，基本就默认插件

m4231007, 2021-06-19:
一样

281677160, 2021-06-19:
等大雕修复呗，哈哈

closed 2021-06-19 by coolsnowwolf