coolsnowwolf/lede #7051: A lot of warning when feeds install

galaxyskyknight, 2021-06-17:
luke@BuildServer:~/lede_x86_new$ ./scripts/feeds install -a
Collecting package info: done
Collecting target info: done
WARNING: Makefile 'package/lean/amule/Makefile' has a dependency on 'libpng', which does not exist
WARNING: Makefile 'package/lean/amule/Makefile' has a dependency on 'libupnp', which does not exist
WARNING: Makefile 'package/lean/amule/Makefile' has a build dependency on 'libgd', which does not exist
WARNING: Makefile 'package/lean/antileech/Makefile' has a build dependency on 'libgd', which does not exist
WARNING: Makefile 'package/lean/autocore/Makefile' has a dependency on 'lm-sensors', which does not exist
WARNING: Makefile 'package/lean/autocore/Makefile' has a dependency on 'bc', which does not exist
WARNING: Makefile 'package/lean/autocore/Makefile' has a dependency on 'lm-sensors', which does not exist
WARNING: Makefile 'package/lean/autosamba/Makefile' has a dependency on 'luci-app-samba', which does not exist
WARNING: Makefile 'package/lean/autosamba/Makefile' has a dependency on 'wsdd2', which does not exist
WARNING: Makefile 'package/lean/baidupcs-web/Makefile' has a build dependency on 'golang/host', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a build dependency on 'libpam', which does not exist
WARNING: Makefile 'package/lean/ddns-scripts_aliyun/Makefile' has a dependency on 'ddns-scripts', which does not exist
WARNING: Makefile 'package/lean/ddns-scripts_dnspod/Makefile' has a dependency on 'ddns-scripts', which does not exist
WARNING: Makefile 'package/lean/default-settings/Makefile' has a dependency on 'luci-base', which does not exist
WARNING: Makefile 'package/lean/default-settings/Makefile' has a dependency on 'luci', which does not exist
WARNING: Makefile 'package/lean/default-settings/Makefile' has a dependency on 'luci-compat', which does not exist
WARNING: Makefile 'package/lean/dnsforwarder/Makefile' has a dependency on 'coreutils-base64', which does not exist
WARNING: Makefile 'package/network/utils/dpdk/Makefile' has a build dependency on 'meson/host', which does not exist
WARNING: Makefile 'package/network/utils/dpdk/Makefile' has a build dependency on 'ninja/host', which does not exist
WARNING: Makefile 'package/network/utils/dpdk/Makefile' has a build dependency on 'ninja/host', which does not exist
WARNING: Makefile 'package/network/utils/dpdk/Makefile' has a build dependency on 'ninja/host', which does not exist
WARNING: Makefile 'package/network/utils/dpdk/Makefile' has a build dependency on 'ninja/host', which does not exist
WARNING: Makefile 'package/network/utils/dpdk/Makefile' has a build dependency on 'ninja/host', which does not exist
WARNING: Makefile 'package/lean/frp/Makefile' has a build dependency on 'golang/host', which does not exist
WARNING: Makefile 'package/lean/gmediarender/Makefile' has a dependency on 'gstreamer1', which does not exist
WARNING: Makefile 'package/lean/gmediarender/Makefile' has a dependency on 'libgstreamer1', which does not exist
WARNING: Makefile 'package/lean/gmediarender/Makefile' has a dependency on 'gstreamer1-libs', which does not exist
WARNING: Makefile 'package/lean/gmediarender/Makefile' has a dependency on 'glib2', which does not exist
WARNING: Makefile 'package/lean/gmediarender/Makefile' has a dependency on 'libupnp', which does not exist
WARNING: Makefile 'package/lean/ipv6-helper/Makefile' has a dependency on 'luci-proto-ipv6', which does not exist
WARNING: Makefile 'package/boot/kexec-tools/Makefile' has a dependency on 'liblzma', which does not exist
WARNING: Makefile 'package/network/services/lldpd/Makefile' has a dependency on 'libnetsnmp', which does not exist
WARNING: Makefile 'package/lean/luci-app-accesscontrol/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-adbyby-plus/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-airplay2/Makefile' has a dependency on 'alsa-utils', which does not exist
WARNING: Makefile 'package/lean/luci-app-airplay2/Makefile' has a dependency on 'shairport-sync-openssl', which does not exist
WARNING: Makefile 'package/lean/luci-app-airplay2/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-amule/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-arpbind/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-autoreboot/Makefile' has a dependency on 'luci', which does not exist
WARNING: Makefile 'package/lean/luci-app-autoreboot/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-baidupcs-web/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-cifs-mount/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-cifsd/Makefile' has a dependency on 'ksmbd-server', which does not exist
WARNING: Makefile 'package/lean/luci-app-cifsd/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-cpufreq/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-diskman/Makefile' has a dependency on 'parted', which does not exist
WARNING: Makefile 'package/lean/luci-app-diskman/Makefile' has a dependency on 'smartmontools', which does not exist
WARNING: Makefile 'package/lean/luci-app-diskman/Makefile' has a dependency on 'btrfs-progs', which does not exist
WARNING: Makefile 'package/lean/luci-app-dnsforwarder/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-docker/Makefile' has a dependency on 'dockerd', which does not exist
WARNING: Makefile 'package/lean/luci-app-docker/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-familycloud/Makefile' has a dependency on 'coreutils', which does not exist
WARNING: Makefile 'package/lean/luci-app-familycloud/Makefile' has a dependency on 'coreutils-nohup', which does not exist
WARNING: Makefile 'package/lean/luci-app-familycloud/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/lean/luci-app-familycloud/Makefile' has a dependency on 'bash', which does not exist
WARNING: Makefile 'package/lean/luci-app-familycloud/Makefile' has a dependency on 'curl', which does not exist
WARNING: Makefile 'package/lean/luci-app-familycloud/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-filetransfer/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-flowoffload/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-frpc/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-frps/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-guest-wifi/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-haproxy-tcp/Makefile' has a dependency on 'haproxy', which does not exist
WARNING: Makefile 'package/lean/luci-app-haproxy-tcp/Makefile' has a build dependency on 'luci-base/host', which does not exist
[... 90 lines cut ...]
WARNING: Makefile 'package/lean/mt/drivers/mt7615d/Makefile' has a dependency on 'maccalc', which does not exist
WARNING: Makefile 'package/lean/n2n_v2/Makefile' has a dependency on 'libzstd', which does not exist
WARNING: Makefile 'package/network/services/noddos/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/network/services/noddos/Makefile' has a dependency on 'libtins', which does not exist
WARNING: Makefile 'package/network/services/noddos/Makefile' has a dependency on 'libyaml-cpp', which does not exist
[... cut ...]
WARNING: Makefile 'package/lean/trojan/Makefile' has a dependency on 'boost-program_options', which does not exist
WARNING: Makefile 'package/lean/trojan/Makefile' has a dependency on 'boost-date_time', which does not exist
WARNING: Makefile 'package/lean/UnblockNeteaseMusic/Makefile' has a dependency on 'node', which does not exist
WARNING: Makefile 'package/lean/UnblockNeteaseMusicGo/Makefile' has a build dependency on 'golang/host', which does not exist
WARNING: Makefile 'package/lean/wxbase/Makefile' has a dependency on 'libexpat', which does not exist

galaxyskyknight, 2021-06-17:
Above happen when I creat a emtpy folder and pull the code, everthing start from original.

EnnawYang, 2021-06-17:
This is normal for the first installation, just ignore it

closed 2021-06-18 by coolsnowwolf

baalwy, 2022-04-03:
it's the same problem until now