coolsnowwolf/lede #7052: 竞斗云编译失败

qpzr, 2021-06-17:
dd: failed to open '/home/runner/work/LOW/LOW/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/image/uboot-fritz4040.bin': No such file or directory 

4109make[5]: *** [Makefile:809: /home/runner/work/LOW/LOW/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/tmp/openwrt-ipq40xx-avm_fritzbox-4040-squashfs-eva.bin] Error 1 

4110make[5]: Leaving directory '/home/runner/work/LOW/LOW/openwrt/target/linux/ipq40xx/image' 

4111make[4]: *** [Makefile:25: install] Error 2 

4112make[4]: Leaving directory '/home/runner/work/LOW/LOW/openwrt/target/linux/ipq40xx' 

4113make[3]: Leaving directory '/home/runner/work/LOW/LOW/openwrt/target/linux' 

4114make[3]: *** [Makefile:11: install] Error 2 

4115time: target/linux/install#147.36#15.68#157.17 

4116 ERROR: target/linux failed to build. 

4117make[2]: *** [target/Makefile:25: target/linux/install] Error 1 

4118make[2]: Leaving directory '/home/runner/work/LOW/LOW/openwrt' 

4119make[1]: *** [target/Makefile:19: /home/runner/work/LOW/LOW/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.target_install] Error 2 

4120make[1]: Leaving directory '/home/runner/work/LOW/LOW/openwrt' 

4121make: *** [/home/runner/work/LOW/LOW/openwrt/include/toplevel.mk:230: world] Error 2 

4122Error: Process completed with exit code 2.

Boos4721, 2021-06-17:
可以编译成功

qpzr, 2021-06-17:
> 可以编译成功

Thu, 17 Jun 2021 18:20:39 GMT
dd: failed to open '/home/runner/work/LOW/LOW/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/image/uboot-fritz4040.bin': No such file or directory
Thu, 17 Jun 2021 18:20:39 GMT
make[5]: *** [Makefile:809: /home/runner/work/LOW/LOW/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/tmp/openwrt-ipq40xx-avm_fritzbox-4040-squashfs-eva.bin] Error 1
Thu, 17 Jun 2021 18:20:39 GMT
make[5]: Leaving directory '/home/runner/work/LOW/LOW/openwrt/target/linux/ipq40xx/image'
Thu, 17 Jun 2021 18:20:39 GMT
make[4]: *** [Makefile:25: install] Error 2
Thu, 17 Jun 2021 18:20:39 GMT
make[4]: Leaving directory '/home/runner/work/LOW/LOW/openwrt/target/linux/ipq40xx'
Thu, 17 Jun 2021 18:20:39 GMT
make[3]: *** [Makefile:11: install] Error 2
Thu, 17 Jun 2021 18:20:39 GMT
make[3]: Leaving directory '/home/runner/work/LOW/LOW/openwrt/target/linux'
Thu, 17 Jun 2021 18:20:39 GMT
time: target/linux/install#128.73#13.51#136.92
Thu, 17 Jun 2021 18:20:39 GMT
    ERROR: target/linux failed to build.
Thu, 17 Jun 2021 18:20:39 GMT
make[2]: *** [target/Makefile:25: target/linux/install] Error 1
Thu, 17 Jun 2021 18:20:39 GMT
make[2]: Leaving directory '/home/runner/work/LOW/LOW/openwrt'
Thu, 17 Jun 2021 18:20:39 GMT
make[1]: *** [target/Makefile:19: /home/runner/work/LOW/LOW/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.target_install] Error 2
Thu, 17 Jun 2021 18:20:39 GMT
make[1]: Leaving directory '/home/runner/work/LOW/LOW/openwrt'
Thu, 17 Jun 2021 18:20:39 GMT
make: *** [/home/runner/work/LOW/LOW/openwrt/include/toplevel.mk:230: world] Error 2
Thu, 17 Jun 2021 18:20:39 GMT
Error: Process completed with exit code 2.

brightsm, 2021-06-18:
竞斗云2 正常。

qpzr, 2021-06-18:
> 可以编译成功

> 竞斗云2 正常。

你们fork祖传的源码肯定正常啊  我直接Actions 这里最新的源码编译就是失败了

Boos4721, 2021-06-18:
> > 可以编译成功
> 
> > 竞斗云2 正常。
> 
> 你们自己fork了一份祖传的源码肯定正常啊 ，我直接Actions 这里最新的源码编译就是失败了

[image]

AmadeusGhost, 2021-06-18:
> 你们自己fork了一份祖传的源码肯定正常啊 ，我直接Actions 这里最新的源码编译就是失败了

那是你的问题，不要开多设备编译（换句话说就是重新生成 .config

qpzr, 2021-06-19:
> > 你们自己fork了一份祖传的源码肯定正常啊 ，我直接Actions 这里最新的源码编译就是失败了
> 
> 那是你的问题，不要开多设备编译（换句话说就是重新生成 .config

多设备？就编译竞斗云一个 . config如下

CONFIG_TARGET_ipq40xx=y

CONFIG_TARGET_ipq40xx_generic_DEVICE_p2w_r619ac-128m=y

我确定一个星期前还是可以编译成功的
[image]

Boos4721, 2021-06-19:
> > > 你们自己fork了一份祖传的源码肯定正常啊 ，我直接Actions 这里最新的源码编译就是失败了
> > 
> > 
> > 那是你的问题，不要开多设备编译（换句话说就是重新生成 .config
> 
> 多设备？就编译竞斗云一个 . config如下
> 
> CONFIG_TARGET_ipq40xx=y
> 
> CONFIG_TARGET_ipq40xx_generic_DEVICE_p2w_r619ac-128m=y
> 
> 我确定一个星期前还是可以编译成功的
> [image]

https://github.com/coolsnowwolf/lede/blob/master/target/linux/ipq40xx/image/Makefile
你的问题

closed 2021-06-19 by coolsnowwolf

Cathgao, 2021-06-19:
设备改名了，把后面什么128m的删掉。最好就是menuconfig之后复制替换一下设备名

qpzr, 2021-06-19:
> 设备改名了，把后面什么128m的删掉。最好就是menuconfig之后复制替换一下设备名

  我用的是128M容量的固件，删了-128m编译成功撸进去目测会翻车，所以我没敢。根据上面@Boos4721的回答，看了他的Actions里成功的车，顺便也看了一下他的.config 既然源码也删掉 _generic 这部分。跟着搬砖照抄删掉，就编译成功了。感谢各位解答

Cathgao, 2021-06-20:
> > 设备改名了，把后面什么128m的删掉。最好就是menuconfig之后复制替换一下设备名
> 
>   我用的是128M容量的固件，删了-128m编译成功撸进去目测会翻车，所以我没敢。根据上面@Boos4721的回答，看了他的Actions里成功的车，顺便也看了一下他的.config 既然源码也删掉 _generic 这部分。跟着搬砖照抄删掉，就编译成功了。感谢各位解答

我都刷进去用两天了，翻不翻车试一下就知道了。源码就一个R619AC，不删你还能怎么编译呢

Boos4721, 2021-06-20:
> > > 设备改名了，把后面什么128m的删掉。最好就是menuconfig之后复制替换一下设备名
> > 
> > 
> > 我用的是128M容量的固件，删了-128m编译成功撸进去目测会翻车，所以我没敢。根据上面@Boos4721的回答，看了他的Actions里成功的车，顺便也看了一下他的.config 既然源码也删掉 _generic 这部分。跟着搬砖照抄删掉，就编译成功了。感谢各位解答
> 
> 我都刷进去用两天了，翻不翻车试一下就知道了。源码就一个R619AC，不删你还能怎么编译呢

[image]
？

PR #7080 (2021-06-19): ipq40xx: some typo and bug fixes
Closed: #7052