coolsnowwolf/lede #7056: 更新代码后内核出现bug，导致某些进程被随机关闭，这是内存溢出导致的吗？

sallyciel, 2021-06-17:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：更新代码后内核出现bug，导致某些进程被随机关闭，内存设置为512MB，用满，设置为1GB也用满，这是内存溢出导致的吗？

### (2) 路由器型号和固件版本

A：ESXi 6.7 U2 | 主机型号 | Intel(R) Core(TM) i5-7200U CPU @ 2.50GHz : 4 Core 4 Thread | CPU 信息 | 2712.000 MHz
固件版本 | OpenWrt R21.6.14 / LuCI Master (git-21.167.28955-fccaab9)

### (3) 详细日志
```
A：Thu Jun 17 23:28:05 2021 kern.info kernel: [   58.265905] [   9725]     0  9725      273        7    32768        0             0 mount
Thu Jun 17 23:28:05 2021 kern.info kernel: [   58.266185] [   9726]     0  9726      273        8    32768        0             0 sh
Thu Jun 17 23:28:05 2021 kern.info kernel: [   58.266492] [   9727]     0  9727      273        8    36864        0             0 mount
Thu Jun 17 23:28:05 2021 kern.info kernel: [   58.266769] [   9728]     0  9728      273        7    40960        0             0 sh
Thu Jun 17 23:28:05 2021 kern.info kernel: [   58.267042] [   9729]     0  9729      273        8    36864        0             0 mount
Thu Jun 17 23:28:05 2021 kern.info kernel: [   58.267320] [   9730]     0  9730      273        8    32768        0             0 sh
Thu Jun 17 23:28:05 2021 kern.info kernel: [   58.267596] [   9731]     0  9731      273        8    40960        0             0 mount
Thu Jun 17 23:28:05 2021 kern.info kernel: [   58.267874] [   9732]     0  9732      273        8    36864        0             0 sh
Thu Jun 17 23:28:05 2021 kern.info kernel: [   58.268149] [   9733]     0  9733      273        7    40960        0             0 mount
Thu Jun 17 23:28:05 2021 kern.info kernel: [   58.268428] [   9734]     0  9734      273        7    36864        0             0 sh
Thu Jun 17 23:28:05 2021 kern.info kernel: [   58.269505] [   9735]     0  9735      273        8    40960        0             0 mount
Thu Jun 17 23:28:05 2021 kern.info kernel: [   58.270090] [   9736]     0  9736      273        7    32768        0             0 sh
Thu Jun 17 23:28:05 2021 kern.info kernel: [   58.270367] [   9737]     0  9737      273        8    36864        0             0 mount
Thu Jun 17 23:28:05 2021 kern.info kernel: [   58.270649] [   9738]     0  9738      273        7    32768        0             0 sh
Thu Jun 17 23:28:05 2021 kern.info kernel: [   58.270926] [   9739]     0  9739      273        7    36864        0             0 mount
Thu Jun 17 23:28:05 2021 kern.info kernel: [   58.271206] [   9740]     0  9740      273        8    36864        0             0 sh
Thu Jun 17 23:28:05 2021 kern.info kernel: [   58.271531] [   9741]     0  9741      273        7    36864        0             0 mount
Thu Jun 17 23:28:05 2021 kern.info kernel: [   58.271814] [   9742]     0  9742      273        8    36864        0             0 sh
Thu Jun 17 23:28:05 2021 kern.info kernel: [   58.272092] [   9743]     0  9743      273        8    32768        0             0 mount
Thu Jun 17 23:28:05 2021 kern.info kernel: [   58.272389] [   9744]     0  9744      273        7    32768        0             0 sh
Thu Jun 17 23:28:05 2021 kern.info kernel: [   58.272666] [   9745]     0  9745      273        8    36864        0             0 mount
Thu Jun 17 23:28:05 2021 kern.info kernel: [   58.272946] [   9746]     0  9746      273        8    32768        0             0 sh
Thu Jun 17 23:28:05 2021 kern.info kernel: [   58.273228] [   9747]     0  9747      273        8    32768        0             0 mount
Thu Jun 17 23:28:05 2021 kern.info kernel: [   58.273506] [   9748]     0  9748      273        8    36864        0             0 sh
Thu Jun 17 23:28:05 2021 kern.info kernel: [   58.273781] [   9749]     0  9749      273        8    36864        0             0 mount
Thu Jun 17 23:28:05 2021 kern.info kernel: [   58.274059] [   9750]     0  9750      273        8    36864        0             0 sh
Thu Jun 17 23:28:05 2021 kern.info kernel: [   58.274339] [   9751]     0  9751      273        7    32768        0             0 mount
Thu Jun 17 23:28:05 2021 kern.info kernel: [   58.274618] [   9752]     0  9752      273        7    36864        0             0 sh
Thu Jun 17 23:28:05 2021 kern.info kernel: [   58.274892] [   9753]     0  9753      273        7    40960        0             0 mount
Thu Jun 17 23:28:05 2021 kern.info kernel: [   58.275170] [   9754]     0  9754      273        7    36864        0             0 sh
Thu Jun 17 23:28:05 2021 kern.info kernel: [   58.275451] [   9755]     0  9755      273        8    32768        0             0 mount
Thu Jun 17 23:28:05 2021 kern.info kernel: [   58.275753] [   9756]     0  9756      273        7    36864        0             0 sh
Thu Jun 17 23:28:05 2021 kern.info kernel: [   58.276045] [   9757]     0  9757      273        8    36864        0             0 mount
Thu Jun 17 23:28:05 2021 kern.info kernel: [   58.276371] [   9758]     0  9758      273        7    36864        0             0 sh
Thu Jun 17 23:28:05 2021 kern.info kernel: [   58.276659] [   9759]     0  9759      273        7    36864        0             0 mount
Thu Jun 17 23:28:05 2021 kern.info kernel: [   58.276953] [   9760]     0  9760      273        8    36864        0             0 sh
Thu Jun 17 23:28:05 2021 kern.info kernel: [   58.277242] [   9761]     0  9761      273        7    36864        0             0 mount
Thu Jun 17 23:28:05 2021 kern.info kernel: [   58.277524] [   9762]     0  9762      273        8    40960        0             0 sh
Thu Jun 17 23:28:05 2021 kern.info kernel: [   58.277801] [   9763]     0  9763      273        8    32768        0             0 mount
Thu Jun 17 23:28:05 2021 kern.info kernel: [   58.278081] [   9764]     0  9764      273        7    36864        0             0 sh
Thu Jun 17 23:28:05 2021 kern.info kernel: [   58.278359] [   9765]     0  9765      273        8    36864        0             0 mount
Thu Jun 17 23:28:05 2021 kern.info kernel: [   58.278639] [   9766]     0  9766      273        7    36864        0             0 sh
Thu Jun 17 23:28:05 2021 kern.info kernel: [   58.280073] [   9767]     0  9767      273        8    36864        0             0 mount
Thu Jun 17 23:28:05 2021 kern.info kernel: [   58.280349] [   9768]     0  9768      273        7    40960        0             0 sh
Thu Jun 17 23:28:05 2021 kern.info kernel: [   58.280621] [   9769]     0  9769      273        7    36864        0             0 mount
Thu Jun 17 23:28:05 2021 kern.info kernel: [   58.280892] [   9770]     0  9770      273        7    36864        0             0 sh
Thu Jun 17 23:28:05 2021 kern.info kernel: [   58.281160] [   9771]     0  9771      273        7    32768        0             0 mount
Thu Jun 17 23:28:05 2021 kern.info kernel: [   58.281432] [   9772]     0  9772      273        8    40960        0             0 sh
Thu Jun 17 23:28:05 2021 kern.info kernel: [   58.281700] [   9773]     0  9773      273        8    36864        0             0 mount
Thu Jun 17 23:28:05 2021 kern.info kernel: [   58.281974] [   9774]     0  9774      273        7    36864        0             0 sh
[... 446 lines cut ...]
Thu Jun 17 23:28:05 2021 kern.info kernel: [   58.423911] [  10239]     0 10239      273        7    32768        0             0 sh
Thu Jun 17 23:28:05 2021 kern.info kernel: [   58.424187] [  10240]     0 10240      273        7    36864        0             0 mount
Thu Jun 17 23:28:05 2021 kern.info kernel: [   58.424519] [  10241]     0 10241      273        8    32768        0             0 sh
Thu Jun 17 23:28:05 2021 kern.info kernel: [   58.424917] [  10242]     0 10242      273        8    32768        0             0 mount
Thu Jun 17 23:28:05 2021 kern.info kernel: [   58.425344] [  10243]     0 10243      273        8    36864        0             0 sh
Thu Jun 17 23:28:05 2021 kern.info kernel: [   58.425780] [  10244]     0 10244      273        7    40960        0             0 mount
Thu Jun 17 23:28:05 2021 kern.info kernel: [   58.426063] [  10245]     0 10245      273        7    32768        0             0 sh
Thu Jun 17 23:28:05 2021 kern.info kernel: [   58.426343] [  10246]     0 10246      273        8    36864        0             0 mount
Thu Jun 17 23:28:05 2021 kern.info kernel: [   58.426619] [  10248]     0 10248      273        8    32768        0             0 sh
Thu Jun 17 23:28:05 2021 kern.info kernel: [   58.426891] [  10250]     0 10250      273        7    40960        0             0 mount
[... cut ...]
Thu Jun 17 23:28:05 2021 kern.info kernel: [   58.429969] oom-kill:constraint=CONSTRAINT_NONE,nodemask=(null),cpuset=/,mems_allowed=0,global_oom,task_memcg=/,task=smbd,pid=3370,uid=0
Thu Jun 17 23:28:05 2021 kern.err kernel: [   58.430534] Out of memory: Killed process 3370 (smbd) total-vm:2888kB, anon-rss:152kB, file-rss:0kB, shmem-rss:144kB, UID:0 pgtables:44kB oom_score_adj:0
Thu Jun 17 23:28:11 2021 daemon.info dnsmasq[10310]: Connected to system UBus
```
[system.log](https://github.com/coolsnowwolf/lede/files/6671576/system.log)

LeanFly, 2021-06-18:
i5-7400 ，vmware，今天拉了新代码编译，启动后出现同样的情况

closed 2021-06-20 by coolsnowwolf

LeanFly, 2021-06-22:
这个咋解决的，我编译的还有这个问题。。。。

sallyciel, 2021-06-26:
> 这个咋解决的，我编译的还有这个问题。。。。

并没有解决