coolsnowwolf/lede #7059: 求救啊，一直报这个错，求大神帮忙

mzl0408, 2021-06-17:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：ake[3]: Entering directory '/home/meng/lede/feeds/kenzo/lua-maxminddb'
mkdir -p /home/meng/lede/dl
SHELL= flock /home/meng/lede/tmp/.lua-maxminddb-0.2.tar.gz.flock -c '  	/home/meng/lede/scripts/download.pl "/home/meng/lede/dl" "lua-maxminddb-0.2.tar.gz" "f9b870968f75c8020a311cca2482139ac8b2b55d4f1b5b0ce00f7844b083c866" "" || (  /home/meng/lede/scripts/dl_github_archive.py --dl-dir="/home/meng/lede/dl" --url="https://github.com/fabled/lua-maxminddb.git" --version="93da9f4e6c814c3a23044dd2cdd22d4a6b4f665b" --subdir="lua-maxminddb-0.2" --source="lua-maxminddb-0.2.tar.gz" --hash="f9b870968f75c8020a311cca2482139ac8b2b55d4f1b5b0ce00f7844b083c866" || ( 	echo "Checking out files from the git repository..."; mkdir -p /home/meng/lede/tmp/dl && cd /home/meng/lede/tmp/dl && rm -rf lua-maxminddb-0.2 && [ \! -d lua-maxminddb-0.2 ] && git clone  https://github.com/fabled/lua-maxminddb.git lua-maxminddb-0.2 && (cd lua-maxminddb-0.2 && git checkout 93da9f4e6c814c3a23044dd2cdd22d4a6b4f665b && git submodule update --init --recursive) && echo "Packing checkout..." && export TAR_TIMESTAMP=`cd lua-maxminddb-0.2 && git log -1 --format='\''@%ct'\''` && rm -rf lua-maxminddb-0.2/.git && 	tar --numeric-owner --owner=0 --group=0 --mode=a-s --sort=name ${TAR_TIMESTAMP:+--mtime="$TAR_TIMESTAMP"} -c lua-maxminddb-0.2 | 	gzip -nc > /home/meng/lede/tmp/dl/lua-maxminddb-0.2.tar.gz && mv /home/meng/lede/tmp/dl/lua-maxminddb-0.2.tar.gz /home/meng/lede/dl/ && rm -rf lua-maxminddb-0.2; );  )    '
Hash of the local file lua-maxminddb-0.2.tar.gz does not match (file: 8aff0958546e4082d48854fdeb32f4fd1ef6a9f78ef066a1886f4d9569898d0e, requested: f9b870968f75c8020a311cca2482139ac8b2b55d4f1b5b0ce00f7844b083c866) - deleting download.
gzip: warning: GZIP environment variable is deprecated; use an alias or script
lua-maxminddb-0.2.tar.gz: Download from https://github.com/fabled/lua-maxminddb.git failed
lua-maxminddb-0.2.tar.gz: Wrong hash (probably caused by .gitattributes), expecting f9b870968f75c8020a311cca2482139ac8b2b55d4f1b5b0ce00f7844b083c866, got 8aff0958546e4082d48854fdeb32f4fd1ef6a9f78ef066a1886f4d9569898d0e
Checking out files from the git repository...
Cloning into 'lua-maxminddb-0.2'...
remote: Enumerating objects: 14, done.
remote: Total 14 (delta 0), reused 0 (delta 0), pack-reused 14
Unpacking objects: 100% (14/14), 3.46 KiB | 443.00 KiB/s, done.
Note: switching to '93da9f4e6c814c3a23044dd2cdd22d4a6b4f665b'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

HEAD is now at 93da9f4 Fixed implicit declaration of gai_strerror() compiler warning, that lead to occational segfualts, i.e. if the address we attempt to look up is invalid
Packing checkout...
touch /home/meng/lede/build_dir/target-aarch64_generic_musl/lua-maxminddb-0.2/.prepared_4c2323bb00c5c880597dc1ca493fa647_6664517399ebbbc92a37c5bb081b5c53_check
. /home/meng/lede/include/shell.sh; gzip -dc /home/meng/lede/dl/lua-maxminddb-0.2.tar.gz | tar -C /home/meng/lede/build_dir/target-aarch64_generic_musl/lua-maxminddb-0.2/.. -xf -
[ ! -d ./src/ ] || cp -fpR ./src/. /home/meng/lede/build_dir/target-aarch64_generic_musl/lua-maxminddb-0.2
touch /home/meng/lede/build_dir/target-aarch64_generic_musl/lua-maxminddb-0.2/.prepared_4c2323bb00c5c880597dc1ca493fa647_6664517399ebbbc92a37c5bb081b5c53
rm -f /home/meng/lede/build_dir/target-aarch64_generic_musl/lua-maxminddb-0.2/.configured_*
rm -f /home/meng/lede/staging_dir/target-aarch64_generic_musl/stamp/.lua-maxminddb_installed
touch /home/meng/lede/build_dir/target-aarch64_generic_musl/lua-maxminddb-0.2/.configured_68b329da9893e34099c7d8ad5cb9c940
rm -f /home/meng/lede/build_dir/target-aarch64_generic_musl/lua-maxminddb-0.2/.built
touch /home/meng/lede/build_dir/target-aarch64_generic_musl/lua-maxminddb-0.2/.built_check
MAKEFLAGS="" /home/meng/lede/staging_dir/host/bin/ninja  -j1 -C /home/meng/lede/build_dir/target-aarch64_generic_musl/lua-maxminddb-0.2 
ninja: Entering directory `/home/meng/lede/build_dir/target-aarch64_generic_musl/lua-maxminddb-0.2'
ninja: error: loading 'build.ninja': No such file or directory
make[3]: *** [Makefile:57: /home/meng/lede/build_dir/target-aarch64_generic_musl/lua-maxminddb-0.2/.built] Error 1
make[3]: Leaving directory '/home/meng/lede/feeds/kenzo/lua-maxminddb'
time: package/feeds/kenzo/lua-maxminddb/compile#0.30#0.07#4.30
    ERROR: package/feeds/kenzo/lua-maxminddb failed to build.
make[2]: *** [package/Makefile:116: package/feeds/kenzo/lua-maxminddb/compile] Error 1
make[2]: Leaving directory '/home/meng/lede'
make[1]: *** [package/Makefile:110: /home/meng/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/meng/lede'
make: *** [/home/meng/lede/include/toplevel.mk:230：world] 错误 2

starxu1971, 2021-06-17:
https://github.com/jerrykuku/lua-maxminddb
用这个库的替换就行了。

mzl0408, 2021-06-18:
> https://github.com/jerrykuku/lua-maxminddb
> 用这个库的替换就行了。

替换这个package/feeds/kenzo/lua-maxminddb里面的makefile吗？

starxu1971, 2021-06-18:
是的

Beginner-Go, 2021-06-18:
原作者Package库不香？非得用杂交Package库？

EnnawYang, 2021-06-18:
> 原作者Package库不香？非得用杂交Package库？

香的一批

mzl0408, 2021-06-18:
> 是的

这个是因为vssr的原因导致的吗？

starxu1971, 2021-06-18:
好几个需要这个依赖

mzl0408, 2021-06-19:
> 好几个需要这个依赖

编译成功了，非常感谢。有个疑问，为什么我编译出来的内核版本不是5.10，还是5.4.125，我的是Rockchip平台

closed 2021-06-19 by coolsnowwolf