coolsnowwolf/lede #7066: K2编译出错

liwenjie119, 2021-06-18:
看log是不是更新后dtb出问题了，缺了k2的dtb？

/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/linux-5.4.125/scripts/dtc/dtc -O dtb -i../dts/ -Wno-unit_address_vs_reg -Wno-simple_bus_reg -Wno-unit_address_format -Wno-pci_bridge -Wno-pci_device_bus_num -Wno-pci_device_reg -Wno-avoid_unnecessary_addr_size -Wno-alias_paths -Wno-graph_child_address -Wno-graph_port -Wno-unique_unit_address   -o /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/image-mt7620a_tplink_archer-c20-v1.dtb /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/image-mt7620a_tplink_archer-c20-v1.dtb.tmp
rm -f /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/image-mt7620a_tplink_archer-c20-v1.dtb.tmp
mipsel-openwrt-linux-musl-cpp -nostdinc -x assembler-with-cpp -I/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/linux-5.4.125/arch/mips/boot/dts -I/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/linux-5.4.125/arch/mips/boot/dts/include -I/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/linux-5.4.125/include/ -undef -D__DTS__  -o /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/image-mt7620a_tplink_archer-c2-v1.dtb.tmp ../dts/mt7620a_tplink_archer-c2-v1.dts
/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/linux-5.4.125/scripts/dtc/dtc -O dtb -i../dts/ -Wno-unit_address_vs_reg -Wno-simple_bus_reg -Wno-unit_address_format -Wno-pci_bridge -Wno-pci_device_bus_num -Wno-pci_device_reg -Wno-avoid_unnecessary_addr_size -Wno-alias_paths -Wno-graph_child_address -Wno-graph_port -Wno-unique_unit_address   -o /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/image-mt7620a_tplink_archer-c2-v1.dtb /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/image-mt7620a_tplink_archer-c2-v1.dtb.tmp
rm -f /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/image-mt7620a_tplink_archer-c2-v1.dtb.tmp
mipsel-openwrt-linux-musl-cpp -nostdinc -x assembler-with-cpp -I/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/linux-5.4.125/arch/mips/boot/dts -I/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/linux-5.4.125/arch/mips/boot/dts/include -I/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/linux-5.4.125/include/ -undef -D__DTS__  -o /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/image-_tplink_archer-c5-v4.dtb.tmp ../dts/_tplink_archer-c5-v4.dts
mipsel-openwrt-linux-musl-cpp: error: ../dts/_tplink_archer-c5-v4.dts: No such file or directory
mipsel-openwrt-linux-musl-cpp: warning: '-x assembler-with-cpp' after last input file has no effect
mipsel-openwrt-linux-musl-cpp: fatal error: no input files
Makefile:206: recipe for target '/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/image-_tplink_archer-c5-v4.dtb' failed
compilation terminated.
make[5]: Leaving directory '/workdir/openwrt/target/linux/ramips/image'
make[5]: *** [/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/image-_tplink_archer-c5-v4.dtb] Error 1
make[4]: *** [install] Error 2
Makefile:24: recipe for target 'install' failed
make[3]: *** [install] Error 2
make[4]: Leaving directory '/workdir/openwrt/target/linux/ramips'
Makefile:11: recipe for target 'install' failed
make[3]: Leaving directory '/workdir/openwrt/target/linux'
time: target/linux/install#51.81#13.57#62.28
    ERROR: target/linux failed to build.
make[2]: *** [target/linux/install] Error 1
target/Makefile:23: recipe for target 'target/linux/install' failed
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [/workdir/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.target_install] Error 2
target/Makefile:19: recipe for target '/workdir/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.target_install' failed
make[1]: Leaving directory '/workdir/openwrt'
/workdir/openwrt/include/toplevel.mk:228: recipe for target 'world' failed
make: *** [world] Error 2

xcf13363175, 2021-06-18:
一样的情况

slipdaly, 2021-06-19:
同样的错误
`2021-06-19T03:19:00.7491173Z /home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/linux-5.4.125/scripts/dtc/dtc -O dtb -i../dts/ -Wno-unit_address_vs_reg -Wno-simple_bus_reg -Wno-unit_address_format -Wno-pci_bridge -Wno-pci_device_bus_num -Wno-pci_device_reg -Wno-avoid_unnecessary_addr_size -Wno-alias_paths -Wno-graph_child_address -Wno-graph_port -Wno-unique_unit_address   -o /home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/image-mt7620a_tplink_archer-c20i.dtb /home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/image-mt7620a_tplink_archer-c20i.dtb.tmp
2021-06-19T03:19:00.7518062Z /home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/linux-5.4.125/scripts/dtc/dtc -O dtb -i../dts/ -Wno-unit_address_vs_reg -Wno-simple_bus_reg -Wno-unit_address_format -Wno-pci_bridge -Wno-pci_device_bus_num -Wno-pci_device_reg -Wno-avoid_unnecessary_addr_size -Wno-alias_paths -Wno-graph_child_address -Wno-graph_port -Wno-unique_unit_address   -o /home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/image-mt7620a_tplink_archer-c20-v1.dtb /home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/image-mt7620a_tplink_archer-c20-v1.dtb.tmp
2021-06-19T03:19:00.7528275Z rm -f /home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/image-mt7620a_tplink_archer-c20i.dtb.tmp
2021-06-19T03:19:00.7549712Z mipsel-openwrt-linux-musl-cpp -nostdinc -x assembler-with-cpp -I/home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/linux-5.4.125/arch/mips/boot/dts -I/home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/linux-5.4.125/arch/mips/boot/dts/include -I/home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/linux-5.4.125/include/ -undef -D__DTS__  -o /home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/image-mt7620a_tplink_archer-c2-v1.dtb.tmp ../dts/mt7620a_tplink_archer-c2-v1.dts
2021-06-19T03:19:00.7567623Z rm -f /home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/image-mt7620a_tplink_archer-c20-v1.dtb.tmp
2021-06-19T03:19:00.7593757Z mipsel-openwrt-linux-musl-cpp -nostdinc -x assembler-with-cpp -I/home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/linux-5.4.125/arch/mips/boot/dts -I/home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/linux-5.4.125/arch/mips/boot/dts/include -I/home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/linux-5.4.125/include/ -undef -D__DTS__  -o /home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/image-_tplink_archer-c5-v4.dtb.tmp ../dts/_tplink_archer-c5-v4.dts
2021-06-19T03:19:00.7619314Z /home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/linux-5.4.125/scripts/dtc/dtc -O dtb -i../dts/ -Wno-unit_address_vs_reg -Wno-simple_bus_reg -Wno-unit_address_format -Wno-pci_bridge -Wno-pci_device_bus_num -Wno-pci_device_reg -Wno-avoid_unnecessary_addr_size -Wno-alias_paths -Wno-graph_child_address -Wno-graph_port -Wno-unique_unit_address   -o /home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/image-mt7620a_tplink_archer-c2-v1.dtb /home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/image-mt7620a_tplink_archer-c2-v1.dtb.tmp
2021-06-19T03:19:00.7623931Z mipsel-openwrt-linux-musl-cpp: error: ../dts/_tplink_archer-c5-v4.dts: No such file or directory
2021-06-19T03:19:00.7630802Z mipsel-openwrt-linux-musl-cpp: warning: '-x assembler-with-cpp' after last input file has no effect
2021-06-19T03:19:00.7633373Z mipsel-openwrt-linux-musl-cpp: fatal error: no input files
2021-06-19T03:19:00.7634553Z compilation terminated.
2021-06-19T03:19:00.7636484Z make[5]: *** [Makefile:206: /home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/image-_tplink_archer-c5-v4.dtb] Error 1
2021-06-19T03:19:00.7637625Z make[5]: *** Waiting for unfinished jobs....
2021-06-19T03:19:00.7666923Z rm -f /home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/image-mt7620a_tplink_archer-c2-v1.dtb.tmp
2021-06-19T03:19:00.7689807Z make[5]: Leaving directory '/home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/target/linux/ramips/image'
2021-06-19T03:19:00.7690707Z make[4]: *** [Makefile:24: install] Error 2
2021-06-19T03:19:00.7693341Z make[3]: *** [Makefile:11: install] Error 2
2021-06-19T03:19:00.7694475Z make[4]: Leaving directory '/home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/target/linux/ramips'
[... cut ...]
2021-06-19T03:19:00.7742779Z make[1]: Leaving directory '/home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt'
2021-06-19T03:19:00.7747569Z make[1]: *** [target/Makefile:19: /home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.target_install] Error 2
2021-06-19T03:19:00.7749380Z make: *** [/home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/include/toplevel.mk:230: world] Error 2
2021-06-19T03:19:00.7777581Z ##[error]Process completed with exit code 2.
`

closed 2021-06-19 by coolsnowwolf