coolsnowwolf/lede #7068: DDNS 3322无法运行

AndyLin2020, 2021-06-18:
### (1) 具体问题

A：之前版本正常，新版出错

Fri Jun 18 14:52:31 2021 user.notice ddns-scripts[8428]: 3322: PID '8428' started at 2021-06-18 14:52
Fri Jun 18 14:52:35 2021 user.crit ddns-scripts[8428]: 3322: Error in 'get_registered_ip()' - no supported Name Server lookup software accessible - TERMINATE
Fri Jun 18 14:52:35 2021 user.warn ddns-scripts[8428]: 3322: PID '8428' exit WITH ERROR '1' at 2021-06-18 14:52

### (3) 详细日志

A：
 145230       : ************ ************** ************** **************
 145231  note : PID '8428' started at 2021-06-18 14:52
 145231       : ddns version  : 2.7.8-3
 145231       : uci configuration:
ddns.3322.domain='a.3322.org'
ddns.3322.enabled='1'
ddns.3322.force_interval='24'
ddns.3322.lookup_host='a.3322.org'
ddns.3322.password='***PW***'
ddns.3322.service_name='3322.org'
ddns.3322.username='a'
ddns.3322=service
 145232       : verbose mode  : 0 - run normal, NO console output
 145233       : check interval: 600 seconds
 145233       : force interval: 86400 seconds
 145233       : retry interval: 60 seconds
 145233       : retry counter : 0 times
 145234       : No old process
 145234       : last update: never
 145234       : Detect registered/public IP
 145234  CRIT : Error in 'get_registered_ip()' - no supported Name Server lookup software accessible - TERMINATE
 145235  WARN : PID '8428' exit WITH ERROR '1' at 2021-06-18 14:52

mkmichael, 2021-06-19:
同样的问题，求大神出现......

coolsnowwolf, 2021-06-19:
[image]
 手动加上这个配置，有些 ISP 封了 3322

mjyhj, 2021-06-19:
还是无效

ddns.F3322=service
 050608       : 
 050609       : Detect registered/public IP
 050609       : verbose mode  : 0 - run normal, NO console output
 050609       : check interval: 600 seconds
 050609       : #> /usr/bin/nslookup f3322.***.net 114.114.114.114 >/var/run/ddns/F3322.dat 2>/var/run/ddns/F3322.err
 050609       : force interval: 259200 seconds
 050610 ERROR : BusyBox nslookup error: '1'
 050610       : retry interval: 60 seconds
 050610       : 
 050610       : retry counter : 0 times
 050610       : 'SIGTERM' was send to old process
 050610  note : PID '28967' terminated by 'SIGTERM' at 2021-06-20 05:06
 050610       : last update: never
 050610       : Verify DNS server '114.114.114.114'
 050611       : #> timeout 2 -- /usr/bin/nc 114.114.114.114 53 </dev/null >/var/run/ddns/F3322.dat 2>/var/run/ddns/F3322.err
 050611       : Detect registered/public IP
 050611       : #> /usr/bin/nslookup f3322.***.net 114.114.114.114 >/var/run/ddns/F3322.dat 2>/var/run/ddns/F3322.err
 050612 ERROR : BusyBox nslookup error: '1'
 050612       : 
 050612       : Detect registered/public IP
 050612       : #> /usr/bin/nslookup f3322.***.net 114.114.114.114 >/var/run/ddns/F3322.dat 2>/var/run/ddns/F3322.err
 050613 ERROR : BusyBox nslookup error: '1'
 050613       : 
 050613  WARN : Get registered/public IP for 'f3322.***.net' failed - retry 1/0 in 60 seconds

AndyLin2020, 2021-06-20:
解决了，发现源码编译没加入nslookup，重新装上即可

closed 2021-06-20 by coolsnowwolf