coolsnowwolf/lede #7070: luci-lib-nixio无法编译成功

nevinen, 2021-06-18:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：luci-lib-nixio无法编译成功，使用的是2021-06-18中午12点左右最新的代码，是完全删除整个lede文件夹后重新clone重新配置feed.conf重新配置menuconfig后编译的，已经如此清空三次编译，均出现同一个错误．

### (2) 路由器型号和固件版本

A：x86软路由

### (3) 详细日志

A：
```
make[3]: Entering directory '/home/evine/lede/feeds/luci/libs/luci-lib-nixio'
touch /home/evine/lede/build_dir/target-x86_64_musl/luci-lib-nixio/.prepared_0f6e76506fb4925cf6e08f2cbd8158eb_6664517399ebbbc92a37c5bb081b5c53_check
for d in luasrc htdocs root src; do if [ -d ./$d ]; then mkdir -p /home/evine/lede/build_dir/target-x86_64_musl/luci-lib-nixio/$d; cp -fpR ./$d/* /home/evine/lede/build_dir/target-x86_64_musl/luci-lib-nixio/$d/; fi; done
true
[ ! -d ./src/ ] || cp -fpR ./src/. /home/evine/lede/build_dir/target-x86_64_musl/luci-lib-nixio
touch /home/evine/lede/build_dir/target-x86_64_musl/luci-lib-nixio/.prepared_0f6e76506fb4925cf6e08f2cbd8158eb_6664517399ebbbc92a37c5bb081b5c53
rm -f /home/evine/lede/build_dir/target-x86_64_musl/luci-lib-nixio/.configured_*
rm -f /home/evine/lede/staging_dir/target-x86_64_musl/stamp/.luci-lib-nixio_installed
touch /home/evine/lede/build_dir/target-x86_64_musl/luci-lib-nixio/.configured_98b31b11569cdb1701bb10c5b1d192c3
rm -f /home/evine/lede/build_dir/target-x86_64_musl/luci-lib-nixio/.built
touch /home/evine/lede/build_dir/target-x86_64_musl/luci-lib-nixio/.built_check
CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/evine/lede/build_dir/target-x86_64_musl/luci-lib-nixio=luci-lib-nixio -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/evine/lede/staging_dir/toolchain-x86_64_gcc-10.3.0_musl/usr/include -I/home/evine/lede/staging_dir/toolchain-x86_64_gcc-10.3.0_musl/include/fortify -I/home/evine/lede/staging_dir/toolchain-x86_64_gcc-10.3.0_musl/include " CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/evine/lede/build_dir/target-x86_64_musl/luci-lib-nixio=luci-lib-nixio -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/evine/lede/staging_dir/toolchain-x86_64_gcc-10.3.0_musl/usr/include -I/home/evine/lede/staging_dir/toolchain-x86_64_gcc-10.3.0_musl/include/fortify -I/home/evine/lede/staging_dir/toolchain-x86_64_gcc-10.3.0_musl/include " LDFLAGS="-L/home/evine/lede/staging_dir/toolchain-x86_64_gcc-10.3.0_musl/usr/lib -L/home/evine/lede/staging_dir/toolchain-x86_64_gcc-10.3.0_musl/lib -znow -zrelro " FPIC="-DPIC -fpic" LUCI_VERSION="git-21.167.28955-fccaab9" LUCI_GITBRANCH="LuCI Master" make -j1 -C /home/evine/lede/build_dir/target-x86_64_musl/luci-lib-nixio/src/ AR="x86_64-openwrt-linux-musl-gcc-ar" AS="x86_64-openwrt-linux-musl-gcc -c -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/evine/lede/build_dir/target-x86_64_musl/luci-lib-nixio=luci-lib-nixio -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=x86_64-openwrt-linux-musl-ld NM="x86_64-openwrt-linux-musl-gcc-nm" CC="x86_64-openwrt-linux-musl-gcc" GCC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" RANLIB="x86_64-openwrt-linux-musl-gcc-ranlib" STRIP=x86_64-openwrt-linux-musl-strip OBJCOPY=x86_64-openwrt-linux-musl-objcopy OBJDUMP=x86_64-openwrt-linux-musl-objdump SIZE=x86_64-openwrt-linux-musl-size CROSS="x86_64-openwrt-linux-musl-" ARCH="x86_64" clean compile;
make[4]: Entering directory '/home/evine/lede/build_dir/target-x86_64_musl/luci-lib-nixio/src'
rm -f *.o *.so *.a *.dll
rm -f axTLS/.prepared
x86_64-openwrt-linux-musl-gcc  -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/evine/lede/build_dir/target-x86_64_musl/luci-lib-nixio=luci-lib-nixio -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/evine/lede/staging_dir/toolchain-x86_64_gcc-10.3.0_musl/usr/include -I/home/evine/lede/staging_dir/toolchain-x86_64_gcc-10.3.0_musl/include/fortify -I/home/evine/lede/staging_dir/toolchain-x86_64_gcc-10.3.0_musl/include  -std=gnu99 -DNO_TLS  -DPIC -fpic -c -o nixio.o nixio.c 
x86_64-openwrt-linux-musl-gcc  -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/evine/lede/build_dir/target-x86_64_musl/luci-lib-nixio=luci-lib-nixio -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/evine/lede/staging_dir/toolchain-x86_64_gcc-10.3.0_musl/usr/include -I/home/evine/lede/staging_dir/toolchain-x86_64_gcc-10.3.0_musl/include/fortify -I/home/evine/lede/staging_dir/toolchain-x86_64_gcc-10.3.0_musl/include  -std=gnu99 -DNO_TLS  -DPIC -fpic -c -o socket.o socket.c 
x86_64-openwrt-linux-musl-gcc  -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/evine/lede/build_dir/target-x86_64_musl/luci-lib-nixio=luci-lib-nixio -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/evine/lede/staging_dir/toolchain-x86_64_gcc-10.3.0_musl/usr/include -I/home/evine/lede/staging_dir/toolchain-x86_64_gcc-10.3.0_musl/include/fortify -I/home/evine/lede/staging_dir/toolchain-x86_64_gcc-10.3.0_musl/include  -std=gnu99 -DNO_TLS  -DPIC -fpic -c -o sockopt.o sockopt.c 
x86_64-openwrt-linux-musl-gcc  -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/evine/lede/build_dir/target-x86_64_musl/luci-lib-nixio=luci-lib-nixio -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/evine/lede/staging_dir/toolchain-x86_64_gcc-10.3.0_musl/usr/include -I/home/evine/lede/staging_dir/toolchain-x86_64_gcc-10.3.0_musl/include/fortify -I/home/evine/lede/staging_dir/toolchain-x86_64_gcc-10.3.0_musl/include  -std=gnu99 -DNO_TLS  -DPIC -fpic -c -o bind.o bind.c 
x86_64-openwrt-linux-musl-gcc  -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/evine/lede/build_dir/target-x86_64_musl/luci-lib-nixio=luci-lib-nixio -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/evine/lede/staging_dir/toolchain-x86_64_gcc-10.3.0_musl/usr/include -I/home/evine/lede/staging_dir/toolchain-x86_64_gcc-10.3.0_musl/include/fortify -I/home/evine/lede/staging_dir/toolchain-x86_64_gcc-10.3.0_musl/include  -std=gnu99 -DNO_TLS  -DPIC -fpic -c -o address.o address.c 
x86_64-openwrt-linux-musl-gcc  -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/evine/lede/build_dir/target-x86_64_musl/luci-lib-nixio=luci-lib-nixio -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/evine/lede/staging_dir/toolchain-x86_64_gcc-10.3.0_musl/usr/include -I/home/evine/lede/staging_dir/toolchain-x86_64_gcc-10.3.0_musl/include/fortify -I/home/evine/lede/staging_dir/toolchain-x86_64_gcc-10.3.0_musl/include  -std=gnu99 -DNO_TLS  -DPIC -fpic -c -o protoent.o protoent.c 
x86_64-openwrt-linux-musl-gcc  -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/evine/lede/build_dir/target-x86_64_musl/luci-lib-nixio=luci-lib-nixio -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/evine/lede/staging_dir/toolchain-x86_64_gcc-10.3.0_musl/usr/include -I/home/evine/lede/staging_dir/toolchain-x86_64_gcc-10.3.0_musl/include/fortify -I/home/evine/lede/staging_dir/toolchain-x86_64_gcc-10.3.0_musl/include  -std=gnu99 -DNO_TLS  -DPIC -fpic -c -o poll.o poll.c 
x86_64-openwrt-linux-musl-gcc  -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/evine/lede/build_dir/target-x86_64_musl/luci-lib-nixio=luci-lib-nixio -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/evine/lede/staging_dir/tool [... cut ...]
/home/evine/lede/staging_dir/toolchain-x86_64_gcc-10.3.0_musl/lib/gcc/x86_64-openwrt-linux-musl/10.3.0/../../../../x86_64-openwrt-linux-musl/bin/ld: user.o:(.rodata+0x0): multiple definition of `nixio__bin2hex'; nixio.o:(.rodata+0x0): first defined here
collect2: error: ld returned 1 exit status
make[4]: *** [Makefile:88: compile] Error 1
make[4]: Leaving directory '/home/evine/lede/build_dir/target-x86_64_musl/luci-lib-nixio/src'
make[3]: *** [../../luci.mk:242: /home/evine/lede/build_dir/target-x86_64_musl/luci-lib-nixio/.built] Error 2
make[3]: Leaving directory '/home/evine/lede/feeds/luci/libs/luci-lib-nixio'
time: package/feeds/luci/luci-lib-nixio/compile#1.33#0.29#1.58
```

nevinen, 2021-06-19:
在`Advanced configuration options`中设置编译器为gcc10后，编译不成功，还原成默认的gcc8就行了。

closed 2021-06-19 by coolsnowwolf

Gaojianli, 2021-07-26:
同样的问题，请问这个问题解决了吗？