coolsnowwolf/lede #7072: 硬盘使用 NTFS3 挂载后中文文件/目录不能显示

Hyy2001X, 2021-06-18:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：ntfs 格式硬盘使用 `mount -t ntfs3 /dev/sdb1 /mnt/sdb1` 命令挂载后看不到带有中文字符的文件或者目录

另外,添加 `-o iocharset=cp936`或者`-o locale=zh_CN.UTF-8` 参数也都不能看到中文目录

### (2) 路由器型号和固件版本

A：x86_64

### (3) 详细日志

A：挂载后执行`ls /mnt/sdb1`时的输出如下,中文路径都变成了`_`
```
ls: /mnt/sdb1/____: No such file or directory
ls: /mnt/sdb1/__: No such file or directory
ls: /mnt/sdb1/___: No such file or directory
21996.1.210529-1541.co_release_CLIENT_CONSUMER_x64FRE_en-us.iso  Downloads   test
```

coolsnowwolf, 2021-06-19:
@LGA1150 确认下有这个问题么~

AmadeusGhost, 2021-06-20:
试试挂载的时候加上 `nls=utf8` 这个参数

gztingting, 2021-06-20:
6月19号编译的版本，ntfs自动挂载正常了，中文能正确显示。
17号编译的版本，ntfs无法正常挂载，看起来已修复这个问题。

Hyy2001X, 2021-06-20:
> 试试挂载的时候加上 `nls=utf8` 这个参数

可以显示中文了 `mount -t ntfs3 -o nls=utf8 /dev/sdb1 /mnt/PT`
```
root@J4105:/# ls /mnt/PT
2021-06-19 16-00-05 的屏幕截图.png                          Downloads                                                        youhua_wr1200js
21996.1.210529-1541.co_release_CLIENT_CONSUMER_x64FRE_en-us.iso  Openwrt-Backups-2021-06-17-c4b44.tar.gz                          桌面备份
BT                                                               board.json                                                       电影
Backups                                                          test                                                             电视剧
Common                                                           x86_64
```

closed 2021-06-20 by Hyy2001X

110120787, 2022-01-04:
现在还是有这个问题