coolsnowwolf/lede #7075: 新版本docker不能用

huang826, 2021-06-18:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：输入/etc/docker-web后报错

### (2) 路由器型号和固件版本

A：x86

### (3) 详细日志

A：
root@OpenWrt:/# /etc/docker-web
Unable to find image 'portainer/portainer-ce:latest' locally
latest: Pulling from portainer/portainer-ce
94cfa856b2b1: Pull complete 
49d59ee0881a: Pull complete 
3fc1bc38fb56: Pull complete 
Digest: sha256:f133bfdd9646f48a8ddb54b9555070f628caa530924045eeff8144a32de2dcc7
Status: Downloaded newer image for portainer/portainer-ce:latest
763292d9aff26627272cc623f59c5f22c32bae56dd59cd6feae9f387e2b2f6bd
docker: Error response from daemon: OCI runtime create failed: /var/lib/docker/overlay2/f3a51b955faf64858cd2d11de54aa3cf9c6ec65c759692c27af8ae2dd8240e94/merged is not an absolute path or is a symlink: unknown.

Promix953, 2021-06-18:
编译时需选中docker

DreamWalkerXZ, 2021-06-19:
> 编译时需选中docker

选中后仍然出现此问题

shu0802, 2021-06-19:
> 反馈bug/问题模板，提建议请删除
> 
> ## 1.关于你要提交的问题
> Q：是否搜索了issue (使用 "x" 选择)
> 
> * [x]  没有类似的issue
> 
> ## 2. 详细叙述
> ### (1) 具体问题
> A：输入/etc/docker-web后报错
> 
> ### (2) 路由器型号和固件版本
> A：x86
> 
> ### (3) 详细日志
> A：
> root@OpenWrt:/# /etc/docker-web
> Unable to find image 'portainer/portainer-ce:latest' locally
> latest: Pulling from portainer/portainer-ce
> 94cfa856b2b1: Pull complete
> 49d59ee0881a: Pull complete
> 3fc1bc38fb56: Pull complete
> Digest: sha256:f133bfdd9646f48a8ddb54b9555070f628caa530924045eeff8144a32de2dcc7
> Status: Downloaded newer image for portainer/portainer-ce:latest
> 763292d9aff26627272cc623f59c5f22c32bae56dd59cd6feae9f387e2b2f6bd
> docker: Error response from daemon: OCI runtime create failed: /var/lib/docker/overlay2/f3a51b955faf64858cd2d11de54aa3cf9c6ec65c759692c27af8ae2dd8240e94/merged is not an absolute path or is a symlink: unknown.

修改下面的根目录与上面的一样，就可以用了
[image]

DreamWalkerXZ, 2021-06-19:
> 修改下面的根目录与上面的一样，就可以用了
> [image]

我找到了等效的方法：
创建`/etc/docker/daemon.json`然后在里面添加如下内容
```
{
	"data-root": "/opt/docker"
}
```
重启
然后就可以用了

coolsnowwolf, 2021-06-19:
fixed

closed 2021-06-19 by coolsnowwolf