coolsnowwolf/lede #7083: 新版5.10固件登录密码问题

Michaol, 2021-06-19:
x86软路由，备份5.4的配置，然后不保留升级5.10。导入配置后，页面root密码无法登陆，既不是原来的个人密码，也不是password。但是通过SSH用原个人密码可以登录，passwd修改root密码，页面登录依然显示密码错误。SSH刷回5.4，一切正常了。

还有没有其他小伙伴有这个问题？

lexin8, 2021-06-19:
R21.6.14这个固件问题太多了，如果代码间隔一两个月以上会有这个情况。现在回滚代码编译R21.6.1失败了，难搞！

EnnawYang, 2021-06-19:
您这操作和不保留配置有什么区别？

EnnawYang, 2021-06-19:
您这操作和保留配置升级有什么区别？

Michaol, 2021-06-19:
> 您这操作和保留配置升级有什么区别？

没有。大佬说x86可以这样上去而已。但是这个跟登录密码有关系吗？

shu0802, 2021-06-19:
> x86软路由，备份5.4的配置，然后不保留升级5.10。导入配置后，页面root密码无法登陆，既不是原来的个人密码，也不是password。但是通过SSH用原个人密码可以登录，passwd修改root密码，页面登录依然显示密码错误。SSH刷回5.4，一切正常了。
> 
> 还有没有其他小伙伴有这个问题？

同样遇到了这人问题，后来重置密码后，重启，仍没有解决。正准备重刷机，连接后再重置密码，又好了。原因不知。

evenwangcn, 2021-06-19:
碰到同样的问题，这个版本的问题确实有点多

Michaol, 2021-06-19:
> 碰到同样的问题，这个版本的问题确实有点多

看来还是暂时待在5.4比较好，稍等个把月，等大佬们修复一下再上不迟。

EnnawYang, 2021-06-19:
大更新**建议**不保留配置，全新安装重新配置

Michaol, 2021-06-19:
> 大更新**建议**不保留配置，全新安装重新配置

臣妾做不到啊……工作量太大了，甚至有些配置好的规则自己都忘了，全部重来会死人的……

coolsnowwolf, 2021-06-19:
官方代码rpcd 的ubus 位置变了，如果需要保留配置一定会不能web登陆

coolsnowwolf, 2021-06-19:
这个不是bug，是必然，不存在所谓修复

closed 2021-06-19 by coolsnowwolf

Michaol, 2021-06-19:
> 这个不是bug，是必然，不存在所谓修复

明解！

hahach, 2021-07-15:
我用了个可能比较奇怪的方式解决了这个问题，但是不清楚有没有不明后遗症。
升级5.10内核的固件后，还原备份，这个时候无法登陆，但是其他功能好像都是正常能用的，然后用ssh登陆，编辑/etc/config/rpcd
将option socket /var/run/ubus.sock 修改为option socket /var/run/ubus/ubus.sock，然后重启rpcd服务，然后就能登陆了...