coolsnowwolf/lede #7085: mt7620这类机器编译的bug

hdm9527, 2021-06-19:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：请删除`lede/linux/ramips/image/mt7620.mk`下的一段代码，文件缺失导致编译失败，删掉这段代码成功编译，openwrt没有这段代码成功编译，不知道这个文件有啥作用
**代码**
```mk
define Device/tplink_archer-c5-v4
  $(Device/Archer)
  TPLINK_FLASHLAYOUT := 8Mmtk
  TPLINK_HWID := 0x04DA857C
  TPLINK_HWREV := 0x0C000600
  TPLINK_HWREVADD := 0x04000000
  IMAGES += factory.bin
  DEVICE_MODEL := Archer C5
  DEVICE_VARIANT := v4
  DEVICE_PACKAGES := kmod-usb2 kmod-usb-ohci kmod-usb-ledtrig-usbport kmod-mt76x2 kmod-switch-rtl8367b
endef
TARGET_DEVICES += tplink_archer-c5-v4
```
### (2) 路由器型号和固件版本

A：斐讯k2 A2版

### (3) 详细日志

A：**编译错误日志**
```bash
mipsel-openwrt-linux-musl-cpp -nostdinc -x assembler-with-cpp -I/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/linux-5.4.125/arch/mips/boot/dts -I/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/linux-5.4.125/arch/mips/boot/dts/include -I/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/linux-5.4.125/include/ -undef -D__DTS__ -o /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/image-_tplink_archer-c5-v4.dtb.tmp ../dts/_tplink_archer-c5-v4.dts
mipsel-openwrt-linux-musl-cpp: error: ../dts/_tplink_archer-c5-v4.dts: No such file or directory
mipsel-openwrt-linux-musl-cpp: warning: '-x assembler-with-cpp' after last input file has no effect
mipsel-openwrt-linux-musl-cpp: fatal error: no input files
Makefile:206: recipe for target '/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/image-_tplink_archer-c5-v4.dtb' failed
compilation terminated.
make[5]: Leaving directory '/workdir/openwrt/target/linux/ramips/image'
make[5]: *** [/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/image-_tplink_archer-c5-v4.dtb] Error 1
make[4]: *** [install] Error 2
```

WongKwongYu, 2021-06-19:
bcm53xx也是这个报错

xcf13363175, 2021-06-19:
复现bug

cdcvip, 2021-06-19:
mipsel-openwrt-linux-musl-cpp: error: ../dts/_tplink_archer-c5-v4.dts: No such file or directory
mipsel-openwrt-linux-musl-cpp: warning: '-x assembler-with-cpp' after last input file has no effect
mipsel-openwrt-linux-musl-cpp: fatal error: no input files
Makefile:206: recipe for target '/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/image-_tplink_archer-c5-v4.dtb' failed
compilation terminated.
make[5]: Leaving directory '/workdir/openwrt/target/linux/ramips/image'
make[5]: *** [/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/image-_tplink_archer-c5-v4.dtb] Error 1
Makefile:24: recipe for target 'install' failed
make[4]: *** [install] Error 2
make[4]: Leaving directory '/workdir/openwrt/target/linux/ramips'
make[3]: *** [install] Error 2
Makefile:11: recipe for target 'install' failed
make[3]: Leaving directory '/workdir/openwrt/target/linux'
time: target/linux/install#8.66#2.77#11.11
    ERROR: target/linux failed to build.
make[2]: *** [target/linux/install] Error 1
target/Makefile:23: recipe for target 'target/linux/install' failed
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [/workdir/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.target_install] Error 2
target/Makefile:19: recipe for target '/workdir/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.target_install' failed
make[1]: Leaving directory '/workdir/openwrt'
make: *** [world] Error 2
/workdir/openwrt/include/toplevel.mk:228: recipe for target 'world' failed
Error: Process completed with exit code 2.

一样...k2

closed 2021-06-19 by coolsnowwolf