coolsnowwolf/lede #7102: RedmiAC2100京东签到插件消失 手动编译ipk nodejs编译失败

kisaragychihaya, 2021-06-20:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [X ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：menuconfig时京东签到插件消失不见，手动单独编译ipk显示nodejs编译报错

### (2) 路由器型号和固件版本

A：Redmi AC2100 commit 1eed21188933daf956f1b44591b4dd92a4f51647

### (3) 详细日志

A：
```
{standard input}: Assembler messages:
{standard input}:130: Error: opcode not supported on this processor: mips32r2 (mips32r2) `ldc1 $f0,24($fp)'
{standard input}:131: Error: opcode not supported on this processor: mips32r2 (mips32r2) `mtc1 $t0,$f1'
{standard input}:132: Error: opcode not supported on this processor: mips32r2 (mips32r2) `sdc1 $f0,24($fp)'
make[4]: *** [tools/v8_gypfiles/v8_libbase.target.mk:169: /home/cjs/lede/build_dir/target-mipsel_24kc_musl/node-v14.17.1/out/Release/obj.target/v8_libbase/deps/v8/src/base/cpu.o] Error 1
make[4]: *** Waiting for unfinished jobs....
  mipsel-openwrt-linux-musl-g++ -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -o /home/cjs/lede/build_dir/target-mipsel_24kc_musl/node-v14.17.1/out/Release/obj.target/v8_libbase/deps/v8/src/base/ieee754.o ../deps/v8/src/base/ieee754.cc '-DV8_GYP_BUILD' '-DV8_TYPED_ARRAY_MAX_SIZE_IN_HEAP=64' '-D__STDC_FORMAT_MACROS' '-DV8_TARGET_ARCH_MIPS' '-DCAN_USE_FPU_INSTRUCTIONS' '-D__mips_hard_float=1' '-DV8_EMBEDDER_STRING="-node.67"' '-DENABLE_DISASSEMBLER' '-DV8_PROMISE_INTERNAL_FIELD_COUNT=1' '-DENABLE_MINOR_MC' '-DOBJECT_PRINT' '-DV8_CONCURRENT_MARKING' '-DV8_ARRAY_BUFFER_EXTENSION' '-DV8_ENABLE_LAZY_SOURCE_POSITIONS' '-DV8_USE_SIPHASH' '-DDISABLE_UNTRUSTED_CODE_MITIGATIONS' '-DV8_WIN64_UNWINDING_INFO' '-DV8_ENABLE_REGEXP_INTERPRETER_THREADED_DISPATCH' '-DV8_SNAPSHOT_COMPRESSION' '-DBUILDING_V8_BASE_SHARED' '-D_MIPS_TARGET_HW' '-D_MIPS_ARCH_MIPS32R2' '-DFPU_MODE_FP32' -I/home/cjs/lede/staging_dir/target-mipsel_24kc_musl/usr/include -I../deps/v8 -I../deps/v8/include  -pthread -Wno-unused-parameter -Wno-return-type -fno-strict-aliasing -EL -Wno-error=array-bounds -mhard-float -mips32r2 -mfp32 -Wa,-mips32r2 -O3 -fno-omit-frame-pointer -fdata-sections -ffunction-sections -O3 -fno-rtti -fno-exceptions -std=gnu++1y -MMD -MF /home/cjs/lede/build_dir/target-mipsel_24kc_musl/node-v14.17.1/out/Release/.deps//home/cjs/lede/build_dir/target-mipsel_24kc_musl/node-v14.17.1/out/Release/obj.target/v8_libbase/deps/v8/src/base/ieee754.o.d.raw  -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -fmacro-prefix-map=/home/cjs/lede/build_dir/target-mipsel_24kc_musl/node-v14.17.1=node-v14.17.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/cjs/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/usr/include -I/home/cjs/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/include/fortify -I/home/cjs/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/include  -c
rm 697a30682ef00e61ae59c1f9981b83ce1b2a9209.intermediate 2e79d13be6a41d1436a0bb21eaf6a8cf6ba26e64.intermediate
make[3]: *** [Makefile:104: node] Error 2
make[3]: Leaving directory '/home/cjs/lede/build_dir/target-mipsel_24kc_musl/node-v14.17.1'
make[2]: *** [Makefile:160: /home/cjs/lede/build_dir/target-mipsel_24kc_musl/node-v14.17.1/.built] Error 2
make[2]: Leaving directory '/home/cjs/lede/feeds/packages/lang/node'
time: package/feeds/packages/node/compile#165.82#9.38#39.24
    ERROR: package/feeds/packages/node failed to build.
make[1]: *** [package/Makefile:116: package/feeds/packages/node/compile] Error 1
make[1]: Leaving directory '/home/cjs/lede'
make: *** [/home/cjs/lede/include/toplevel.mk:230: package/lean/luci-app-jd-dailybonus/compile] Error 2
```

minlang112, 2021-06-20:
是的，node编译会失败，试了两次都是如此

coolsnowwolf, 2021-06-20:
nodejs 新版本不再支持没有 FPU 的 CPU 了，换其他 CPU 吧

coolsnowwolf, 2021-06-20:
[image]

minlang112, 2021-06-20:
版本好像回滚到14.7.0了……

coolsnowwolf, 2021-06-20:
加入了最后一个支持 MIPS 的 v12.22.1 版本，以后不会有新版本支持了

其他有 FPU 的继续用 v14

kisaragychihaya, 2021-06-20:
AX3600 menuconfig还是没有签到插件 ，但是手动编译是成功了
```
installing /home/xxx/lede/build_dir/target-aarch64_cortex-a53_musl/node-v14.17.0/ipkg-install/usr/include/node/cppgc/internal/gc-info.h
installing /home/xxx/lede/build_dir/target-aarch64_cortex-a53_musl/node-v14.17.0/ipkg-install/usr/include/node/cppgc/internal/api-constants.h
installing /home/xxx/lede/build_dir/target-aarch64_cortex-a53_musl/node-v14.17.0/ipkg-install/usr/include/node/cppgc/internal/accessors.h
installing /home/xxx/lede/build_dir/target-aarch64_cortex-a53_musl/node-v14.17.0/ipkg-install/usr/include/node/cppgc/internal/prefinalizer-handler.h
installing /home/xxx/lede/build_dir/target-aarch64_cortex-a53_musl/node-v14.17.0/ipkg-install/usr/include/node/cppgc/internal/pointer-policies.h
installing /home/xxx/lede/build_dir/target-aarch64_cortex-a53_musl/node-v14.17.0/ipkg-install/usr/include/node/libplatform/v8-tracing.h
installing /home/xxx/lede/build_dir/target-aarch64_cortex-a53_musl/node-v14.17.0/ipkg-install/usr/include/node/libplatform/libplatform.h
installing /home/xxx/lede/build_dir/target-aarch64_cortex-a53_musl/node-v14.17.0/ipkg-install/usr/include/node/libplatform/libplatform-export.h
make[3]: Leaving directory '/home/xxx/lede/build_dir/target-aarch64_cortex-a53_musl/node-v14.17.0'
touch /home/xxx/lede/build_dir/target-aarch64_cortex-a53_musl/node-v14.17.0/.built
rm -rf /home/xxx/lede/tmp/stage-node
mkdir -p /home/xxx/lede/tmp/stage-node/host /home/xxx/lede/staging_dir/target-aarch64_cortex-a53_musl/packages
install -d -m0755 /home/xxx/lede/tmp/stage-node/usr/include
cp -fpR /home/xxx/lede/build_dir/target-aarch64_cortex-a53_musl/node-v14.17.0/ipkg-install/usr/include/* /home/xxx/lede/tmp/stage-node/usr/include/
find /home/xxx/lede/tmp/stage-node -name '*.la' | xargs -r rm -f;
if [ -f /home/xxx/lede/staging_dir/target-aarch64_cortex-a53_musl/packages/node.list ]; then /home/xxx/lede/scripts/clean-package.sh "/home/xxx/lede/staging_dir/target-aarch64_cortex-a53_musl/packages/node.list" "/home/xxx/lede/staging_dir/target-aarch64_cortex-a53_musl"; fi
if [ -d /home/xxx/lede/tmp/stage-node ]; then (cd /home/xxx/lede/tmp/stage-node; find ./ > /home/xxx/lede/tmp/stage-node.files);        SHELL= flock /home/xxx/lede/tmp/.staging-dir.flock -c ' mv /home/xxx/lede/tmp/stage-node.files /home/xxx/lede/staging_dir/target-aarch64_cortex-a53_musl/packages/node.list && cp -fpR /home/xxx/lede/tmp/stage-node/* /home/xxx/lede/staging_dir/target-aarch64_cortex-a53_musl/; '; fi
rm -rf /home/xxx/lede/tmp/stage-node
touch /home/xxx/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.node_installed
make[2]: Leaving directory '/home/xxx/lede/feeds/packages/lang/node'
time: package/feeds/packages/node/compile#5384.92#221.62#965.02
make[2]: Entering directory '/home/xxx/lede/package/lean/luci-app-jd-dailybonus'
make[2]: Nothing to be done for 'compile'.
make[2]: Leaving directory '/home/xxx/lede/package/lean/luci-app-jd-dailybonus'
time: package/lean/luci-app-jd-dailybonus/compile#0.11#0.02#0.16
make[1]: Leaving directory '/home/xxx/lede'
```

closed 2021-06-20 by coolsnowwolf

kisaragychihaya, 2021-06-20:
小米AX3600应该可以通过 make 包\compile V=s编译成功node v14了，但是menuconfig里面还是没有
[image]

```
installing /home/xxx/lede/build_dir/target-aarch64_cortex-a53_musl/node-v14.17.0/ipkg-install/usr/include/node/cppgc/internal/gc-info.h
installing /home/xxx/lede/build_dir/target-aarch64_cortex-a53_musl/node-v14.17.0/ipkg-install/usr/include/node/cppgc/internal/api-constants.h
installing /home/xxx/lede/build_dir/target-aarch64_cortex-a53_musl/node-v14.17.0/ipkg-install/usr/include/node/cppgc/internal/accessors.h
installing /home/xxx/lede/build_dir/target-aarch64_cortex-a53_musl/node-v14.17.0/ipkg-install/usr/include/node/cppgc/internal/prefinalizer-handler.h
installing /home/xxx/lede/build_dir/target-aarch64_cortex-a53_musl/node-v14.17.0/ipkg-install/usr/include/node/cppgc/internal/pointer-policies.h
installing /home/xxx/lede/build_dir/target-aarch64_cortex-a53_musl/node-v14.17.0/ipkg-install/usr/include/node/libplatform/v8-tracing.h
installing /home/xxx/lede/build_dir/target-aarch64_cortex-a53_musl/node-v14.17.0/ipkg-install/usr/include/node/libplatform/libplatform.h
installing /home/xxx/lede/build_dir/target-aarch64_cortex-a53_musl/node-v14.17.0/ipkg-install/usr/include/node/libplatform/libplatform-export.h
make[3]: Leaving directory '/home/xxx/lede/build_dir/target-aarch64_cortex-a53_musl/node-v14.17.0'
touch /home/xxx/lede/build_dir/target-aarch64_cortex-a53_musl/node-v14.17.0/.built
rm -rf /home/xxx/lede/tmp/stage-node
mkdir -p /home/xxx/lede/tmp/stage-node/host /home/xxx/lede/staging_dir/target-aarch64_cortex-a53_musl/packages
install -d -m0755 /home/xxx/lede/tmp/stage-node/usr/include
cp -fpR /home/xxx/lede/build_dir/target-aarch64_cortex-a53_musl/node-v14.17.0/ipkg-install/usr/include/* /home/xxx/lede/tmp/stage-node/usr/include/
find /home/xxx/lede/tmp/stage-node -name '*.la' | xargs -r rm -f;
if [ -f /home/xxx/lede/staging_dir/target-aarch64_cortex-a53_musl/packages/node.list ]; then /home/xxx/lede/scripts/clean-package.sh "/home/xxx/lede/staging_dir/target-aarch64_cortex-a53_musl/packages/node.list" "/home/xxx/lede/staging_dir/target-aarch64_cortex-a53_musl"; fi
if [ -d /home/xxx/lede/tmp/stage-node ]; then (cd /home/xxx/lede/tmp/stage-node; find ./ > /home/xxx/lede/tmp/stage-node.files);        SHELL= flock /home/xxx/lede/tmp/.staging-dir.flock -c ' mv /home/xxx/lede/tmp/stage-node.files /home/xxx/lede/staging_dir/target-aarch64_cortex-a53_musl/packages/node.list && cp -fpR /home/xxx/lede/tmp/stage-node/* /home/xxx/lede/staging_dir/target-aarch64_cortex-a53_musl/; '; fi
rm -rf /home/xxx/lede/tmp/stage-node
touch /home/xxx/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.node_installed
make[2]: Leaving directory '/home/xxx/lede/feeds/packages/lang/node'
time: package/feeds/packages/node/compile#5384.92#221.62#965.02
make[2]: Entering directory '/home/xxx/lede/package/lean/luci-app-jd-dailybonus'
make[2]: Nothing to be done for 'compile'.
make[2]: Leaving directory '/home/xxx/lede/package/lean/luci-app-jd-dailybonus'
time: package/lean/luci-app-jd-dailybonus/compile#0.11#0.02#0.16
make[1]: Leaving directory '/home/xxx/lede'
```