coolsnowwolf/lede #7109: 最新代码编译后固件无法更新

dunkeng, 2021-06-20:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：无法更新，在网页提示：不支持所上传的映像文件格式，请选择适合当前平台的通用映像文件。
在后台提示：
root@OpenWrt:/tmp# sysupgrade openwrt-mvebu-cortexa9-linksys_wrt1200ac-squashfs-sysupgrade.bin
Device linksys,wrt1200ac not supported by this image
Supported devices: linksys,wrt1200ac armada-385-linksys-caiman linksys,caiman - Image version mismatch: image 1.1, device 1.0. Please wipe config during upgrade (force required) or reinstall. Reason: Config cannot be migrated from swconfig to DSA
Image check failed.

### (2) 路由器型号和固件版本

A：路由器是LINKSYS WRT1200AC
固件取最新代码：
87e2f2912806ab7d8cc5dd90fd3069f1ff9c5fcb

### (3) 详细日志

A：
root@OpenWrt:/tmp# sysupgrade openwrt-mvebu-cortexa9-linksys_wrt1200ac-squashfs-sysupgrade.bin
Device linksys,wrt1200ac not supported by this image
Supported devices: linksys,wrt1200ac armada-385-linksys-caiman linksys,caiman - Image version mismatch: image 1.1, device 1.0. Please wipe config during upgrade (force required) or reinstall. Reason: Config cannot be migrated from swconfig to DSA
Image check failed.
附带.config文件与关键的机型行：
CONFIG_TARGET_mvebu=y
CONFIG_TARGET_mvebu_cortexa9=y
CONFIG_TARGET_mvebu_cortexa9_DEVICE_linksys_wrt1200ac=y
CONFIG_TARGET_BOARD="mvebu"
CONFIG_TARGET_SUBTARGET="cortexa9"
CONFIG_TARGET_PROFILE="DEVICE_linksys_wrt1200ac"
[.config.zip](https://github.com/coolsnowwolf/lede/files/6682429/default.config.zip)

coolsnowwolf, 2021-06-20:
Please wipe config during upgrade (force required) or reinstall. Reason: Config cannot be migrated from swconfig to DSA
Image check failed.

coolsnowwolf, 2021-06-20:
不能保留配置升级

minlang112, 2021-06-20:
ARM的升级5.10的话不能保留配置

dunkeng, 2021-06-20:
成功了，用-n参数不行，网上查了，用了-n -v -F三个参数成功升级，谢谢

日志：
root@OpenWrt:/tmp# sysupgrade -n openwrt-mvebu-cortexa9-linksys_wrt1200ac-squashfs-sysupgrade.bin
Device linksys,wrt1200ac not supported by this image
Supported devices: linksys,wrt1200ac armada-385-linksys-caiman linksys,caiman - Image version mismatch: image 1.1, device 1.0. Please wipe config during upgrade (force required) or reinstall. Reason: Config cannot be migrated from swconfig to DSA
Image check failed.
root@OpenWrt:/tmp# sysupgrade -n -v -F openwrt-mvebu-cortexa9-linksys_wrt1200ac-squashfs-sysupgrade.bin
Device linksys,wrt1200ac not supported by this image
Supported devices: linksys,wrt1200ac armada-385-linksys-caiman linksys,caiman - Image version mismatch: image 1.1, device 1.0. Please wipe config during upgrade (force required) or reinstall. Reason: Config cannot be migrated from swconfig to DSA
Image check failed but --force given - will update anyway!
Commencing upgrade. Closing all shell sessions.

Remote side unexpectedly closed network connection

dunkeng, 2021-06-20:
> ARM的升级5.10的话不能保留配置

已经不保留配置升级了，谢谢，但内核还是5.4，并没有升级
root@OpenWrt:~# uname -a
Linux OpenWrt 5.4.125 #0 SMP Sun Jun 20 07:45:22 2021 armv7l GNU/Linux

closed 2021-06-20 by dunkeng