coolsnowwolf/lede #7113: 6-20编译版本用了docker后cpu负荷高

gztingting, 2021-06-21:
开启docker后，
拉取镜像的过程，CPU usage在70~100%，以前正常2~3分钟搞定，现在要十几分钟。
创建容器的过程，CPU usage一直满负荷，以前正常3~5分钟搞定，现在要二十几分钟。
开启容器后，进入容器，明显感觉比较卡顿，在容器内安装组件，非常慢，修改apt sources为国内镜像后，连apt update也都超级慢，添加一个Mariadb要一个多小时，记得以前很快的。
机器还是那台i5 4300的Thinkpad X250电脑，
高CPU负荷，导致luci更新数据切换页面时比较慢，以前瞬间刷新，现在有的页面要几十秒，按理说docker拿到了系统，容器内部应该比较快，但容器内部也比较慢。
请教各路大神，到底是什么原因。
Intel(R) Core(TM) i5-4300U CPU @ 1.90GHz : 2 Core 4 Thread
固件OpenWrt R21.6.14 / LuCI Master (git-21.170.32194-b1a5082)
内核5.10.43
编译时选了luci-app-dockerman，luci-app-docker是最新版本，dockerd几乎全选编译。

xiaobaishu1, 2021-06-21:
cpufreq 这个插件前几天修复之后一直都是满载运行的

coolsnowwolf, 2021-06-21:
cpufreq 看看是不是正确频率了

gztingting, 2021-06-21:
luci状态--概况页的cpu型号显示是Intel(R) Core(TM) i5-4300U CPU @ 1.90GHz : 2 Core 4 Thread，貌似这是对的。
请教在哪里设置cpufreq正确频率？

gztingting, 2021-06-21:
概况页可以看到CPU频率不断在变化

lexin8, 2021-06-21:
居然还能运行，我提交issues都关了
[image]

xingkong0113, 2021-06-21:
我这里是正常的，以下是机器运行信息：
Intel(R) Core(TM) i3-6100U CPU @ 2.30GHz : 2 Core 4 Thread (CpuMark : 48943.882779 Scores)
固件版本
OpenWrt R21.6.14 / LuCI Master (git-21.170.32194-b1a5082)
内核版本
5.10.43
平均负载
0.15, 0.04, 0.01
CPU 使用率（%）
3 %

gztingting, 2021-06-21:
> 我这里是正常的，以下是机器运行信息：
> Intel(R) Core(TM) i3-6100U CPU @ 2.30GHz : 2 Core 4 Thread (CpuMark : 48943.882779 Scores)
> 固件版本
> OpenWrt R21.6.14 / LuCI Master (git-21.170.32194-b1a5082)
> 内核版本
> 5.10.43
> 平均负载
> 0.15, 0.04, 0.01
> CPU 使用率（%）
> 3 %

以前我编译的版本，还有cpumark信息，我昨天编译的版本就没有了呢。
我哪里设置出了问题？

coolsnowwolf, 2021-06-21:
干净编译测试，没有发现问题

gztingting, 2021-06-21:
我回退到3月19日编译的一个版本（5.4.105内核openWrt R21.2.1 / LuCI Master (git-21.035.34289-13f96d1)），测试后发现木有这个问题，docker正常，在cpu高负荷时刷新页面正常，真不知道哪里有问题。
我编译的版本，没有git clone任何外部的包，都是云编译https://github.com/coolsnowwolf/lede主分支。为什么以前的版本和别人最新编译的版本有cpumark，我的就没有了呢。

lexin8, 2021-06-21:
> 我回退到3月19日编译的一个版本（5.4.105内核openWrt R21.2.1 / LuCI Master (git-21.035.34289-13f96d1)），测试后发现木有这个问题，docker正常，在cpu高负荷时刷新页面正常，真不知道哪里有问题。
> 我编译的版本，没有git [...]

怎么回退编译docker? 退到5月份编译提示缺少docker-ce
https://github.com/coolsnowwolf/lede/issues/7104#issue-925537626

ZqinKing, 2021-06-22:
> > 我这里是正常的，以下是机器运行信息：
> > Intel(R) Core(TM) i3-6100U CPU @ 2.30GHz : 2 Core 4 Thread (CpuMark : 48943.882779 Scores)
> > 固件版本
> > OpenWrt R21.6.14 / LuCI Master (git-21.170.32194-b1a5082)
> > 内核版本
> > 5.10.43
> > 平均负载
> > 0.15, 0.04, 0.01
> > CPU 使用率（%）
> > 3 %
> 
> 以前我编译的版本，还有cpumark信息，我昨天编译的版本就没有了呢。
> 我哪里设置出了问题？

cpumark是定时任务执行的只会执行一次。等着第二天就有了。

closed 2021-06-22 by gztingting