coolsnowwolf/lede #7118: 是否可能移殖openwrt官方的luci-app-advanced-reboot

dunkeng, 2021-06-21:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：是否可能移殖openwrt官方的luci-app-advanced-reboot
目前lede里面带的luci-app-advanced-reboot是不可用的
我自己试过openwrt官方版本是可以用的，但直接复制他的代码替换掉lede里面的版本编译，或者下载官方IPK安装均不出WEB界面
我搞不懂原因，百度不出来相关信息

### (2) 路由器型号和固件版本

A：wrt1200ac，最新代码

### (3) 详细日志

A：没有可提供的日志

coolsnowwolf, 2021-06-21:
不可用 是什么意思

dunkeng, 2021-06-21:
> 不可用 是什么意思

抱歉，没注意打错字了，是WEB界面没有出来
我查那个/usr/lib/lua/luci/view/admin_system目录里面没有对应的htm文件

AmadeusGhost, 2021-06-21:
试试这个 commit：https://github.com/AmadeusGhost/luci/commit/56b116da4c98096603a17be914aac283d432e283

dunkeng, 2021-06-21:
在我WRT1200AC测试过了，还是和以前一样，麻烦你看看我这操作有没有问题？
[image]
[image]
刚刚跑完

恢复环境：
git fetch --all
git reset --hard origin/master 
git pull

改feeds.conf.default，只列出这两行，其他没有没列出
#src-git luci https://github.com/coolsnowwolf/luci
src-git luci https://github.com/AmadeusGhost/luci

刷新并重新配置
./scripts/feeds update -a 
./scripts/feeds install -a
make menuconfig 

编译，保留配置刷新

dunkeng, 2021-06-21:
这个是否有对应命令行？我可以执行后把日志发上来

AmadeusGhost, 2021-06-21:
> 在我WRT1200AC测试过了，还是和以前一样，麻烦你看看我这操作有没有问题？

这样改的话补丁没打上的，你试试这样：
```
cd feeds/luci
wget https://github.com/AmadeusGhost/luci/commit/56b116da4c98096603a17be914aac283d432e283.patch
git apply *.patch
```

dunkeng, 2021-06-21:
问题解决了，功能存在，测试了可以正常切换分区，就是界面格式没有了，有点奇怪，不过不影响使用了。
重启前在2分区：
[image]

重启后在1分区：
root@OpenWrt:~# fw_printenv boot_part
boot_part=1

谢谢了。

coolsnowwolf, 2021-06-22:
[image]
这一段建议保留之前 table 的写法而非 div，否则css 样式乱了

closed 2021-06-22 by coolsnowwolf

dunkeng, 2021-06-22:
还是有问题，但我不知道完整日志文件在哪
[image]

dunkeng, 2021-06-22:
@coolsnowwolf @AmadeusGhost 请看一下，还有问题
[image]