coolsnowwolf/lede #7121: fatal error: execinfo.h: No such file or directory  #include <execinfo.h>

minlang112, 2021-06-21:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述
编译勾选DPDK，编译时报错
### (1) 具体问题

A：fatal error: execinfo.h: No such file or directory
 #include <execinfo.h>

### (2) 路由器型号和固件版本

A：X86平台，同步最新版本

### (3) 详细日志
[image]

####(4)似乎可行的修改方法
修改 openwrt/lede/build_dir/target-x86_64_musl/dpdk-21.05/drivers/net/nfp/nfpcore里的nfp_cpp_pcie_ops.c
参照0001-Handling-backtrace-and-execinfo-header-file.patch 里的方式，修改 #include这部分为

+#ifdef __GLIBC__
 #ifdef RTE_BACKTRACE
 #include <execinfo.h>
 #endif
+#endif

然后我编译了一下通过了……

xiexiansen, 2021-10-30:
大哥 你改的方式没看懂 能详细截个图是修改哪个部分么

minlang112, 2021-10-30:
> 大哥 你改的方式没看懂 能详细截个图是修改哪个部分么

[image]

xiexiansen, 2021-11-05:
> 
> 
> > 大哥 你改的方式没看懂 能详细截个图是修改哪个部分么
> 
> [image]

确实可以通过编译，就是不知道这种方式dpdk有没有起作用哈

minlang112, 2021-11-05:
> > > 大哥 你改的方式没看懂 能详细截个图是修改哪个部分么
> > 
> > 
> > [image]
> 
> 确实可以通过编译，就是不知道这种方式dpdk有没有起作用哈

你想多了，这只是一个SDK

xiexiansen, 2021-11-06:
> 
> 
> > > > 大哥 你改的方式没看懂 能详细截个图是修改哪个部分么
> > > 
> > > 
> > > [image]
> > 
> > 
> > 确实可以通过编译，就是不知道这种方式dpdk有没有起作用哈
> 
> 你想多了，这只是一个SDK

本人菜鸟，不太懂，这个确实编译出来ipk文件了 系统里面也集成了，说到底忙了半天没用啊？

closed 2022-02-12 by Beginner-Go