coolsnowwolf/lede #7130: LEDE上部署docker时提示443端口被占用，请问如何关闭？

kingzleshe, 2021-06-22:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：LEDE上部署docker时提示443端口被占用，请问如何关闭？
      应为是反向代理和ssl签发相关的docker image，
      文档上要求部署443端口不能被占用

### (2) 路由器型号和固件版本

A：树莓派4， sulingGG最新的编译版本

### (3) 详细日志

A：

coolsnowwolf, 2021-06-22:
uhttpd 里面关闭

kingzleshe, 2021-06-22:
uhttpd的配置我没有去动过，配置里面没看到有在用443。
是不是得加配置才能关闭？

$ cat /etc/config/uhttpd

config uhttpd 'main'
        list listen_http '0.0.0.0:80'
        list listen_http '[::]:80'
        option redirect_https '0'
        option home '/www'
        option max_connections '100'
        option cert '/etc/uhttpd.crt'
        option key '/etc/uhttpd.key'
        option cgi_prefix '/cgi-bin'
        list lua_prefix '/cgi-bin/luci=/usr/lib/lua/luci/sgi/uhttpd.lua'
        option network_timeout '30'
        option http_keepalive '20'
        option tcp_keepalive '1'
        option ubus_prefix '/ubus'
        option script_timeout '360'
        option max_requests '50'
        list index_page 'cgi-bin/luci'
        option rfc1918_filter '0'

config cert 'defaults'
        option days '730'
        option key_type 'rsa'
        option bits '2048'
        option ec_curve 'P-256'
        option country 'ZZ'
        option state 'Somewhere'
        option location 'Unknown'
        option commonname 'OpenWrt'

kingzleshe, 2021-06-22:
暂时不用了，发现arm平台搞https还是太麻烦

closed 2021-06-22 by kingzleshe