coolsnowwolf/lede #7131: 拉最新源码出现的错误

jxph1123, 2021-06-22:
全部删除重新拉最新源码，执行./scripts/feeds install -a出现这个错误：
Build dependency: Please install the GNU C Compiler (gcc) 6 or later
Build dependency: Please install the GNU C++ Compiler (g++) 6 or later
Build dependency: Please install Python >= 3.6
Build dependency: Please install Python >= 3.6

/home/etek/lede/include/prereq.mk:9: recipe for target 'prereq' failed
Prerequisite check failed. Use FORCE=1 to override.
/home/etek/lede/include/toplevel.mk:179: recipe for target 'staging_dir/host/.prereq-build' failed
make: *** [staging_dir/host/.prereq-build] Error 1

有人知道怎么解决吗。谢谢

missorforget, 2021-06-22:
Please install Python >= 3.6

jxph1123, 2021-06-22:
> Please install Python >= 3.6

如何升级呢，谢谢

jxph1123, 2021-06-22:
搞定了，升级GCC6和Python3.6

closed 2021-06-22 by jxph1123