coolsnowwolf/lede #7132: 友善 r4s 1GB 固件刷了无法启动

KLT007, 2021-06-22:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x ] 没有类似的issue

## 2. 详细叙述
友善 r4s 1GB 固件刷了无法启动
### (1) 具体问题

A：通过编译后，刷上固件，开机没量，无法启动，不过发现写卡日志第二个分区无法写入

### (2) 路由器型号和固件版本

A：R4S 1GB版本

EnnawYang, 2021-06-22:
1GB version is not supported.

KLT007, 2021-06-22:
> 1GB version is not supported.

什么时候支持？

EnnawYang, 2021-06-22:
> > 1GB version is not supported.
> 
> 什么时候支持？

自己换个大点的卡

KLT007, 2021-06-22:
128G TF卡吗？

KLT007, 2021-06-22:
是1G内存卡不支持了还是R4S-1G不支持了？

EnnawYang, 2021-06-22:
> Partition 1:  
> Size: 80 MB (83886080 bytes)
> Partition 2:  
> Size: 1.2 GB (1258291200 bytes)  
> 
总大小超过1G，1G的tf卡当然刷不进

EnnawYang, 2021-06-22:
你是昨晚群里问的那个？

KLT007, 2021-06-22:
昨晚我看他也和我一样问题，所以才来问

KLT007, 2021-06-22:
按你这么说，我用128核重新编译了固件，改了总大小，依然那样问题
Partition 1:
  Type: GNU/Linux (0x83)
  Size: 64 MB (67108864 bytes)
  Start Sector: 65536, Boot: Yes
Partition 2:
  Type: GNU/Linux (0x83)
  Size: 960 MB (1006632960 bytes)
  Start Sector: 262144, Boot: No

coolsnowwolf, 2021-06-22:
采用了官方的代码，根本不支持 1GB 版本了

KLT007, 2021-06-22:
> 采用了官方的代码，根本不支持 1GB 版本了

这样啊，感谢回复

closed 2021-06-22 by KLT007