coolsnowwolf/lede #7133: 自更新dropbear.init后系统日志有报错记录

shikkoku, 2021-06-22:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
https://github.com/coolsnowwolf/lede/blob/a3f1e837fde8f4cebcd41fa7be7dec35d599bc69/package/network/services/dropbear/files/dropbear.init

自此次更新后系统日志出现错误提示

### (1) 具体问题

A：系统日志出现错误提示

### (2) 路由器型号和固件版本

A：x86_64 R21.6.14

### (3) 详细日志

A：Tue Jun 22 14:42:00 2021 daemon.err uhttpd[2626]: sh: /etc/init.d/dropbear: line 13: extra_command: not found

coolsnowwolf, 2021-06-22:
你需要重新编译了，你的base-file 没更新

closed 2021-06-22 by coolsnowwolf

tanst, 2023-11-24:
`/etc/rc.common` 内容已经更新过了，这是固件里面的:
```
extra_command() {
	local cmd="$1"
	local help="$2"

	local extra="$(printf "%-16s%s" "${cmd}" "${help}")"
	ALL_HELP="${ALL_HELP}\t${extra}\n"
	ALL_COMMANDS="${ALL_COMMANDS} ${cmd}"
}
```
这是 `/etc/init.d/dropbear:13` :
```
extra_command "killclients" "Kill ${NAME} processes except servers and yourself"
```
依然报错：
```
Fri Nov 24 09:07:23 2023 daemon.err uhttpd[3337]: sh: /etc/init.d/dropbear: line 13: extra_command: not found
```