coolsnowwolf/lede #7136: 最近固件，加入DIY插件后，无法编译成功？

souwei168, 2021-06-22:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：最近固件，加入DIY插件后，无法编译成功？

### (2) 路由器型号和固件版本

A：X86

### (3) 详细日志

A：

kisaragychihaya, 2021-06-22:
你这啥具体插件 错误日志 都没有 怎么看啊

EnnawYang, 2021-06-22:
啥都不提供，自觉关了吧

souwei168, 2021-06-22:
> 反馈bug/问题模板，提建议请删除
> 
> ## 1.关于你要提交的问题
> Q：是否搜索了issue (使用 "x" 选择)
> 
> * [ ]  没有类似的issue
> 
> ## 2. 详细叙述
> ### (1) 具体问题
> A：最近固件，加入DIY插件后，无法编译成功？
> 
> ### (2) 路由器型号和固件版本
> A：X86
> 
> ### (3) 详细日志
> A：
 make[3] -C feeds/luci/libs/luci-lib-ip compile
 make[3] -C feeds/luci/libs/luci-lib-jsonc compile
make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
/home/sw/openwrt/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 1

EnnawYang, 2021-06-22:
跟没有一样

closed 2021-06-22 by coolsnowwolf