coolsnowwolf/lede #7141: 干净环境编译，tools/ccache编译失败

softtip, 2021-06-22:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

干净环境编译，tools/ccache编译失败

### (2) 路由器型号和固件版本

X86

### (3) 详细日志

[44/77] Building C object src/third_party/CMakeFiles/third_party_lib.dir/xxhash.c.o
[45/77] Building C object src/third_party/CMakeFiles/third_party_lib.dir/xxh_x86dispatch.c.o
[46/77] Linking CXX static library src/third_party/libthird_party_lib.a
[47/77] Building C object src/third_party/blake3/CMakeFiles/blake3.dir/blake3.c.o
[48/77] Building C object src/third_party/blake3/CMakeFiles/blake3.dir/blake3_avx512.c.o
FAILED: src/third_party/blake3/CMakeFiles/blake3.dir/blake3_avx512.c.o 
/usr/bin/env CCACHE_BASEDIR=/mnt/x86/tmp/lede/build_dir/host/ccache-4.2.1 CCACHE_CPP2=true /usr/bin/ccache /mnt/x86/tmp/lede/staging_dir/host/bin/gcc   -O2 -I/mnt/x86/tmp/lede/staging_dir/host/include -DNDEBUG -std=c99 -mavx512f -mavx512vl -MD -MT src/third_party/blake3/CMakeFiles/blake3.dir/blake3_avx512.c.o -MF src/third_party/blake3/CMakeFiles/blake3.dir/blake3_avx512.c.o.d -o src/third_party/blake3/CMakeFiles/blake3.dir/blake3_avx512.c.o -c src/third_party/blake3/blake3_avx512.c
/mnt/x86/tmp/lede/tmp/cc3NXwFB.s: Assembler messages:
/mnt/x86/tmp/lede/tmp/cc3NXwFB.s:16: Error: operand size mismatch for `vmovdqu64'
/mnt/x86/tmp/lede/tmp/cc3NXwFB.s:19: Error: operand size mismatch for `vmovdqu64'
/mnt/x86/tmp/lede/tmp/cc3NXwFB.s:22: Error: operand size mismatch for `vmovdqu64'
/mnt/x86/tmp/lede/tmp/cc3NXwFB.s:25: Error: operand size mismatch for `vmovdqu64'
/mnt/x86/tmp/lede/tmp/cc3NXwFB.s:28: Error: operand size mismatch for `vmovdqu64'
/mnt/x86/tmp/lede/tmp/cc3NXwFB.s:29: Error: operand type mismatch for `vpxorq'
/mnt/x86/tmp/lede/tmp/cc3NXwFB.s:32: Error: operand type mismatch for `vprord'
/mnt/x86/tmp/lede/tmp/cc3NXwFB.s:35: Error: operand type mismatch for `vpxorq'
/mnt/x86/tmp/lede/tmp/cc3NXwFB.s:38: Error: operand type mismatch for `vprord'
/mnt/x86/tmp/lede/tmp/cc3NXwFB.s:42: Error: operand type mismatch for `vpxorq'
/mnt/x86/tmp/lede/tmp/cc3NXwFB.s:45: Error: operand type mismatch for `vprord'

中间省略...

/mnt/x86/tmp/lede/tmp/ccNzD8KY.s:3767: Error: operand size mismatch for `vmovdqu32'
/mnt/x86/tmp/lede/tmp/ccNzD8KY.s:3768: Error: operand size mismatch for `vmovdqu32'
/mnt/x86/tmp/lede/tmp/ccNzD8KY.s:3769: Error: operand size mismatch for `vmovdqu32'
/mnt/x86/tmp/lede/tmp/ccNzD8KY.s:3770: Error: operand size mismatch for `vmovdqu32'
ninja: build stopped: subcommand failed.
make[2]: *** [Makefile:43: /mnt/x86/tmp/lede/build_dir/host/ccache-4.2.1/.built] Error 1
make[2]: Leaving directory '/mnt/x86/tmp/lede/tools/ccache'
time: tools/ccache/compile#3.99#1.08#5.86
    ERROR: tools/ccache failed to build.
make[1]: *** [tools/Makefile:159: tools/ccache/compile] Error 1
make[1]: Leaving directory '/mnt/x86/tmp/lede'
make: *** [/mnt/x86/tmp/lede/include/toplevel.mk:230: tools/ccache/compile] Error 2
root@cs-virtual-machine:/mnt/x86/tmp/lede# 

昨天都还OK
今天git pull后编译失败，
重新git clone拉取后干净编译默认配置x86
失败，求解

luodaoyi, 2021-06-22:
看起来像是你的文件拉取的不对 试试重新 `make -j8 download V=s`

coolsnowwolf, 2021-06-24:
可以禁用 ccache 试试

softtip, 2021-06-24:
搞定了，之前用多线程编译失败，后来再单线程也失败

最后，彻底rm -rf后单线程编译通过

closed 2021-06-24 by softtip