coolsnowwolf/lede #7142: ffmpeg-custom编译失败

yunhai20082008, 2021-06-22:
1. 先选择ffmpeg，libffmpeg-full会被强制选上且不可取消，此时再选中libffmpeg-custom编译时提示冲突
2. 取消ffmpeg和libffmpeg-full，手动选上libffmpeg-custom，此时会出现ffmpeg-custom，选中且增加h264支持，无论编译整个固件还是单独的ipk都报错，前者错误信息：
`SHELL= flock /mnt/sdb/op/lede1/tmp/.ffmpeg-4.3.2.tar.xz.flock -c '      /mnt/sdb/op/lede1/scripts/download.pl "/mnt/sdb/op/lede1/dl" "ffmpeg-4.3.2.tar.xz" "46e4e64f1dd0233cbc0934b9f1c0da676008cad34725113fb7f802cfa84ccddb" "" "https://ffmpeg.org/releases/"    '
mkdir -p /mnt/sdb/op/lede1/bin/targets/x86/64/packages /mnt/sdb/op/lede1/build_dir/target-x86_64_musl/ffmpeg-custom/ffmpeg-4.3.2/ipkg-x86_64/ffmpeg-custom/CONTROL /mnt/sdb/op/lede1/staging_dir/target-x86_64_musl/pkginfo
install -d -m0755 /mnt/sdb/op/lede1/build_dir/target-x86_64_musl/ffmpeg-custom/ffmpeg-4.3.2/ipkg-x86_64/ffmpeg-custom/usr/bin
cp -fpR /mnt/sdb/op/lede1/build_dir/target-x86_64_musl/ffmpeg-custom/ffmpeg-4.3.2/ipkg-install/usr/bin/ffmpeg /mnt/sdb/op/lede1/build_dir/target-x86_64_musl/ffmpeg-custom/ffmpeg-4.3.2/ipkg-x86_64/ffmpeg-custom/usr/bin/
cp: cannot stat '/mnt/sdb/op/lede1/build_dir/target-x86_64_musl/ffmpeg-custom/ffmpeg-4.3.2/ipkg-install/usr/bin/ffmpeg': No such file or directory
Makefile:750: recipe for target '/mnt/sdb/op/lede1/bin/packages/x86_64/packages/ffmpeg-custom_4.3.2-1_x86_64.ipk' failed
make[3]: *** [/mnt/sdb/op/lede1/bin/packages/x86_64/packages/ffmpeg-custom_4.3.2-1_x86_64.ipk] Error 1
make[3]: Leaving directory '/mnt/sdb/op/lede1/feeds/packages/multimedia/ffmpeg'
time: package/feeds/packages/ffmpeg/custom/compile#0.35#0.22#0.81
    ERROR: package/feeds/packages/ffmpeg failed to build (build variant: custom).
package/Makefile:114: recipe for target 'package/feeds/packages/ffmpeg/compile' failed
make[2]: *** [package/feeds/packages/ffmpeg/compile] Error 1
make[2]: Leaving directory '/mnt/sdb/op/lede1'
package/Makefile:110: recipe for target '/mnt/sdb/op/lede1/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/mnt/sdb/op/lede1/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/mnt/sdb/op/lede1'
/mnt/sdb/op/lede1/include/toplevel.mk:228: recipe for target 'world' failed
make: *** [world] Error 2
`

yonggedebaqi, 2021-06-23:
https://blog.51cto.com/neicole/1363552

yunhai20082008, 2021-06-23:
> https://blog.51cto.com/neicole/1363552
这是在ubuntu编译它能用的ffmpeg吧，怎么ubuntu做op定制版的ffmpeg呢，能否指导下

coolsnowwolf, 2021-06-24:
这个需要统一为 ffmpeg-full ，不然编译会冲突

yunhai20082008, 2021-06-24:
> 这个需要统一为 ffmpeg-full ，不然编译会冲突

也就是说要想定制ffmpeg只能改它本身的make file和libffmpeg-full，不能用ffmpeg-custom和libffmpeg-custom 吗？

yonggedebaqi, 2021-06-26:
编译选 ffmpeg-full 
修改/lede/feeds/packages/multimedia/ffmpeg/Makefile文件添加试试吧
	--enable-libx264 \
	--enable-libx265 \

yunhai20082008, 2021-06-26:
> 编译选 ffmpeg-full
> 修改/lede/feeds/packages/multimedia/ffmpeg/Makefile文件添加试试吧
> --enable-libx264
> --enable-libx265 \
感谢老铁解答，问题已经解决。
可能由于自行精简的.config,未编译global building setting中的compile with support for patented functionality导致libx264不能出现在列表中，只要改后，编译进libx264，默认编译的ffmpeg和libffmpeg-full就自动支持h264，可以操作rtsp的h264流，实现用openwrt拉取监控摄像头的数据到本地。
另外分享下经验，经过在VMware workstation测试，ffmpeg添加 -c:v copy xxx.mp4参数，可能是不对流重编码，只是简单的封装，对cpu的耗费，相比只添加-f mp4 xxx.mp4低太多，可以用在x86上定时录像。

closed 2021-06-26 by yunhai20082008

yonggedebaqi, 2021-07-14:
[image]
[image]
[image]
我编译把libx265加入了，固件也编译成功了

yonggedebaqi, 2021-07-14:
[image]

yonggedebaqi, 2021-07-15:
> > 编译选 ffmpeg-full
> > 修改/lede/feeds/packages/multimedia/ffmpeg/Makefile文件添加试试吧
> > --enable-libx264
> > --enable-libx265
> > 感谢老铁解答，问题已经解决，以下内容留给别人做参考。
> > 可能由于自行精简的.config,未编译global building setting中的compile with support for patented functionality导致libx264不能出现在列表中，只要改后，编译进libx264，默认编译的ffmpeg和libffmpeg-full就自动支持h264，可以操作rtsp的h264流，实现用openwrt拉取监控摄像头的数据到本地。
> > 另外分享下经验，经过在VMware workstation测试，ffmpeg添加 -c:v copy xxx.mp4参数，可能是不对流重编码，只是简单的封装，对cpu的耗费，相比只添加-f mp4 xxx.mp4低太多，可以用在x86上定时录像。

你可以写个脚本用OpenWRT，实现ffmpeg定时录

zikura88, 2021-10-16:
我HC5962编译时不管在make menuconfig中选没选中ffmpeg都报ffmpeg的错误，不知还跟哪个插件挂钩？
make[4]: *** [Makefile:124: ffmpeg_g] Error 1
make[4]: Leaving directory '/workdir/openwrt/build_dir/target-mipsel_24kc_musl/ffmpeg-full/ffmpeg-4.4'
make[3]: *** [Makefile:751: /workdir/openwrt/build_dir/target-mipsel_24kc_musl/ffmpeg-full/ffmpeg-4.4/.built] Error 2
make[3]: Leaving directory '/workdir/openwrt/feeds/packages/multimedia/ffmpeg'
time: package/feeds/packages/ffmpeg/full/compile#0.52#0.13#0.62
    ERROR: package/feeds/packages/ffmpeg failed to build (build variant: full).
make[2]: *** [package/Makefile:119: package/feeds/packages/ffmpeg/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2
Error: Process completed with exit code 2.