coolsnowwolf/lede #7144: 无法设置DUID，导致路由不能分配指定后缀的ipv6

iam993, 2021-06-22:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

有类似问题 #6800 

## 2. 详细叙述

### (1) 具体问题

A：如 #6800 所说，路由是用上层链路获取到的PD，再依据与设备DUID关联的ipv6后缀设置项分配ipv6。这个结论是参考了各类衍生固件以及官方原版OpenWrt 19.07.7 r11306-c4a6851c72 后得出的。 

手动修改文件"
 vi /etc/config/dhcp

`config host
  option duid '00xxxx'
  option hostid '2001'`

重启可正常分配ipv6

由此可说明DUID确实与ipv6后缀有关联

另外~！另外~！另外~！并不是所有设备都会根据DUID分配，部分设备不需要DUID也能正常分配ipv6，目前还没找到关联点。

### (2) 路由器型号和固件版本

A：近期版本

shikkoku, 2021-06-22:
[image]
类似这样？

iam993, 2021-06-22:
大概是这样，只是图里这duid好像不大对。

[image]

在
 `/usr/lib/lua/luci/model/cbi/admin_network/dhcp.lua`  
中加入 
 `duid = s:option(Value, "duid", translate("duid"))` 
可手动填写并且正常运作。
但不太懂luci，不知道如何才能下拉。

tanst, 2021-06-23:
之前 5.4 内核版本是正常的，能根据静态 ipv4 的后缀来分配 ipv6
比如你绑定设置了静态 IPv4 : NAS    192.168.99.5
那么 IPv6 就会自动获取：240e:299:4d7:c0e0::5/128 这样后缀都是 5 的地址，很是方便，新版本固件无效，啥也获取不了。

iam993, 2021-06-23:
> 之前 5.4 内核版本是正常的，能根据静态 ipv4 的后缀来分配 ipv6
> 比如你绑定设置了静态 IPv4 : NAS 192.168.99.5
> 那么 IPv6 就会自动获取：240e:299:4d7:c0e0::5/128 这样后缀都是 5 的地址，很是方便，新版本固件无效，啥也获取不了。我之前提过这问题 #7055

看我上面说的设置，改源码可以正常获取对应ipv6后缀的。
虽然路由上分配的ipv6都是固定的，使用上来说没啥区别，就是好看些。

closed 2022-02-12 by Beginner-Go