coolsnowwolf/lede #7146: 因为cache问题编译失败

miansan, 2021-06-23:
Ubuntu编译X64的固件，出现如下错误，这个cache出问题请问如何解决，谢谢！

Signing package index...
make[2]: Leaving directory '/home/miansan/lede'
export MAKEFLAGS= ;make -w -r json_overview_image_info
make[2]: Entering directory '/home/miansan/lede'
make[2]: Nothing to be done for 'json_overview_image_info'.
make[2]: Leaving directory '/home/miansan/lede'
export MAKEFLAGS= ;make -w -r checksum
make[2]: Entering directory '/home/miansan/lede'
make[2]: Leaving directory '/home/miansan/lede'
/home/miansan/lede/staging_dir/host/bin/ccache -s
cache directory                     /home/miansan/lede/.ccache
primary config                      /home/miansan/lede/.ccache/ccache.conf
secondary config (readonly)         /home/miansan/lede/staging_dir/host/etc/ccache.conf
stats updated                       Wed Jun 23 02:57:50 2021
cache hit (direct)                 11718
cache hit (preprocessed)            1164
cache miss                         22336
cache hit rate                     36.58 %
called for link                     1713
called for preprocessing            1924
multiple source files                  4
compiler produced empty output        71
compile failed                      1345
ccache internal error                 32
preprocessor error                   594
bad compiler arguments               176
unsupported source language            6
autoconf compile/link               4238
unsupported compiler option          240
unsupported code directive             9
output to stdout                       4
no input file                        677
cleanups performed                     0
files in cache                     44558
cache size                         567.5 MB
max cache size                       5.0 GB
make[1]: Leaving directory '/home/miansan/lede'

EnnawYang, 2021-06-23:
嗯？ 这个叫错误？

miansan, 2021-06-23:
不好意思，检查了一下，固件编译成功了。

closed 2021-06-23 by miansan