coolsnowwolf/lede #7147: 新的代码编译失败，高手帮忙看看

keeno1982, 2021-06-23:
Generating index for package ./wget-ssl_1.21.1-3_aarch64_cortex-a53.ipk
Signing package index...
make[2]: Leaving directory '/home/dietpi/lede'
export MAKEFLAGS= ;make -w -r json_overview_image_info
make[2]: Entering directory '/home/dietpi/lede'
make[2]: Nothing to be done for 'json_overview_image_info'.
make[2]: Leaving directory '/home/dietpi/lede'
export MAKEFLAGS= ;make -w -r checksum
make[2]: Entering directory '/home/dietpi/lede'
make[2]: Leaving directory '/home/dietpi/lede'
/home/dietpi/lede/staging_dir/host/bin/ccache -s
cache directory                     /home/dietpi/lede/.ccache
primary config                      /home/dietpi/lede/.ccache/ccache.conf
secondary config (readonly)         /home/dietpi/lede/staging_dir/host/etc/ccache.conf
stats updated                       Wed Jun 23 02:38:07 2021
cache hit (direct)                   327
cache hit (preprocessed)               4
cache miss                          2205
cache hit rate                     13.05 %
called for link                       35
called for preprocessing              28
multiple source files                  3
compiler produced empty output         9
compile failed                        11
preprocessor error                    68
unsupported compiler option            3
unsupported code directive             2
no input file                         95
cleanups performed                     0
files in cache                      4383
cache size                         118.6 MB
max cache size                       5.0 GB
make[1]: Leaving directory '/home/dietpi/lede'

EnnawYang, 2021-06-23:
谁说了这是失败了？

keeno1982, 2021-06-23:
@EnnawYang 那这几个Failed是什么意思啊，以前没遇过
compile failed 11
preprocessor error 68
unsupported compiler option 3
unsupported code directive 2
no input file 95

keeno1982, 2021-06-23:
大概明白了，这个一个编译后的统计项

closed 2021-06-23 by keeno1982