coolsnowwolf/lede #7151: 编译出错 trojan-go

Yofo-me, 2021-06-23:
- Making log

```php
...
SHELL= flock /home/kin/lede/tmp/.trojan-go-0.10.4.tar.gz.flock -c '     /home/kin/lede/scripts/downloa          d.pl "/home/kin/lede/dl" "trojan-go-0.10.4.tar.gz" "487aee96ce31e6f5939ef464c997f7ce6cfb888915ca0e9294          0bdcda65f2e0e0" "" "https://codeload.github.com/p4gefau1t/trojan-go/tar.gz/v0.10.4?"    '
rm -f /home/kin/lede/build_dir/target-x86_64_musl/trojan-go-0.10.4/.built
touch /home/kin/lede/build_dir/target-x86_64_musl/trojan-go-0.10.4/.built_check
CONFIG_GOLANG_MOD_CACHE_WORLD_READABLE="" GO_BUILD_CACHE_DIR="/home/kin/lede/tmp/go-build" GO_MOD_CACH          E_DIR="/home/kin/lede/dl/go-mod-cache" GO_MOD_ARGS="-modcacherw" GO_PKG="github.com/p4gefau1t/trojan-g          o" GO_INSTALL_EXTRA="" GO_INSTALL_ALL="" GO_SOURCE_ONLY="" GO_BUILD_PKG="github.com/p4gefau1t/trojan-g          o" GO_EXCLUDES="" GO_GO_GENERATE="" GO_INSTALL_BIN_PATH="/usr/bin" BUILD_DIR="/home/kin/lede/build_dir          /target-x86_64_musl/trojan-go-0.10.4" GO_BUILD_DIR="/home/kin/lede/build_dir/target-x86_64_musl/trojan          -go-0.10.4/.go_work/build" GO_BUILD_BIN_DIR="/home/kin/lede/build_dir/target-x86_64_musl/trojan-go-0.1          0.4/.go_work/build/bin" GO_BUILD_DEPENDS_PATH="/usr/share/gocode" GO_BUILD_DEPENDS_SRC="/home/kin/lede          /staging_dir/target-x86_64_musl/usr/share/gocode/src" GOOS="linux" GOARCH="amd64" GO386="" GOARM="" GO          MIPS="" GOMIPS64="" CGO_ENABLED=1 CC="ccache_cc" CXX="ccache_cxx" CGO_CFLAGS="-Os -pipe -fno-caller-sa          ves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix          -map=/home/kin/lede/build_dir/target-x86_64_musl/trojan-go-0.10.4=trojan-go-0.10.4 -Wformat -Werror=fo          rmat-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CGO_CPPFLAGS="-I/home/kin          /lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/include -I/home/kin/lede/staging_dir/toolchain-x          86_64_gcc-8.4.0_musl/include/fortify -I/home/kin/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/incl          ude" CGO_CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-varia          ble -Wno-error=unused-result -fmacro-prefix-map=/home/kin/lede/build_dir/target-x86_64_musl/trojan-go-          0.10.4=trojan-go-0.10.4 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,          now -Wl,-z,relro" CGO_LDFLAGS="-L/home/kin/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/lib -L          /home/kin/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib -znow -zrelro" GOPATH="/home/kin/lede/b          uild_dir/target-x86_64_musl/trojan-go-0.10.4/.go_work/build" GOCACHE="/home/kin/lede/tmp/go-build" GOM          ODCACHE="/home/kin/lede/dl/go-mod-cache" GOENV=off /usr/bin/env bash /home/kin/lede/feeds/packages/lan          g/golang//golang-build.sh build -v -trimpath -ldflags "all=-buildid '1624370117' -linkmode external -e          xtldflags '-L/home/kin/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/lib -L/home/kin/lede/stagi          ng_dir/toolchain-x86_64_gcc-8.4.0_musl/lib -Wl,-z,now -Wl,-z,relro'"        -ldflags "-s -w -X github.          com/p4gefau1t/trojan-go/constant.Version=0.10.4 -X github.com/p4gefau1t/trojan-go/constant.Commit=v0.1          0.4 -buildid '1624370117' -linkmode external -extldflags '-L/home/kin/lede/staging_dir/toolchain-x86_6          4_gcc-8.4.0_musl/usr/lib -L/home/kin/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib -Wl,-z,now -          Wl,-z,relro'" -tags "full"
Finding targets

Building targets
go: downloading google.golang.org/genproto v0.0.0-20200526211855-cb27e3aa2013
go: downloading github.com/txthinking/x v0.0.0-20210326105829-476fab902fbe
go: downloading golang.org/x/crypto v0.0.0-20210506145944-38f3c27a63bf
go: downloading golang.org/x/text v0.3.6
/home/kin/lede/dl/go-mod-cache/github.com/shadowsocks/go-shadowsocks2@v0.1.5/shadowaead/cipher.go:11:2          : golang.org/x/crypto@v0.0.0-20210506145944-38f3c27a63bf: local error: tls: bad record MAC
/home/kin/lede/dl/go-mod-cache/github.com/refraction-networking/utls@v0.0.0-20201210053706-2179f286686          b/handshake_messages.go:9:2: golang.org/x/crypto@v0.0.0-20210506145944-38f3c27a63bf: local error: tls:           bad record MAC
/home/kin/lede/dl/go-mod-cache/github.com/refraction-networking/utls@v0.0.0-20201210053706-2179f286686          b/key_schedule.go:12:2: golang.org/x/crypto@v0.0.0-20210506145944-38f3c27a63bf: local error: tls: bad           record MAC
/home/kin/lede/dl/go-mod-cache/github.com/shadowsocks/go-shadowsocks2@v0.1.5/shadowaead/cipher.go:12:2          : golang.org/x/crypto@v0.0.0-20210506145944-38f3c27a63bf: local error: tls: bad record MAC
/home/kin/lede/dl/go-mod-cache/github.com/refraction-networking/utls@v0.0.0-20201210053706-2179f286686          b/u_prng.go:22:2: golang.org/x/crypto@v0.0.0-20210506145944-38f3c27a63bf: local error: tls: bad record           MAC
/home/kin/lede/dl/go-mod-cache/golang.org/x/net@v0.0.0-20210525063256-abc453219eb5/idna/idna10.0.0.go:          26:2: golang.org/x/text@v0.3.6: local error: tls: bad record MAC
/home/kin/lede/dl/go-mod-cache/golang.org/x/net@v0.0.0-20210525063256-abc453219eb5/idna/idna10.0.0.go:          27:2: golang.org/x/text@v0.3.6: local error: tls: bad record MAC
/home/kin/lede/dl/go-mod-cache/golang.org/x/net@v0.0.0-20210525063256-abc453219eb5/idna/idna10.0.0.go:          28:2: golang.org/x/text@v0.3.6: local error: tls: bad record MAC
/home/kin/lede/dl/go-mod-cache/google.golang.org/grpc@v1.38.0/status/status.go:34:2: google.golang.org          /genproto@v0.0.0-20200526211855-cb27e3aa2013: local error: tls: bad record MAC

make[3]: *** [Makefile:78: /home/kin/lede/build_dir/target-x86_64_musl/trojan-go-0.10.4/.built] Error           1
make[3]: Leaving directory '/home/kin/lede/feeds/small/trojan-go'
time: package/feeds/small/trojan-go/compile#0.88#0.93#12.48
    ERROR: package/feeds/small/trojan-go failed to build.
make[2]: *** [package/Makefile:116: package/feeds/small/trojan-go/compile] Error 1
make[2]: Leaving directory '/home/kin/lede'
make[1]: *** [package/Makefile:110: /home/kin/lede/staging_dir/target-x86_64_musl/stamp/.package_compi          le] Error 2
make[1]: Leaving directory '/home/kin/lede'
make: *** [/home/kin/lede/include/toplevel.mk:230：world] 错误 2

```

- Screenshot
[image]

执行`make clean && make download -j8 V=s && make -j1 V=s`仍然报这个错误 
编译环境 `Ubuntu 20.04 lts with Python3.9 on the Contabo dedicated server or local VM Ubuntu system env(Host is on win10).`

Beginner-Go, 2021-06-23:
使用杂交package后报错,不要赖本源码

chobitsos, 2021-06-23:
rm -rf 当前源码全部，重新拉新源码编译

closed 2021-06-23 by coolsnowwolf