coolsnowwolf/lede #7152: ubuntu 20.04.2 package/libselinux编译出错

softtip, 2021-06-23:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

make[3]: Entering directory '/home/cs/disk/x86/tmp/lede/build_dir/target-x86_64_musl/libselinux-3.2'
pkg-config.real: error while loading shared libraries: libpkgconf.so.3: cannot open shared object file: No such file or directory
pkg-config.real: error while loading shared libraries: libpkgconf.so.3: cannot open shared object file: No such file or directory
make[4]: Entering directory '/home/cs/disk/x86/tmp/lede/build_dir/target-x86_64_musl/libselinux-3.2/include'
make[4]: Nothing to be done for 'all'.
make[4]: Leaving directory '/home/cs/disk/x86/tmp/lede/build_dir/target-x86_64_musl/libselinux-3.2/include'
make[4]: Entering directory '/home/cs/disk/x86/tmp/lede/build_dir/target-x86_64_musl/libselinux-3.2/src'
ccache_cc -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/cs/disk/x86/tmp/lede/build_dir/target-x86_64_musl/libselinux-3.2=libselinux-3.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/cs/disk/x86/tmp/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/include -I/home/cs/disk/x86/tmp/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/fortify -I/home/cs/disk/x86/tmp/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include  -I../include -D_GNU_SOURCE  -DNO_ANDROID_BACKEND  -L/home/cs/disk/x86/tmp/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/lib -L/home/cs/disk/x86/tmp/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib -znow -zrelro  -shared -o libselinux.so.1 avc.lo avc_internal.lo avc_sidtab.lo booleans.lo callbacks.lo canonicalize_context.lo checkAccess.lo check_context.lo checkreqprot.lo compute_av.lo compute_create.lo compute_member.lo compute_relabel.lo compute_user.lo context.lo deny_unknown.lo disable.lo enabled.lo fgetfilecon.lo freecon.lo freeconary.lo fsetfilecon.lo get_context_list.lo get_default_type.lo get_initial_context.lo getenforce.lo getfilecon.lo getpeercon.lo init.lo is_customizable_type.lo label.lo label_db.lo label_file.lo label_media.lo label_support.lo label_x.lo lgetfilecon.lo load_policy.lo lsetfilecon.lo mapping.lo matchmediacon.lo matchpathcon.lo policyvers.lo procattr.lo query_user_context.lo regex.lo reject_unknown.lo selinux_check_securetty_context.lo selinux_config.lo selinux_restorecon.lo sestatus.lo setenforce.lo setexecfilecon.lo setfilecon.lo setrans_client.lo seusers.lo sha1.lo stringrep.lo validatetrans.lo  -lfts -ldl -Wl,-soname,libselinux.so.1,--version-script=libselinux.map,-z,defs,-z,relro
/home/cs/disk/x86/tmp/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/bin/../lib/gcc/x86_64-openwrt-linux-musl/8.4.0/../../../../x86_64-openwrt-linux-musl/bin/ld: regex.lo: in function `regex_version':
regex.c:(.text+0x54): undefined reference to `pcre_version'
/home/cs/disk/x86/tmp/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/bin/../lib/gcc/x86_64-openwrt-linux-musl/8.4.0/../../../../x86_64-openwrt-linux-musl/bin/ld: regex.lo: in function `regex_writef':
regex.c:(.text+0x9e): undefined reference to `pcre_fullinfo'
/home/cs/disk/x86/tmp/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/bin/../lib/gcc/x86_64-openwrt-linux-musl/8.4.0/../../../../x86_64-openwrt-linux-musl/bin/ld: regex.c:(.text+0x10c): undefined reference to `pcre_fullinfo'
/home/cs/disk/x86/tmp/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/bin/../lib/gcc/x86_64-openwrt-linux-musl/8.4.0/../../../../x86_64-openwrt-linux-musl/bin/ld: regex.lo: in function `regex_data_free':
regex.c:(.text+0x19a): undefined reference to `pcre_free'
/home/cs/disk/x86/tmp/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/bin/../lib/gcc/x86_64-openwrt-linux-musl/8.4.0/../../../../x86_64-openwrt-linux-musl/bin/ld: regex.c:(.text+0x1ab): undefined reference to `pcre_free_study'
/home/cs/disk/x86/tmp/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/bin/../lib/gcc/x86_64-openwrt-linux-musl/8.4.0/../../../../x86_64-openwrt-linux-musl/bin/ld: regex.lo: in function `regex_match':
regex.c:(.text+0x216): undefined reference to `pcre_exec'
/home/cs/disk/x86/tmp/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/bin/../lib/gcc/x86_64-openwrt-linux-musl/8.4.0/../../../../x86_64-openwrt-linux-musl/bin/ld: regex.lo: in function `regex_cmp':
regex.c:(.text+0x250): undefined reference to `pcre_fullinfo'
/home/cs/disk/x86/tmp/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/bin/../lib/gcc/x86_64-openwrt-linux-musl/8.4.0/../../../../x86_64-openwrt-linux-musl/bin/ld: regex.c:(.text+0x276): undefined reference to `pcre_fullinfo'
/home/cs/disk/x86/tmp/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/bin/../lib/gcc/x86_64-openwrt-linux-musl/8.4.0/../../../../x86_64-openwrt-linux-musl/bin/ld: regex.lo: in function `regex_prepare_data':
regex.c:(.text+0x31d): undefined reference to `pcre_compile'
/home/cs/disk/x86/tmp/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/bin/../lib/gcc/x86_64-openwrt-linux-musl/8.4.0/../../../../x86_64-openwrt-linux-musl/bin/ld: regex.c:(.text+0x35a): undefined reference to `pcre_study'
/home/cs/disk/x86/tmp/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/bin/../lib/gcc/x86_64-openwrt-linux-musl/8.4.0/../../../../x86_64-openwrt-linux-musl/bin/ld: regex.lo: in function `regex_load_mmap':
regex.c:(.text+0x409): undefined reference to `pcre_fullinfo'
/home/cs/disk/x86/tmp/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/bin/../lib/gcc/x86_64-openwrt-linux-musl/8.4.0/../../../../x86_64-openwrt-linux-musl/bin/ld: regex.c:(.text+0x477): undefined reference to `pcre_fullinfo'
collect2: error: ld returned 1 exit status
make[4]: *** [Makefile:157: libselinux.so.1] Error 1
make[4]: Leaving directory '/home/cs/disk/x86/tmp/lede/build_dir/target-x86_64_musl/libselinux-3.2/src'
make[3]: *** [Makefile:48: all] Error 1
make[3]: Leaving directory '/home/cs/disk/x86/tmp/lede/build_dir/target-x86_64_musl/libselinux-3.2'
make[2]: *** [Makefile:157: /home/cs/disk/x86/tmp/lede/build_dir/target-x86_64_musl/libselinux-3.2/.built] Error 2
make[2]: Leaving directory '/home/cs/disk/x86/tmp/lede/package/libs/libselinux'
time: package/libs/libselinux/compile#0.32#0.14#0.42
    ERROR: package/libs/libselinux failed to build.
make[1]: *** [package/Makefile:116: package/libs/libselinux/compile] Error 1
make[1]: Leaving directory '/home/cs/disk/x86/tmp/lede'
make: *** [/home/cs/disk/x86/tmp/lede/include/toplevel.mk:230：package/libselinux/compile] 错误 2
cs@ubuntu:~/disk/x86/tmp/lede$ 

上一个问题ccache编译问题,ubuntu14.04无果，只好更换系统为ubuntu20.04.2
但是package/libselinux又报错了，无法解决求助

OleStore, 2021-06-23:
rm -rf

softtip, 2021-06-23:
> rm -rf

rm -rf lede 无数次了

softtip, 2021-06-23:
无解

hahaher123, 2021-06-24:
首页就写着推荐使用ubuntu18来编译，为啥不用14就用20。。。。

xnxy2012, 2021-06-24:
一样的错误，换了18依然存在，但是有时候会编译成功，不知道是什么问题

adamgonglj, 2021-06-24:
是不是磁盘空间不够啦？

adamgonglj, 2021-06-24:
我就出现过磁盘空间不够的情况。

EnnawYang, 2021-06-24:
> 是不是磁盘空间不够啦？

No

softtip, 2021-06-24:
搞定了，之前用多线程编译失败，后来再单线程也失败

最后，彻底rm -rf后单线程编译通过

closed 2021-06-24 by softtip