coolsnowwolf/lede #7154: luci-app-advanced-reboot界面报错

dunkeng, 2021-06-23:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ *] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：[image]

### (2) 路由器型号和固件版本

A：WRT1200AC，
OpenWrt R21.6.22 / LuCI Master (git-21.173.12022-bfe9018)

### (3) 详细日志

A：不知道完整日志在哪

4860575, 2021-06-23:
使用17.01分支的luci-app-advanced-reboot就可以

dunkeng, 2021-06-23:
> 使用17.01分支的luci-app-advanced-reboot就可以,也就是上一个版本42那个
> git-21.174.56281-edd4d24-42
> [image]

 谢谢
我是想让开发者们能修复一下问题，所以暂时不考虑回退。
你这个内核版本好旧啊，openwrt官方都5.4了。

closed 2021-06-24 by coolsnowwolf