coolsnowwolf/lede #7156: HiWiFi HC5861 最新源码PPPoE掉线

AndyLin2020, 2021-06-23:
### (1) 具体问题

A：PPPOE拔号上网，刷今日头条的视频就很容易掉线

### (2) 路由器型号和固件版本

A：HiWiFi HC5861，OpenWrt R21.6.22 / LuCI Master (git-21.173.12022-bfe9018)

### (3) 详细日志

A：
Wed Jun 23 18:42:46 2021 kern.err kernel: [ 1479.899588] ralink_soc_eth 10100000.ethernet eth0: transmit timed out
Wed Jun 23 18:42:46 2021 kern.info kernel: [ 1479.906196] ralink_soc_eth 10100000.ethernet eth0: dma_cfg:80000065
Wed Jun 23 18:42:46 2021 kern.info kernel: [ 1479.912628] ralink_soc_eth 10100000.ethernet eth0: tx_ring=0, base=066f4000, max=1024, ctx=874, dtx=874, fdx=823, next=874
Wed Jun 23 18:42:46 2021 kern.info kernel: [ 1479.923920] ralink_soc_eth 10100000.ethernet eth0: rx_ring=0, base=0409c000, max=1024, calc=801, drx=209
Wed Jun 23 18:42:46 2021 kern.info kernel: [ 1479.933686] ralink_soc_eth 10100000.ethernet eth0: port 5 link down
Wed Jun 23 18:42:46 2021 daemon.notice netifd: Network device 'eth0' link is down
Wed Jun 23 18:42:46 2021 daemon.notice netifd: VLAN 'eth0.1' link is down
Wed Jun 23 18:42:46 2021 daemon.notice netifd: VLAN 'eth0.2' link is down
Wed Jun 23 18:42:46 2021 daemon.notice netifd: Interface 'wan' has link connectivity loss
Wed Jun 23 18:42:46 2021 daemon.notice netifd: Interface 'wan6' has link connectivity loss
Wed Jun 23 18:42:46 2021 kern.info kernel: [ 1479.950989] br-lan: port 1(eth0.1) entered disabled state
Wed Jun 23 18:42:46 2021 daemon.info pppd[13540]: Terminating on signal 15
Wed Jun 23 18:42:46 2021 daemon.info pppd[13540]: Connect time 6.0 minutes.
Wed Jun 23 18:42:46 2021 daemon.info pppd[13540]: Sent 1954888 bytes, received 19797193 bytes.
Wed Jun 23 18:42:46 2021 daemon.err odhcp6c[13770]: Failed to send RS (Permission denied)
Wed Jun 23 18:42:46 2021 daemon.notice netifd: Network device 'pppoe-wan' link is down
Wed Jun 23 18:42:46 2021 daemon.notice netifd: Network alias 'pppoe-wan' link is down
Wed Jun 23 18:42:46 2021 daemon.notice netifd: Interface 'wan_6' has link connectivity loss
Wed Jun 23 18:42:46 2021 daemon.err odhcp6c[13770]: Failed to send RELEASE message to ff02::1:2 (Permission denied)
Wed Jun 23 18:42:46 2021 daemon.notice netifd: wan_6 (13770): Command failed: Permission denied
Wed Jun 23 18:42:46 2021 daemon.notice netifd: Interface 'wan6' is now down
Wed Jun 23 18:42:46 2021 daemon.err odhcp6c[13770]: Failed to send SOLICIT message to ff02::1:2 (Permission denied)
Wed Jun 23 18:42:46 2021 daemon.notice netifd: Interface 'wan_6' is now down
Wed Jun 23 18:42:46 2021 daemon.notice netifd: Interface 'wan_6' is disabled
Wed Jun 23 18:42:49 2021 daemon.notice netifd: Network device 'eth0' link is up
Wed Jun 23 18:42:49 2021 daemon.notice netifd: VLAN 'eth0.1' link is up
Wed Jun 23 18:42:49 2021 daemon.notice netifd: VLAN 'eth0.2' link is up
Wed Jun 23 18:42:49 2021 daemon.notice netifd: Interface 'wan' has link connectivity
Wed Jun 23 18:42:49 2021 daemon.notice netifd: Interface 'wan6' has link connectivity
Wed Jun 23 18:42:49 2021 daemon.notice netifd: Interface 'wan6' is setting up now
Wed Jun 23 18:42:49 2021 kern.info kernel: [ 1482.389974] br-lan: port 1(eth0.1) entered blocking state
Wed Jun 23 18:42:49 2021 kern.info kernel: [ 1482.395556] br-lan: port 1(eth0.1) entered forwarding state
Wed Jun 23 18:42:49 2021 kern.info kernel: [ 1482.544888] ralink_soc_eth 10100000.ethernet eth0: port 5 link up (1000Mbps/Full duplex)
Wed Jun 23 18:42:51 2021 daemon.notice netifd: Interface 'wan' is now down
Wed Jun 23 18:42:51 2021 daemon.notice netifd: Interface 'wan' is setting up now
Wed Jun 23 18:42:51 2021 daemon.err insmod: module is already loaded - slhc
Wed Jun 23 18:42:51 2021 daemon.err insmod: module is already loaded - ppp_generic
Wed Jun 23 18:42:51 2021 daemon.err insmod: module is already loaded - pppox
Wed Jun 23 18:42:51 2021 daemon.err insmod: module is already loaded - pppoe
Wed Jun 23 18:42:51 2021 daemon.notice netifd: wan (16868): uci: Entry not found
Wed Jun 23 18:42:51 2021 daemon.notice netifd: wan (16868): sh: out of range
Wed Jun 23 18:42:51 2021 daemon.info pppd[16906]: Plugin rp-pppoe.so loaded.
Wed Jun 23 18:42:51 2021 daemon.info pppd[16906]: RP-PPPoE plugin version 3.8p compiled against pppd 2.4.8
Wed Jun 23 18:42:51 2021 daemon.notice pppd[16906]: pppd 2.4.8 started by root, uid 0
Wed Jun 23 18:42:59 2021 kern.err kernel: [ 1487.622349] ralink_soc_eth 10100000.ethernet eth0: transmit timed out
Wed Jun 23 18:42:59 2021 kern.info kernel: [ 1487.628944] ralink_soc_eth 10100000.ethernet eth0: dma_cfg:80000065
Wed Jun 23 18:42:59 2021 kern.info kernel: [ 1487.635373] ralink_soc_eth 10100000.ethernet eth0: tx_ring=0, base=06700000, max=1024, ctx=386, dtx=386, fdx=32, next=386
Wed Jun 23 18:42:59 2021 kern.info kernel: [ 1487.646569] ralink_soc_eth 10100000.ethernet eth0: rx_ring=0, base=04000000, max=1024, calc=13, drx=14
Wed Jun 23 18:42:59 2021 kern.info kernel: [ 1487.656608] ralink_soc_eth 10100000.ethernet eth0: port 5 link down
Wed Jun 23 18:42:59 2021 daemon.notice netifd: Network device 'eth0' link is down
Wed Jun 23 18:42:59 2021 daemon.notice netifd: VLAN 'eth0.1' link is down
Wed Jun 23 18:42:59 2021 daemon.notice netifd: VLAN 'eth0.2' link is down
Wed Jun 23 18:42:59 2021 daemon.notice netifd: Interface 'wan' has link connectivity loss
Wed Jun 23 18:42:59 2021 daemon.notice netifd: Interface 'wan6' has link connectivity loss
Wed Jun 23 18:42:59 2021 kern.info kernel: [ 1487.669470] br-lan: port 1(eth0.1) entered disabled state
Wed Jun 23 18:42:59 2021 daemon.notice netifd: Interface 'wan6' is now down
Wed Jun 23 18:42:59 2021 daemon.err pppd[16906]: select (waitForPADO): Interrupted system call
Wed Jun 23 18:42:59 2021 daemon.warn pppd[16906]: Timeout waiting for PADO packets
Wed Jun 23 18:42:59 2021 daemon.err pppd[16906]: Unable to complete PPPoE Discovery
[... 44 lines cut ...]
Wed Jun 23 18:43:54 2021 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Wed Jun 23 18:43:55 2021 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Wed Jun 23 18:43:59 2021 daemon.info dnsmasq[18758]: Connected to system UBus
Wed Jun 23 18:44:35 2021 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Wed Jun 23 18:44:35 2021 user.notice dnsmasq: Allowing 127.0.0.0/8 responses

内核日志：
[ 1479.899588] ralink_soc_eth 10100000.ethernet eth0: transmit timed out
[ 1479.906196] ralink_soc_eth 10100000.ethernet eth0: dma_cfg:80000065
[ 1479.912628] ralink_soc_eth 10100000.ethernet eth0: tx_ring=0, base=066f4000, max=1024, ctx=874, dtx=874, fdx=823, next=874
[ 1479.923920] ralink_soc_eth 10100000.ethernet eth0: rx_ring=0, base=0409c000, max=1024, calc=801, drx=209
[ 1479.933686] ralink_soc_eth 10100000.ethernet eth0: port 5 link down
[ 1479.950989] br-lan: port 1(eth0.1) entered disabled state
[ 1482.389974] br-lan: port 1(eth0.1) entered blocking state
[ 1482.395556] br-lan: port 1(eth0.1) entered forwarding state
[ 1482.544888] ralink_soc_eth 10100000.ethernet eth0: port 5 link up (1000Mbps/Full duplex)
[ 1487.622349] ralink_soc_eth 10100000.ethernet eth0: transmit timed out
[ 1487.628944] ralink_soc_eth 10100000.ethernet eth0: dma_cfg:80000065
[ 1487.635373] ralink_soc_eth 10100000.ethernet eth0: tx_ring=0, base=06700000, max=1024, ctx=386, dtx=386, fdx=32, next=386
[ 1487.646569] ralink_soc_eth 10100000.ethernet eth0: rx_ring=0, base=04000000, max=1024, calc=13, drx=14
[ 1487.656608] ralink_soc_eth 10100000.ethernet eth0: port 5 link down
[ 1487.669470] br-lan: port 1(eth0.1) entered disabled state
[ 1490.450556] br-lan: port 1(eth0.1) entered blocking state
[ 1490.456110] br-lan: port 1(eth0.1) entered forwarding state
[ 1490.616428] ralink_soc_eth 10100000.ethernet eth0: port 5 link up (1000Mbps/Full duplex)

FisherWY, 2021-06-25:
同样的问题

1. 具体问题
- 校园网，使用Mentohust认证，隔段时间就会出现掉线的情况
- 大概2个星期前编译的固件没有出现这种问题

2. 路由器型号和固件版本
- HiWiFi HC5861
- OpenWrt R21.6.22 / LuCI Master (git-21.173.12022-bfe9018)

3. 掉线时的内核日志
```
[ 6749.020164] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
[ 6749.025777] mt76x2e 0000:01:00.0: Build: 1
[ 6749.030033] mt76x2e 0000:01:00.0: Build Time: 201507311614____
[ 6749.060374] mt76x2e 0000:01:00.0: Firmware running!
[ 6749.069440] ieee80211 phy0: Hardware restart was requested
[ 6750.548664] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
[ 6750.554322] mt76x2e 0000:01:00.0: Build: 1
[ 6750.558550] mt76x2e 0000:01:00.0: Build Time: 201507311614____
[ 6750.581052] mt76x2e 0000:01:00.0: Firmware running!
[ 6750.586834] ieee80211 phy0: Hardware restart was requested
[ 6752.295816] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
[ 6752.301429] mt76x2e 0000:01:00.0: Build: 1
[ 6752.305703] mt76x2e 0000:01:00.0: Build Time: 201507311614____
[ 6752.334572] mt76x2e 0000:01:00.0: Firmware running!
[ 6752.340393] ieee80211 phy0: Hardware restart was requested
[ 6752.409462] ralink_soc_eth 10100000.ethernet eth0: transmit timed out
[ 6752.416102] ralink_soc_eth 10100000.ethernet eth0: dma_cfg:80000065
[ 6752.422528] ralink_soc_eth 10100000.ethernet eth0: tx_ring=0, base=060d0000, max=1024, ctx=77, dtx=77, fdx=48, next=77
[ 6752.433468] ralink_soc_eth 10100000.ethernet eth0: rx_ring=0, base=06178000, max=1024, calc=164, drx=166
[ 6752.521070] ralink_soc_eth 10100000.ethernet eth0: port 5 link down
[ 6753.480153] ralink_soc_eth 10100000.ethernet eth0: port 5 link up (100Mbps/Full duplex)
[ 6757.009966] ralink_soc_eth 10100000.ethernet eth0: transmit timed out
[ 6757.016562] ralink_soc_eth 10100000.ethernet eth0: dma_cfg:80000065
[ 6757.023011] ralink_soc_eth 10100000.ethernet eth0: tx_ring=0, base=060d0000, max=1024, ctx=58, dtx=58, fdx=16, next=58
[ 6757.033939] ralink_soc_eth 10100000.ethernet eth0: rx_ring=0, base=06b58000, max=1024, calc=132, drx=133
[ 6757.044258] ralink_soc_eth 10100000.ethernet eth0: port 5 link down
[ 6758.988911] ralink_soc_eth 10100000.ethernet eth0: port 5 link up (100Mbps/Full duplex)
```

4. 完整的日志如下
[HC5861_kernel_log_full](https://github.com/coolsnowwolf/lede/files/6715828/hc5861_log_full.txt)

### 感谢开发者们的辛勤付出❤️

AndyLin2020, 2021-06-28:
还未修复，一刷视频就掉线

AndyLin2020, 2021-06-29:
> 同样的问题
> 
> 1. 具体问题
> 
> * 校园网，使用Mentohust认证，隔段时间就会出现掉线的情况
> * 大概2个星期前编译的固件没有出现这种问题
> 
> 1. 路由器型号和固件版本
> 
> * HiWiFi HC5861
> * OpenWrt R21.6.22 / LuCI Master (git-21.173.12022-bfe9018)
> 
> 1. 掉线时的内核日志
> 
> ```
> [ 6749.020164] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
> [ 6749.025777] mt76x2e 0000:01:00.0: Build: 1
> [ 6749.030033] mt76x2e 0000:01:00.0: Build Time: 201507311614____
> [ 6749.060374] mt76x2e 0000:01:00.0: Firmware running!
> [ 6749.069440] ieee80211 phy0: Hardware restart was requested
> [ 6750.548664] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
> [ 6750.554322] mt76x2e 0000:01:00.0: Build: 1
> [ 6750.558550] mt76x2e 0000:01:00.0: Build Time: 201507311614____
> [ 6750.581052] mt76x2e 0000:01:00.0: Firmware running!
> [ 6750.586834] ieee80211 phy0: Hardware restart was requested
> [ 6752.295816] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
> [ 6752.301429] mt76x2e 0000:01:00.0: Build: 1
> [ 6752.305703] mt76x2e 0000:01:00.0: Build Time: 201507311614____
> [ 6752.334572] mt76x2e 0000:01:00.0: Firmware running!
> [ 6752.340393] ieee80211 phy0: Hardware restart was requested
> [ 6752.409462] ralink_soc_eth 10100000.ethernet eth0: transmit timed out
> [ 6752.416102] ralink_soc_eth 10100000.ethernet eth0: dma_cfg:80000065
> [ 6752.422528] ralink_soc_eth 10100000.ethernet eth0: tx_ring=0, base=060d0000, max=1024, ctx=77, dtx=77, fdx=48, next=77
> [ 6752.433468] ralink_soc_eth 10100000.ethernet eth0: rx_ring=0, base=06178000, max=1024, calc=164, drx=166
> [ 6752.521070] ralink_soc_eth 10100000.ethernet eth0: port 5 link down
> [ 6753.480153] ralink_soc_eth 10100000.ethernet eth0: port 5 link up (100Mbps/Full duplex)
> [ 6757.009966] ralink_soc_eth 10100000.ethernet eth0: transmit timed out
> [ 6757.016562] ralink_soc_eth 10100000.ethernet eth0: dma_cfg:80000065
> [ 6757.023011] ralink_soc_eth 10100000.ethernet eth0: tx_ring=0, base=060d0000, max=1024, ctx=58, dtx=58, fdx=16, next=58
> [ 6757.033939] ralink_soc_eth 10100000.ethernet eth0: rx_ring=0, base=06b58000, max=1024, calc=132, drx=133
> [ 6757.044258] ralink_soc_eth 10100000.ethernet eth0: port 5 link down
> [ 6758.988911] ralink_soc_eth 10100000.ethernet eth0: port 5 link up (100Mbps/Full duplex)
> ```
> 
> 1. 完整的日志如下
>    [HC5861_kernel_log_full](https://github.com/coolsnowwolf/lede/files/6715828/hc5861_log_full.txt)
> 
> ### 感谢开发者们的辛勤付出

焦急啊，掉线太频繁了，得手动重连才行

AndyLin2020, 2021-06-29:
L大大可以帮忙看一下这个问是吗？ @coolsnowwolf

coolsnowwolf, 2021-06-30:
没有 MT7620 的设备了

AmadeusGhost, 2021-07-02:
似乎是 mt76 (mt7612) 不稳定引起的

AndyLin2020, 2021-08-11:
这个版本正常：
    固件版本: OpenWrt R21.7.26 / LuCI Master (git-21.206.25982-60d43ca)
    内核版本: 5.4.132

这几天更新后又是同样问题了

pzcn, 2021-08-19:
同问题 k2 就算什么也不干每隔一段时间自动断开
ralink_soc_eth 10100000.ethernet eth0: dma_cfg:80000065

AndyLin2020, 2021-08-20:
试了最新的5.10和5.4内核都有一样问题，现在只有用5.4.132正常

`固件版本	OpenWrt R21.7.15 / LuCI Master (git-21.211.30217-03784c3)
内核版本	5.4.132`

closed 2024-11-19 by coolsnowwolf