coolsnowwolf/lede #7157: Dnspod 出错

leone424, 2021-06-23:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
DDNS 里 Dnspod 出错
Q：是否搜索了issue (使用 "x" 选择)
[ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：OpenWrt R21.6.22 / LuCI Master (git-21.173.12022-bfe9018)

### (3) 详细日志

A：
[image]

wspzf, 2021-06-23:
你不是一个人，我也一样

swxk521, 2021-06-23:
看看有没有安装wget-ssl组件
没有的话就安装上或编译时加上。
或者等大佬改下DDNS脚本。

wspzf, 2021-06-23:
> 看看有没有安装 wget-ssl 组件
> 没有的话就安装上或编译时加上。
> 或者等大佬改下 DDNS 脚本。
[image]
编译加上这个就可以了

leone424, 2021-06-23:
> > 看看有没有安装 wget-ssl 组件
> > 没有的话就安装上或编译时加上。
> > 或者等大佬改下 DDNS 脚本。
> 
> [image]
> 
> 编译加上这个就可以了

按照此方法已解决，感谢！

closed 2021-06-23 by leone424