coolsnowwolf/lede #7158: LEDE docker Storage Driver

kingzleshe, 2021-06-23:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：容器创建，启动，升级，删除等操作太慢，太痛苦

docker 文档 关于 vfs storage driver 说明：

The vfs storage driver is intended for testing purposes, and for situations where no copy-on-write filesystem can be used. Performance of this storage driver is poor, and is not generally recommended for production use.

目前LEDE默认使用的vfs格式

参考下面的链接，修改为devicemapper发现并不生效
https://www.raspberrypi.org/forums/viewtopic.php?t=275070
https://docs.docker.com/storage/storagedriver/device-mapper-driver/

是LEDE系统目前不支持么？

### (2) 路由器型号和固件版本

A：树莓派4  shulingGG源码编译最新版本

### (3) 详细日志

A：

gztingting, 2021-06-23:
确实，最近新编译的LEDE固件，用Docker时比旧版本的固件+docker_ce慢很多，一直找不到原因。

kingzleshe, 2021-06-23:
网上查了下 SquashFS 文件系统是支持  copy-on-write 特性的
vfs 的特点是慢，但是可以使用在 no copy-on-write 的文件系统上面

https://docs.docker.com/storage/storagedriver/overlayfs-driver/
docker 推荐的OverlayFS storage driver中描述的前提条件并没感觉不支持
至于是否其他问题导致，能力有限，我就不知道了。

shu0802, 2021-06-23:
手动更新这两个文件版本试试
luci-app-dockerman_v0.5.21_all
luci-lib-docker_v0.3.4_all

kingzleshe, 2021-06-24:
> 手动更新这两个文件版本试试
> luci-app-dockerman_v0.5.21_all
> luci-lib-docker_v0.3.4_all

升级好了，OverlayFS storage driver试了 overlay和 overlay2还是不行。
不过升级后界面信息更好看和实用了，菜单都独立出来了，也算是以外的惊喜。

kingzleshe, 2021-06-24:
重新安装了OpenWrt-Mini
https://github.com/SuLingGG/OpenWrt-Mini

然后安装docker 20.10.7 版本
现在显示是 Storage Driver: overlay2  正常了

closed 2021-06-24 by kingzleshe