coolsnowwolf/lede #7162: 请教各位大神关于ipv6的问题

rongcenter, 2021-06-23:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

我一般不勾选ipv6编译，21日之前的版本没有问题，编译出来不会分配ipv6地址，22日之后的版本编译后自动分配ipv6地址，古怪的是wan6口没有获取到地址，lan口下面的设备全分到240开头的电信地址，这有什么办法解决？

### (2) 路由器型号和固件版本

x86

### (3) 详细日志

无

rongcenter, 2021-06-23:
如有大神解惑，感激涕零

shu0802, 2021-06-23:
LAN-高级设置-使用内置的ipv6后的钩掉
WAN-高级设置-使用内置的ipv6后的钩掉、获取ipv6地址选择禁用。

hahaher123, 2021-06-24:
dnsmasq-full默认增加了DHCP V6的选项，编译前去掉就行

xcf13363175, 2021-06-24:
https://github.com/coolsnowwolf/lede/commit/07863551c8845ddbb5739e213e1a69ebb373adf5#diff-f0faed86d3aea4e7c4424e924fb504b26e190f0adf7c4e7dca473ca0cef212ad
默认加了

coolsnowwolf, 2021-06-24:
fixed

gztingting, 2021-06-24:
把光猫桥接改为仅ipv4就不会获取到240*开头的v6地址。openwrt关了v6，但光猫桥接选项还是v4/v6的话，虽然路由器不再下发v6地址，它自身还是获取到有效的v6地址的。

closed 2021-06-24 by coolsnowwolf