coolsnowwolf/lede #7163: 硬盘分区不能自动挂载，如何调整？

gztingting, 2021-06-24:
**版本信息：X86-64 OpenWrt R21.6.22 / LuCI Master (git-21.173.12022-bfe9018) Kernel 5.4.43**

把img刷到u盘上启动Thinkpad x250笔记本电脑，image选的是squashfs文件系统，第三个分区是空分区，格成了ext4，另还有固态m2 ssd和一个sata机械硬盘，除了原windows系统的EFI分区以外，都是ntfs分区。

**发现两个问题：**
1、电脑重启后，U盘、固态硬盘、机械硬盘分配的dev/sda、dev/sdb、dev/sdc顺序经常会变。U盘，一会是sda，一会是sdc。如何锁死？
2、不管了ext4分区还是ntfs分区，不能自动挂载。

在磁盘管理页面，可以看到：
[image]

挂载点列表里，显示多个设备**不存在**，但实际这里的**uuid都是正确的**，
开始怀疑是保留配置刷写新的image导致的（实际上uuid是不会变的），
后来把U盘分区全删了，重新把新编译的image写盘，**全新环境下测试，问题依旧。**

通过手动添加或者自动挂载的方式，都显示多个设备不存在，

图中所显示文件系统为ntfs3、选型为rw,relatime,nls=utf8，都是**手动输入**，否则就是一个问号？，手动添加和自动挂载方式都需要修改文件系统参数和选型参数。
[image]

经过这么修改，看到对应挂载到/mnt/sda*了，**但依然是无法读取文件。**
TTYD终端里用lsblk看到没有挂载成功
[image]

一定要在TTYD终端里，输入如下命令（加上相应a3，b1、c3等），才算正确挂载。**_但重启后就失效了。_**
[image]

在Diskman磁盘管理那里，手动添加和在ttyd终端里输入mount效果是一样的，**_重启后就失效了，_**还需要手动添加。
[image]

**请教各路大神，以上问题如何正确设置方能保证顺序不变，挂载点不变，自动挂载**

xfmly, 2021-06-24:
启动项 加 mount -t ntfs3 /dev/sda1 /mnt/sda1就好了

# Put your custom commands here that should be executed once
# the system init finished. By default this file does nothing.

mount -t ntfs3 /dev/sda1 /mnt/sda1

exit 0

coolsnowwolf, 2021-06-24:
ntfs3 还没有正式进入内核的 fstools 支持，暂时需要手动命令

gztingting, 2021-06-24:
> ntfs3 还没有正式进入内核的 fstools 支持，暂时需要手动命令

有一个ext4分区也需要手动mount呢？

closed 2021-06-24 by gztingting

rxhdiy123, 2021-06-28:
> 启动项 加 mount -t ntfs3 /dev/sda1 /mnt/sda1就好了
> 
> # Put your custom commands here that should be executed once
> # the system init finished. By default this file does nothing.
> mount -t ntfs3 /dev/sda1 /mnt/sda1
> 
> exit 0

我的手动命令也不行