coolsnowwolf/lede #7171: 大大，新的源码怎么没有tplink wdr6500 v2这个机型了？

jacelift, 2021-06-24:
大大，新的源码怎么没有tplink wdr6500 v2这个机型了？

zpj7377327, 2021-06-24:
换个源码就行了，tp类很多都被移除了

jacelift, 2021-06-24:
> 换个源码就行了，tp类很多都被移除了

换哪个源码？lienol的源码也是没有这个型号的，lean的源码之前是有tp这个型号的，最近更新后就没有了！！

zpj7377327, 2021-06-24:
> > 换个源码就行了，tp类很多都被移除了
> 
> 换哪个源码？lienol的源码也是没有这个型号的，lean的源码之前是有tp这个型号的，最近更新后就没有了！！

换成coolsnowwolf/openwrt，lede里面这个了

jacelift, 2021-06-24:
> > > 换个源码就行了，tp类很多都被移除了
> > 
> > 换哪个源码？lienol的源码也是没有这个型号的，lean的源码之前是有tp这个型号的，最近更新后就没有了！！
> 
> 换成coolsnowwolf/openwrt，lede里面这个了

这个是17.0版本的啊，之前有wdr6500的是21.0版本的，感觉时光倒流了(●—●)

coolsnowwolf, 2021-06-25:
同步了官方的代码，tplink 4MB rom 系列都已经被删除了

yjd, 2021-06-26:
选比较旧的版本编译

coolsnowwolf, 2021-06-29:
[image]

特别优待，重新添加了这个型号的支持，在 ATH79 里面

jacelift, 2021-06-30:
> [image]
> 
> 特别优待，重新添加了这个型号的支持，在 ATH79 里面

给力，↖(^ω^)↗

closed 2021-08-15 by AmadeusGhost

yqxuezhe, 2024-12-30:
请问TPlink-WDR6500 千兆版怎么刷