coolsnowwolf/lede #7175: 编译报错

airline12138, 2021-06-24:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ √ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译时不错

### (2) 路由器型号和固件版本

A：Ubuntu18.04 LTS 64bit

### (3) 详细日志

A：install: cannot change permissions of '/home/ubuntu/lede/staging_dir/host/bin': 
Operation not permitted                                                         
Makefile:39: recipe for target '/home/ubuntu/lede/staging_dir/host/stamp/.ninja_
installed' failed                                                               
make[3]: *** [/home/ubuntu/lede/staging_dir/host/stamp/.ninja_installed] Error 1
make[3]: Leaving directory '/home/ubuntu/lede/tools/ninja'                      
time: tools/ninja/compile#0.03#0.01#0.05                                        
    ERROR: tools/ninja failed to build.                                         
tools/Makefile:157: recipe for target 'tools/ninja/compile' failed              
make[2]: *** [tools/ninja/compile] Error 1                                      
make[2]: Leaving directory '/home/ubuntu/lede'                                  
tools/Makefile:155: recipe for target '/home/ubuntu/lede/staging_dir/host/stamp/
.tools_compile_yyynyynnyyyyynyyyyyynyynyyyynyyyyyyyyyyyyyyyyynynnyyyyyyyyyy' fai
led                                                                             
make[1]: *** [/home/ubuntu/lede/staging_dir/host/stamp/.tools_compile_yyynyynnyy
yyynyyyyyynyynyyyynyyyyyyyyyyyyyyyyynynnyyyyyyyyyy] Error 2                     
make[1]: Leaving directory '/home/ubuntu/lede'                                  
/home/ubuntu/lede/include/toplevel.mk:228: recipe for target 'world' failed     
make: *** [world] Error 2

zhouxiaolong920, 2021-06-25:
Operation not permitted
一看就是权限问题，是不是在root下打开过。然后在换到ubuntu帐户下，在弄的。

closed 2021-08-12 by AmadeusGhost

dd-bpir3, 2024-07-11:
> Operation not permitted 一看就是权限问题，是不是在root下打开过。然后在换到ubuntu帐户下，在弄的。

我也是这问题，我是vps操作的，也是普通用户