coolsnowwolf/lede #7176: 最新版本编译docker 失败或者依赖golang的都会失败

zhouxiaolong920, 2021-06-24:
/home/op/lede/staging_dir/target-x86_64_musl/usr/share/gocode/src does not exist, skipping symlinks

touch /home/op/lede/build_dir/target-x86_64_musl/runc-1.0.0-rc95/.configured_1cd727ac7b9ebbe4ee453b5ddb8504b3
rm -f /home/op/lede/build_dir/target-x86_64_musl/runc-1.0.0-rc95/.built
touch /home/op/lede/build_dir/target-x86_64_musl/runc-1.0.0-rc95/.built_check
CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/op/lede/build_dir/target-x86_64_musl/runc-1.0.0-rc95=runc-1.0.0-rc95 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/op/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/include -I/home/op/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/fortify -I/home/op/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include " CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/op/lede/build_dir/target-x86_64_musl/runc-1.0.0-rc95=runc-1.0.0-rc95 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/op/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/include -I/home/op/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/fortify -I/home/op/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include " LDFLAGS="-L/home/op/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/lib -L/home/op/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib -znow -zrelro " GOOS="linux" GOARCH="amd64" GO386="" GOARM="" GOMIPS="" GOMIPS64="" CGO_ENABLED=1 CC="ccache_cc" CXX="ccache_cxx" CGO_CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/op/lede/build_dir/target-x86_64_musl/runc-1.0.0-rc95=runc-1.0.0-rc95 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CGO_CPPFLAGS="-I/home/op/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/include -I/home/op/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/fortify -I/home/op/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include" CGO_CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/op/lede/build_dir/target-x86_64_musl/runc-1.0.0-rc95=runc-1.0.0-rc95 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CGO_LDFLAGS="-L/home/op/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/lib -L/home/op/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib -znow -zrelro" GOPATH="/home/op/lede/build_dir/target-x86_64_musl/runc-1.0.0-rc95/.go_work/build" GOCACHE="/home/op/lede/tmp/go-build" GOMODCACHE="/home/op/lede/dl/go-mod-cache" GOENV=off make  -C /home/op/lede/build_dir/target-x86_64_musl/runc-1.0.0-rc95/.go_work/build/src/github.com/opencontainers/runc AR="x86_64-openwrt-linux-musl-gcc-ar" AS="ccache_cc -c -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/op/lede/build_dir/target-x86_64_musl/runc-1.0.0-rc95=runc-1.0.0-rc95 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=x86_64-openwrt-linux-musl-ld NM="x86_64-openwrt-linux-musl-gcc-nm" CC="ccache_cc" GCC="ccache_cc" CXX="ccache_cxx" RANLIB="x86_64-openwrt-linux-musl-gcc-ranlib" STRIP=x86_64-openwrt-linux-musl-strip OBJCOPY=x86_64-openwrt-linux-musl-objcopy OBJDUMP=x86_64-openwrt-linux-musl-objdump SIZE=x86_64-openwrt-linux-musl-size CROSS="x86_64-openwrt-linux-musl-" ARCH="x86_64" COMMIT=b9ee9c6314599f1b4a7f497e1f1f856fe433d3b7 BUILDTAGS=' seccomp' ;
make[4]: Entering directory '/home/op/lede/build_dir/target-x86_64_musl/runc-1.0.0-rc95/.go_work/build/src/github.com/opencontainers/runc'
go build -trimpath "-mod=vendor" "-buildmode=pie"  -tags "seccomp" -ldflags "-X main.gitCommit=b9ee9c6314599f1b4a7f497e1f1f856fe433d3b7 -X main.version=1.0.0-rc95 " -o runc .
# github.com/opencontainers/runc
/home/op/lede/staging_dir/hostpkg/lib/go-cross/pkg/tool/linux_amd64/link: running ccache_cc failed: signal: aborted
ccache: Util.cpp:731: std::string Util::get_relative_path(nonstd::sv_lite::string_view, nonstd::sv_lite::string_view): failed assertion: Util::is_absolute_path(dir)

Makefile:36: recipe for target 'runc' failed
make[4]: *** [runc] Error 2
make[4]: Leaving directory '/home/op/lede/build_dir/target-x86_64_musl/runc-1.0.0-rc95/.go_work/build/src/github.com/opencontainers/runc'
Makefile:64: recipe for target '/home/op/lede/build_dir/target-x86_64_musl/runc-1.0.0-rc95/.built' failed
make[3]: *** [/home/op/lede/build_dir/target-x86_64_musl/runc-1.0.0-rc95/.built] Error 2
make[3]: Leaving directory '/home/op/lede/feeds/packages/utils/runc'
time: package/feeds/packages/runc/compile#40.38#9.82#31.72
    ERROR: package/feeds/packages/runc failed to build.
package/Makefile:114: recipe for target 'package/feeds/packages/runc/compile' failed
make[2]: *** [package/feeds/packages/runc/compile] Error 1
make[2]: Leaving directory '/home/op/lede'
package/Makefile:110: recipe for target '/home/op/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/op/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/op/lede'
/home/op/lede/include/toplevel.mk:228: recipe for target 'world' failed
make: *** [world] Error 2
[op@jiuweihu1.local lede]$ git log
commit a2c991e755897d7fb22d53749c5db2e4861c9e41
Author: Dingcon <gztingting@gmail.com>
Date:   Thu Jun 24 17:37:57 2021 +0800

zhouxiaolong920, 2021-06-24:
我是centos7,
每次，更新了。。我都追，最新编译x86
这次是真过不了。全部删除了，所有的都重新来的

不安装docker和openclash之类。依赖于golang的编译，通过

lexin8, 2021-06-25:
不存在你说的问题，而且只有你一个人反馈，找找自己的原因。追求最新源码用云编译，你可以到我仓库去拉取。

zhouxiaolong920, 2021-06-25:
@lexin8 ok，已经在测试了。

zhouxiaolong920, 2021-06-26:
@lexin8 ，我一直用公司的虚拟机centos7。编译。
恩。我测试过了。centos7的会有这个问题，表现就是runc这里编译不过。具体原因还没有找到。
ubuntu编译没问题，很轻松过

closed 2021-06-30 by coolsnowwolf

codemenworld, 2023-08-17:
ubuntu 20.04 编译docker相关的，同样有这个问题