coolsnowwolf/lede #7183: 编译报错

zhunian0322, 2021-06-25:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
下载源码后，更新安装 feeds 后，编译（想编译x86的 看了menuconfig 后选用默认的配置），系统是 Ubuntu1804LTS

### (1) 具体问题

A：编译到 fullconenat 出现下载失败，手动添加网上的 fullconenat-2019-10-21-0cf3b48f.tar.xz（有校验过该文件的 sha256 值 跟 package 下的 Makefile 中的是一致） 后依旧报错。

### (3) 详细日志

A：make[3]: Entering directory '/home/gino/lede/package/lean/openwrt-fullconenat'
mkdir -p /home/gino/lede/dl
SHELL= flock /home/gino/lede/tmp/.fullconenat-2019-10-21-0cf3b48f.tar.xz.flock -c '     /home/gino/lede/scripts/download.pl "/home/gino/lede/dl" "fullconenat-2019-10-21-0cf3b48f.tar.xz" "e7bb77d9916d190b3c02975faf62d1794557d4a4a774cd0cf1f3a6ddb7403e3e" "" || (  /home/gino/lede/scripts/dl_github_archive.py --dl-dir="/home/gino/lede/dl" --url="https://github.com/Chion82/netfilter-full-cone-nat.git" --version="0cf3b48fd7d2fa81d0297d1fff12bbd0580fc435" --subdir="fullconenat-2019-10-21-0cf3b48f" --source="fullconenat-2019-10-21-0cf3b48f.tar.xz" --hash="e7bb77d9916d190b3c02975faf62d1794557d4a4a774cd0cf1f3a6ddb7403e3e" || (        echo "Checking out files from the git repository..."; mkdir -p /home/gino/lede/tmp/dl && cd /home/gino/lede/tmp/dl && rm -rf fullconenat-2019-10-21-0cf3b48f && [ \! -d fullconenat-2019-10-21-0cf3b48f ] && git clone  https://github.com/Chion82/netfilter-full-cone-nat.git fullconenat-2019-10-21-0cf3b48f && (cd fullconenat-2019-10-21-0cf3b48f && git checkout 0cf3b48fd7d2fa81d0297d1fff12bbd0580fc435 && git submodule update --init --recursive) && echo "Packing checkout..." && export TAR_TIMESTAMP=`cd fullconenat-2019-10-21-0cf3b48f && git log -1 --format='\''@%ct'\''` && rm -rf fullconenat-2019-10-21-0cf3b48f/.git &&         tar --numeric-owner --owner=0 --group=0 --mode=a-s --sort=name ${TAR_TIMESTAMP:+--mtime="$TAR_TIMESTAMP"} -c fullconenat-2019-10-21-0cf3b48f |      xz -zc -7e > /home/gino/lede/tmp/dl/fullconenat-2019-10-21-0cf3b48f.tar.xz && mv /home/gino/lede/tmp/dl/fullconenat-2019-10-21-0cf3b48f.tar.xz /home/gino/lede/dl/ && rm -rf fullconenat-2019-10-21-0cf3b48f; );  )    '
rm -f /home/gino/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2019-10-21-0cf3b48f/.built
touch /home/gino/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2019-10-21-0cf3b48f/.built_check
make -j1 -C "/home/gino/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.44" CROSS_COMPILE="x86_64-openwrt-linux-musl-" ARCH="x86" M="/home/gino/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2019-10-21-0cf3b48f" EXTRA_CFLAGS="" modules
make[4]: Entering directory '/home/gino/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.44'
make[5]: *** No rule to make target '/home/gino/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2019-10-21-0cf3b48f/xt_FULLCONENAT.o', needed by '/home/gino/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2019-10-21-0cf3b48f/xt_FULLCONENAT.mod'.  Stop.
Makefile:1821: recipe for target '/home/gino/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2019-10-21-0cf3b48f' failed
make[4]: *** [/home/gino/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2019-10-21-0cf3b48f] Error 2
make[4]: Leaving directory '/home/gino/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.44'
Makefile:65: recipe for target '/home/gino/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2019-10-21-0cf3b48f/.built' failed
make[3]: *** [/home/gino/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2019-10-21-0cf3b48f/.built] Error 2
make[3]: Leaving directory '/home/gino/lede/package/lean/openwrt-fullconenat'
time: package/lean/openwrt-fullconenat/compile#0.62#0.38#0.99
    ERROR: package/lean/openwrt-fullconenat failed to build.
package/Makefile:114: recipe for target 'package/lean/openwrt-fullconenat/compile' failed
make[2]: *** [package/lean/openwrt-fullconenat/compile] Error 1
make[2]: Leaving directory '/home/gino/lede'
package/Makefile:110: recipe for target '/home/gino/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/gino/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/gino/lede'
/home/gino/lede/include/toplevel.mk:228: recipe for target 'world' failed
make: *** [world] Error 2

coolsnowwolf, 2021-06-25:
make clean

closed 2021-06-30 by coolsnowwolf

zhangguanzhang, 2021-08-13:
我在action里编译也报这个错了，怎么解决的

netlovehf, 2022-01-12:
一样的错误
make[3]: Entering directory '/home/netlove/lede/package/lean/openwrt-fullconenat'
mkdir -p /home/netlove/lede/dl
SHELL= flock /home/netlove/lede/tmp/.fullconenat-2021-12-04-7fe02604.tar.xz.flock -c '  	/home/netlove/lede/scripts/download.pl "/home/netlove/lede/dl" "fullconenat-2021-12-04-7fe02604.tar.xz" "857062e07606d00ef9ce1b6c5d457ffc1ccfb7667694c1958bc59f1d7c4f08a4" "" || (  /home/netlove/lede/scripts/dl_github_archive.py --dl-dir="/home/netlove/lede/dl" --url="https://github.com/llccd/netfilter-full-cone-nat.git" --version="7fe02604e073945a3b081c18efbb4d3c799a31f6" --subdir="fullconenat-2021-12-04-7fe02604" --source="fullconenat-2021-12-04-7fe02604.tar.xz" --hash="857062e07606d00ef9ce1b6c5d457ffc1ccfb7667694c1958bc59f1d7c4f08a4" || ( 	echo "Checking out files from the git repository..."; mkdir -p /home/netlove/lede/tmp/dl && cd /home/netlove/lede/tmp/dl && rm -rf fullconenat-2021-12-04-7fe02604 && [ \! -d fullconenat-2021-12-04-7fe02604 ] && git clone  https://github.com/llccd/netfilter-full-cone-nat.git fullconenat-2021-12-04-7fe02604 && (cd fullconenat-2021-12-04-7fe02604 && git checkout 7fe02604e073945a3b081c18efbb4d3c799a31f6 && git submodule update --init --recursive) && echo "Packing checkout..." && export TAR_TIMESTAMP=`cd fullconenat-2021-12-04-7fe02604 && git log -1 --format='\''@%ct'\''` && rm -rf fullconenat-2021-12-04-7fe02604/.git && 	tar --numeric-owner --owner=0 --group=0 --mode=a-s --sort=name ${TAR_TIMESTAMP:+--mtime="$TAR_TIMESTAMP"} -c fullconenat-2021-12-04-7fe02604 | 	xz -zc -7e > /home/netlove/lede/tmp/dl/fullconenat-2021-12-04-7fe02604.tar.xz && mv /home/netlove/lede/tmp/dl/fullconenat-2021-12-04-7fe02604.tar.xz /home/netlove/lede/dl/ && rm -rf fullconenat-2021-12-04-7fe02604; );  )    '
touch /home/netlove/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/.prepared_42645540f7ac912202af65068e987065_6664517399ebbbc92a37c5bb081b5c53_check
. /home/netlove/lede/include/shell.sh; xzcat /home/netlove/lede/dl/fullconenat-2021-12-04-7fe02604.tar.xz | tar -C /home/netlove/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/.. -xf -
[ ! -d ./src/ ] || cp -fpR ./src/. /home/netlove/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604
cp -fpR ./files/Makefile /home/netlove/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/
touch /home/netlove/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/.prepared_42645540f7ac912202af65068e987065_6664517399ebbbc92a37c5bb081b5c53
rm -f /home/netlove/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/.configured_*
rm -f /home/netlove/lede/staging_dir/target-x86_64_musl/stamp/.openwrt-fullconenat_installed
(cd /home/netlove/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/./; if [ -x ./configure ]; then find /home/netlove/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/ -name config.guess | xargs -r chmod u+w; find /home/netlove/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/ -name config.guess | xargs -r -n1 cp --remove-destination /home/netlove/lede/scripts/config.guess; find /home/netlove/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/ -name config.sub | xargs -r chmod u+w; find /home/netlove/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/ -name config.sub | xargs -r -n1 cp --remove-destination /home/netlove/lede/scripts/config.sub; AR="x86_64-openwrt-linux-musl-gcc-ar" AS="x86_64-openwrt-linux-musl-gcc -c -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/netlove/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604=fullconenat-2021-12-04-7fe02604 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=x86_64-openwrt-linux-musl-ld NM="x86_64-openwrt-linux-musl-gcc-nm" CC="x86_64-openwrt-linux-musl-gcc" GCC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" RANLIB="x86_64-openwrt-linux-musl-gcc-ranlib" STRIP=x86_64-openwrt-linux-musl-strip OBJCOPY=x86_64-openwrt-linux-musl-objcopy OBJDUMP=x86_64-openwrt-linux-musl-objdump SIZE=x86_64-openwrt-linux-musl-size CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/netlove/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604=fullconenat-2021-12-04-7fe02604 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/netlove/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604=fullconenat-2021-12-04-7fe02604 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CPPFLAGS="-I/home/netlove/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/include -I/home/netlove/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/fortify -I/home/netlove/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include " LDFLAGS="-L/home/netlove/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/lib -L/home/netlove/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/li [... cut ...]
make[3]: Leaving directory '/home/netlove/lede/package/lean/openwrt-fullconenat'
time: package/lean/openwrt-fullconenat/compile#0.74#0.34#1.14
    ERROR: package/lean/openwrt-fullconenat failed to build.
make[2]: *** [package/Makefile:116: package/lean/openwrt-fullconenat/compile] Error 1
make[2]: Leaving directory '/home/netlove/lede'
make[1]: *** [package/Makefile:110: /home/netlove/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/netlove/lede'
make: *** [/home/netlove/lede/include/toplevel.mk:230: world] Error 2