coolsnowwolf/lede #7185: 别再发编译失败的问题了

lexin8, 2021-06-25:
每次来这里查issues都有一大堆反馈编译失败的问题！
对于新手或者小白来说最好用云编译。
以我编译的经验来说，基本编译失败的原因是节点不好，资源链接不上导致。极少源码问题也会很快修复，自行加入第三方插件除外！
所以先用云编译编译默认配置，没有问题再添加编译需要的插件，最后再加入第三方插件，云编译可以同时进行，三四个小时就完成了。你们别再这里issues无人问津的！大佬们没空理你们这些低级错误的。

以下是云编译自己去拉取
https://github.com/P3TERX/Actions-OpenWrt

minlang112, 2021-06-25:
我怀疑你是高级黑，但是我没有证据

shikkoku, 2021-06-25:
我fork后用KFERMercer的云编译脚本最多的错误就是添加的插件名称或者格式没写对才会报错……比本地编译要安全和快。

yjd, 2021-06-26:
那可不一定。上次错误是因为最新的改动太多跟旧的有可能不兼容。作者还发了置顶帖，需要创建个新目录重新下载。
我编译都走美国代理。

realconnor, 2021-06-26:
确实，编译错误又不是源码bug...而且基本没办法复现要人怎么回🤦‍♂️ 如果基本的对着日志debug的能力都没有，真心建议先找别人编译好的固件，折腾的过程其实也是个学习的过程，慢慢自己也能编译定制自己需要的固件了。
另外，云编译确实省事，以前每次大规模合入官方代码自己都要折腾一阵，强迫症。现在心态特别佛系，半年升级一次的样子。最近这波更新除了一次编译超时，我的编译配置没有出过错，虽然我是懒得更新就是了…

closed 2021-06-28 by coolsnowwolf

aglent, 2023-05-22:
现在云编译 插件不多 每次都要超过6小时 失败

yysw, 2023-08-19:
> 现在云编译 插件不多 每次都要超过6小时 失败

我也遇到了同样的问题，编译的package没有任何变化，结果编译时间变长了，不知道怎么解决。

zhouzikk, 2023-10-05:
> 现在云编译 插件不多 每次都要超过6小时 失败
我也是第一次不到3显示 现在6小时都编不完