coolsnowwolf/lede #7187: mwan3 启动异常导致整机断网，提示 interface wan is error (15) and tracking is paused 错误

jshensh, 2021-06-25:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：mwan3 开机自启无法正常启动，执行 `mwan3 status` 提示错误，手动执行 `mwan3 restart` 后可恢复正常：

```
Interface status:
 interface wan is error (15) and tracking is paused
 interface wan_gwbn is error (15) and tracking is paused
```

### (2) 路由器型号和固件版本

A：斐讯 k3，OpenWrt R21.6.22
coolsnowwolf/lede 版本：a2c991e755897d7fb22d53749c5db2e4861c9e41
coolsnowwolf/packages 版本：165562e2356db29c30d7cfb01ab1fcb94eed8ba7

### (3) 详细日志

A：内核日志中无 mwan3 相关内容

```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.4.128 (jshensh@debian) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r3336-37805151b)) #0 SMP Thu Jun 24 09:37:57 2021
[    0.000000] CPU: ARMv7 Processor [413fc090] revision 0 (ARMv7), cr=10c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Phicomm K3
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Hit pending asynchronous external abort (FSR=0x00001c06) during first unmask, this is most likely caused by a firmware/bootloader bug.
[    0.000000] On node 0 totalpages: 131072
[    0.000000]   Normal zone: 288 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 32768 pages, LIFO batch:7
[    0.000000]   HighMem zone: 98304 pages, LIFO batch:31
[    0.000000] percpu: Embedded 14 pages/cpu s26700 r8192 d22452 u57344
[    0.000000] pcpu-alloc: s26700 r8192 d22452 u57344 alloc=14*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 130784
[    0.000000] Kernel command line: console=ttyS0,115200
[    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 510756K/524288K available (4973K kernel code, 149K rwdata, 1244K rodata, 1024K init, 283K bss, 13532K reserved, 0K cma-reserved, 393216K highmem)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=2, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] L2C: DT/platform modifies aux control register: 0x0a130000 -> 0x3a530000
[    0.000000] L2C-310 enabling early BRESP for Cortex-A9
[    0.000000] L2C-310 full line of zeros enabled for Cortex-A9
[    0.000000] L2C-310 ID prefetch enabled, offset 1 lines
[    0.000000] L2C-310 dynamic clock gating enabled, standby mode enabled
[    0.000000] L2C-310 cache controller enabled, 16 ways, 256 kB
[    0.000000] L2C-310: CACHE_ID 0x410000c8, AUX_CTRL 0x7e530001
[    0.000000] random: get_random_bytes called from start_kernel+0x2f0/0x49c with crng_init=0
[    0.000007] sched_clock: 64 bits at 700MHz, resolution 1ns, wraps every 4398046511103ns
[    0.000020] clocksource: arm_global_timer: mask: 0xffffffffffffffff max_cycles: 0xa17102bcf3, max_idle_ns: 440795224838 ns
[    0.000038] Switching to timer-based delay loop, resolution 1ns
[    0.000181] Calibrating delay loop (skipped), value calculated using timer frequency.. 1400.00 BogoMIPS (lpj=7000000)
[    0.000189] pid_max: default: 32768 minimum: 301
[    0.000285] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.000292] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.000840] CPU: Testing write buffer coherency: ok
[    0.000870] CPU0: Spectre v2: using BPIALL workaround
[... 490 lines cut ...]
        option failure_loss '20'
        option recovery_loss '5'
        option interval '5'
        option down '3'
        option up '8'

config interface 'wan_gwbn'
        option enabled '1'
        option initial_state 'online'
        option family 'ipv4'
        option track_method 'ping'
        option count '1'
        option size '56'
        option check_quality '0'
        option timeout '2'
        option interval '5'
        option failure_interval '5'
        option recovery_interval '5'
        option down '3'
        option reliability '2'
        option up '3'
        list track_ip '223.5.5.5'
        list track_ip '223.6.6.6'
        list track_ip '114.114.114.114'
```

hgh188, 2021-06-25:
我也是

yssyye, 2021-06-26:
一样，新版多拨不显示接口数

jshensh, 2021-06-26:
看到 https://github.com/coolsnowwolf/packages/commit/bd22627cfa07359f9824cd419c01a44594c4177b 回退了 mwan3 的版本，目前使用正常

closed 2021-06-28 by coolsnowwolf