coolsnowwolf/lede #7192: 云编译X86编译错误，请各位大神们帮忙看看什么问题，感谢！

281677160, 2021-06-26:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：
2021-06-26T05:09:59.2619152Z /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe/sfe_ipv4.c: In function 'sfe_ipv4_recv_tcp':
2021-06-26T05:09:59.2622751Z /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe/sfe_ipv4.c:2052:5: error: 'struct sk_buff' has no member named 'fast_forwarded'
2021-06-26T05:09:59.2624224Z   skb->fast_forwarded = 1;
2021-06-26T05:09:59.2624897Z      ^~
2021-06-26T05:09:59.2658871Z /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe/sfe_ipv4.c: In function 'sfe_ipv4_periodic_sync':
2021-06-26T05:09:59.2662210Z /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe/sfe_ipv4.c:2868:46: error: 'struct timer_list' has no member named 'cust_data'
2021-06-26T05:09:59.2663574Z   struct sfe_ipv4 *si = (struct sfe_ipv4 *)arg->cust_data;
2021-06-26T05:09:59.2664293Z                                               ^~
2021-06-26T05:09:59.2686699Z /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe/sfe_ipv4.c: In function 'sfe_ipv4_init':
2021-06-26T05:09:59.2690116Z /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe/sfe_ipv4.c:3552:11: error: 'struct timer_list' has no member named 'cust_data'
2021-06-26T05:09:59.2691676Z   si->timer.cust_data = (unsigned long)si;
2021-06-26T05:09:59.2692348Z            ^
2021-06-26T05:09:59.3003768Z make[4]: Leaving directory '/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.44'
2021-06-26T05:09:59.3006081Z make[5]: *** [scripts/Makefile.build:279: /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe/sfe_ipv4.o] Error 1
2021-06-26T05:09:59.3007324Z make[3]: Leaving directory '/workdir/openwrt/package/lean/shortcut-fe'
2021-06-26T05:09:59.3009430Z make[4]: *** [Makefile:1821: /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe] Error 2
2021-06-26T05:09:59.3010436Z time: package/lean/shortcut-fe/compile#1.16#0.32#1.54
2021-06-26T05:09:59.3012034Z make[3]: *** [Makefile:79: /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe/.built] Error 2
2021-06-26T05:09:59.3017477Z     ERROR: package/lean/shortcut-fe failed to build.
2021-06-26T05:09:59.3025436Z make[2]: *** [package/Makefile:116: package/lean/shortcut-fe/compile] Error 1
2021-06-26T05:09:59.3026513Z make[2]: *** Waiting for unfinished jobs....
2021-06-26T05:09:59.3042338Z   g++ -o /workdir/openwrt/build_dir/hostpkg/node-v12.22.1/out/Release/obj.host/icutools/deps/icu-small/source/common/uniset_props.o ../deps/icu-small/source/common/uniset_props.cpp '-DV8_DEPRECATION_WARNINGS' '-DV8_IMMINENT_DEPRECATION_WARNINGS' '-D__STDC_FORMAT_MACROS' '-DOPENSSL_NO_PINSHARED' '-DOPENSSL_THREADS' '-DU_COMMON_IMPLEMENTATION=1' '-DU_I18N_IMPLEMENTATION=1' '-DU_IO_IMPLEMENTATION=1' '-DU_TOOLUTIL_IMPLEMENTATION=1' '-DU_ATTRIBUTE_DEPRECATED=' '-D_CRT_SECURE_NO_DEPRECATE=' '-DU_STATIC_IMPLEMENTATION=1' '-DUCONFIG_NO_SERVICE=1' '-DU_ENABLE_DYLOAD=0' '-DU_HAVE_STD_STRING=1' '-DUCONFIG_NO_BREAK_ITERATION=0' -I../deps/icu-small/source/common -I../deps/icu-small/source/i18n -I../deps/icu-small/source/tools/toolutil  -pthread -Wall -Wextra -Wno-unused-parameter -m64 -Wno-deprecated-declarations -Wno-strict-aliasing -O3 -fno-omit-frame-pointer -fno-exceptions -std=gnu++1y -frtti -MMD -MF /workdir/openwrt/build_dir/hostpkg/node-v12.22.1/out/Release/.deps//workdir/openwrt/build_dir/hostpkg/node-v12.22.1/out/Release/obj.host/icutools/deps/icu-small/source/common/uniset_props.o.d.raw   -c
2021-06-26T05:09:59.4831387Z   g++ -o /workdir/openwrt/build_dir/hostpkg/node-v12.22.1/out/Release/obj.host/icutools/deps/icu-small/source/common/ucnvscsu.o ../deps/icu-small/source/common/ucnvscsu.cpp '-DV8_DEPRECATION_WARNINGS' '-DV8_IMMINENT_DEPRECATION_WARNINGS' '-D__STDC_FORMAT_MACROS' '-DOPENSSL_NO_PINSHARED' '-DOPENSSL_THREADS' '-DU_COMMON_IMPLEMENTATION=1' '-DU_I18N_IMPLEMENTATION=1' '-DU_IO_IMPLEMENTATION=1' '-DU_TOOLUTIL_IMPLEMENTATION=1' '-DU_ATTRIBUTE_DEPRECATED=' '-D_CRT_SECURE_NO_DEPRECATE=' '-DU_STATIC_IMPLEMENTATION=1' '-DUCONFIG_NO_SERVICE=1' '-DU_ENABLE_DYLOAD=0' '-DU_HAVE_STD_STRING=1' '-DUCONFIG_NO_BREAK_ITERATION=0' -I../deps/icu-small/source/common -I../deps/icu-small/source/i18n -I../deps/icu-small/source/tools/toolutil  -pthread -Wall -Wextra -Wno-unused-parameter -m64 -Wno-deprecated-declarations -Wno-strict-aliasing -O3 -fno-omit-frame-pointer -fno-exceptions -std=gnu++1y -frtti -MMD -MF /workdir/openwrt/build_dir/hostpkg/node-v12.22.1/out/Release/.deps//workdir/openwrt/build_dir/hostpkg/node-v12.22.1/out/Release/obj.host/icutools/deps/icu-small/source/common/ucnvscsu.o.d.raw   -c

我同时编译两个机型的固件，一个是N1的，一个是X86的，X86昨天晚上开始编译错误

xiangfeidexiaohuo, 2021-06-26:
本地跑 也是shortcut-fe这玩意编译出错

xiangfeidexiaohuo, 2021-06-26:
奇怪的是 K3这种没出错

281677160, 2021-06-26:
我编译N1的没问题，编译x86就不行

xiangfeidexiaohuo, 2021-06-26:
> 我编译N1的没问题，编译x86就不行
> […](#)

应该是 5.10内核原因

281677160, 2021-06-27:
> > 我编译N1的没问题，编译x86就不行
> > […](#)
> 
> 应该是 5.10内核原因

是的，我把内核版本改成5.4的就编译成功了，等大雕修复吧

suizhenmin, 2021-06-27:
我本地编译没有问题，是不是添加了其他的源导致冲突

281677160, 2021-06-27:
> 我本地编译没有问题，是不是添加了其他的源导致冲突

没搞明白，前几天还没问题，就前天晚上吧，编译就开始出现这个错误了，用5.4内核又可以成功编译

xiangfeidexiaohuo, 2021-06-27:
> > 我本地编译没有问题，是不是添加了其他的源导致冲突
> 
> 没搞明白，前几天还没问题，就前天晚上吧，编译就开始出现这个错误了，用5.4内核又可以成功编译

只能等大雕修修5.10内核了，应该是新的内核不支持sfe ，或者不兼容

281677160, 2021-06-27:
> > > 我本地编译没有问题，是不是添加了其他的源导致冲突
> > 
> > 
> > 没搞明白，前几天还没问题，就前天晚上吧，编译就开始出现这个错误了，用5.4内核又可以成功编译
> 
> 只能等大雕修修5.10内核了，应该是新的内核不支持sfe ，或者不兼容

我都没选sfe这个插件啊

shikkoku, 2021-06-28:
#7204 看看修好了没？

281677160, 2021-06-28:
> #7204 看看修好了没？

不知道啊，昨天晚上睡觉前编译不成功，我现在再去编译试试

281677160, 2021-06-28:
> #7204 看看修好了没？

感谢您提醒，我编译成功了，谢谢！！！，也感谢大雕辛勤付出，感谢感谢感谢！！！

281677160, 2021-06-28:
出了点状况了，x86能编译成功了，现在到N1编译失败了，昨天晚上我睡觉之前(3点左右)，编译了一次，X86还是失败，N1还是成功的，现在反过来了，X86成功了，N1的失败了

2021-06-28T05:59:43.5225329Z SHELL= flock /workdir/openwrt/tmp/.vlmcsd-svn1113.tar.gz.flock -c '  	/workdir/openwrt/scripts/download.pl "/workdir/openwrt/dl" "vlmcsd-svn1113.tar.gz" "9510cf4e48e809b09fea60f29e41d3a0fb0f58257644dc60929ad503e8a156af" "" "https://codeload.github.com/Wind4/vlmcsd/tar.gz/svn1113?"    '
2021-06-28T05:59:43.5506401Z + curl -f --connect-timeout 20 --retry 5 --location --insecure https://codeload.github.com/Wind4/vlmcsd/tar.gz/svn1113?/vlmcsd-svn1113.tar.gz
2021-06-28T05:59:43.5573657Z   % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
2021-06-28T05:59:43.5574382Z                                  Dload  Upload   Total   Spent    Left  Speed
2021-06-28T05:59:43.5574736Z 
2021-06-28T05:59:43.8832550Z   0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
2021-06-28T05:59:43.8835482Z 100  232k    0  232k    0     0   713k      0 --:--:-- --:--:-- --:--:--  713k
2021-06-28T05:59:43.8894939Z Hash of the downloaded file does not match (file: 62f55c48f5de1249c2348ab6b96dabbe7e38899230954b0c8774efb01d9c42cc, requested: 9510cf4e48e809b09fea60f29e41d3a0fb0f58257644dc60929ad503e8a156af) - deleting download.
2021-06-28T05:59:43.8972709Z + curl -f --connect-timeout 20 --retry 5 --location --insecure https://sources.cdn.openwrt.org/vlmcsd-svn1113.tar.gz
2021-06-28T05:59:43.9038900Z   % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
2021-06-28T05:59:43.9044533Z                                  Dload  Upload   Total   Spent    Left  Speed
2021-06-28T05:59:43.9044929Z 
2021-06-28T05:59:44.0672384Z   0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
2021-06-28T05:59:44.0676276Z   0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
2021-06-28T05:59:44.0678629Z curl: (22) The requested URL returned error: 404 
2021-06-28T05:59:44.0698997Z Download failed.
2021-06-28T05:59:44.0774243Z + curl -f --connect-timeout 20 --retry 5 --location --insecure https://sources.openwrt.org/vlmcsd-svn1113.tar.gz
2021-06-28T05:59:44.0835083Z   % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
2021-06-28T05:59:44.0836003Z                                  Dload  Upload   Total   Spent    Left  Speed
2021-06-28T05:59:44.0836465Z 
2021-06-28T05:59:44.4937243Z   0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
2021-06-28T05:59:44.8329252Z   0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
2021-06-28T05:59:44.8330092Z   0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
2021-06-28T05:59:44.8336363Z curl: (22) The requested URL returned error: 404 Not Found
2021-06-28T05:59:44.8351338Z Download failed.
2021-06-28T05:59:44.8423833Z + curl -f --connect-timeout 20 --retry 5 --location --insecure https://mirror2.openwrt.org/sources/vlmcsd-svn1113.tar.gz
2021-06-28T05:59:44.8491799Z   % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
2021-06-28T05:59:44.8494703Z                                  Dload  Upload   Total   Spent    Left  Speed
2021-06-28T05:59:44.8495066Z 
2021-06-28T05:59:45.5157254Z   0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
2021-06-28T05:59:45.5158098Z   0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
2021-06-28T05:59:45.5163895Z curl: (22) The requested URL returned error: 404 Not Found
2021-06-28T05:59:45.5179955Z Download failed.
2021-06-28T05:59:45.5180891Z No more mirrors to try - giving up.
2021-06-28T05:59:45.5194475Z make[3]: *** [Makefile:49: /workdir/openwrt/dl/vlmcsd-svn1113.tar.gz] Error 2
2021-06-28T05:59:45.5195626Z make[3]: Leaving directory '/workdir/openwrt/package/lean/vlmcsd'
2021-06-28T05:59:45.5196714Z time: package/lean/vlmcsd/compile#0.25#0.08#2.18
2021-06-28T05:59:45.5202173Z     ERROR: package/lean/vlmcsd failed to build.
2021-06-28T05:59:45.5206339Z make[2]: *** [package/Makefile:116: package/lean/vlmcsd/compile] Error 1
2021-06-28T05:59:45.5213472Z make[2]: Leaving directory '/workdir/openwrt'
2021-06-28T05:59:45.5229915Z make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
2021-06-28T05:59:45.5231210Z make[1]: Leaving directory '/workdir/openwrt'
2021-06-28T05:59:45.5232167Z make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2
2021-06-28T05:59:45.5278975Z ##[error]Process completed with exit code 2.
2021-06-28T05:59:45.5569570Z Post job cleanup.
2021-06-28T05:59:45.7760620Z [command]/usr/bin/git version
2021-06-28T05:59:45.7798137Z git version 2.32.0
2021-06-28T05:59:45.7843678Z [command]/usr/bin/git config --local --name-only --get-regexp core\.sshCommand
2021-06-28T05:59:45.7879837Z [command]/usr/bin/git submodule foreach --recursive git config --local --name-only --get-regexp 'core\.sshCommand' && git config --local --unset-all 'core.sshCommand' || :
2021-06-28T05:59:45.8189488Z [command]/usr/bin/git config --local --name-only --get-regexp http\.https\:\/\/github\.com\/\.extraheader
2021-06-28T05:59:45.8190432Z http.https://github.com/.extraheader
2021-06-28T05:59:45.8191534Z [command]/usr/bin/git config --local --unset-all http.https://github.com/.extraheader
2021-06-28T05:59:45.8193544Z [command]/usr/bin/git submodule foreach --recursive git config --local --name-only --get-regexp 'http\.https\:\/\/github\.com\/\.extraheader' && git config --local --unset-all 'http.https://github.com/.extraheader' || :
2021-06-28T05:59:45.8471751Z Cleaning up orphan processes

shikkoku, 2021-06-28:
2021-06-28T05:59:44.8423833Z + curl -f --connect-timeout 20 --retry 5 --location --insecure https://mirror2.openwrt.org/sources/vlmcsd-svn1113.tar.gz

源挂了

281677160, 2021-06-28:
> 2021-06-28T05:59:44.8423833Z + curl -f --connect-timeout 20 --retry 5 --location --insecure https://mirror2.openwrt.org/sources/vlmcsd-svn1113.tar.gz
> 
> 源挂了

package/lean/vlmcsd 这个文件有问题，用L大今天修改之前的内容替换就可以编译成功了，等大雕修复了

EnnawYang, 2021-06-28:
> package/lean/vlmcsd 这个文件有问题，用L大今天修改之前的内容替换就可以编译成功了，等大雕修复了
> 
测试并无问题，有缓存的先清除dl目录下原有的包
```
~/openwrt/lede   master ±  make package/lean/vlmcsd/download V=s -j1

time: target/linux/prereq#0.12#0.02#0.15
make[1]: Entering directory '/home/xw/openwrt/lede'
make[2]: Entering directory '/home/xw/openwrt/lede/package/lean/vlmcsd'
mkdir -p /home/xw/openwrt/lede/dl
/home/xw/openwrt/lede/scripts/download.pl "/home/xw/openwrt/lede/dl" "vlmcsd-svn1113.tar.gz" "62f55c48f5de1249c2348ab6b96dabbe7e38899230954b0c8774efb01d9c42cc" "" "https://codeload.github.com/Wind4/vlmcsd/tar.gz/svn1113?"    
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://codeload.github.com/Wind4/vlmcsd/tar.gz/svn1113?/vlmcsd-svn1113.tar.gz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  232k  100  232k    0     0   176k      0  0:00:01  0:00:01 --:--:--  176k
make[2]: Leaving directory '/home/xw/openwrt/lede/package/lean/vlmcsd'
time: package/lean/vlmcsd/download#0.18#0.04#1.52
make[1]: Leaving directory '/home/xw/openwrt/lede'
```

281677160, 2021-06-28:
> 测试并无问题，有缓存的先清除dl目录下原有的包
>
> ```
> ~/openwrt/lede > master ± > make package/lean/vlmcsd/download V=s -j1
>
> time: target/linux/prereq#0.12#0.02#0.15
> make[1]: Entering directory '/home/xw/openwrt/lede' [...]

呃，那我就不知道什么问题了，云编译不存在缓存的吧，反正我用大雕今天修改的文件就编译不成功，用回老的文件内容替换就没问题

281677160, 2021-06-30:
感谢大雕修复，前两天没空测试，昨晚上编译x86跟N1的都成功了，感谢！！！

closed 2021-06-30 by 281677160