coolsnowwolf/lede #7193: 新源码编译出来的竞斗云固件没有温度显示。

HUANGJHON, 2021-06-26:
[image]

Boos4721, 2021-06-26:
[image]
?

AmadeusGhost, 2021-06-26:
你把 WiFi 关掉就没有温度显示了。

icsfy, 2021-06-26:
> 你把 WiFi 关掉就没有温度显示了。

我这里acrh17同样没有温度显示，wifi是开了的，看日志报错如下：
`
Sat Jun 26 18:50:40 2021 daemon.err uhttpd[3000]: awk: /sys/class/ieee80211/phy*/device/hwmon/hwmon*/temp1_input: No such file or directory
`

Boos4721, 2021-06-26:
> > 你把 WiFi 关掉就没有温度显示了。
> 
> 我这里acrh17同样没有温度显示，wifi是开了的，看日志报错如下：
> `Sat Jun 26 18:50:40 2021 daemon.err uhttpd[3000]: awk: /sys/class/ieee80211/phy*/device/hwmon/hwmon*/temp1_input: No such file or directory`

[image]

icsfy, 2021-06-26:
> > > 你把 WiFi 关掉就没有温度显示了。
> > 
> > 
> > 我这里acrh17同样没有温度显示，wifi是开了的，看日志报错如下：
> > `Sat Jun 26 18:50:40 2021 daemon.err uhttpd[3000]: awk: /sys/class/ieee80211/phy*/device/hwmon/hwmon*/temp1_input: No such file or directory`
> 
> [image]

看了下，我这里没有hwmon这个目录：

```
root@OpenWrt:~# ls -A /sys/class/ieee80211/phy*/device
/sys/class/ieee80211/phy0/device:
aer_dev_correctable       device                    local_cpus                reset
aer_dev_fatal             devspec                   max_link_speed            resource
aer_dev_nonfatal          dma_mask_bits             max_link_width            resource0
ari_enabled               driver                    modalias                  revision
broken_parity_status      driver_override           msi_bus                   subsystem
class                     enable                    msi_irqs                  subsystem_device
config                    ieee80211                 net                       subsystem_vendor
consistent_dma_mask_bits  irq                       of_node                   uevent
current_link_speed        leds                      remove                    vendor
current_link_width        local_cpulist             rescan

/sys/class/ieee80211/phy1/device:
driver           driver_override  ieee80211        modalias         of_node          subsystem        uevent
root@OpenWrt:~#
```

shikkoku, 2021-06-26:
不知默认有无选了autocore-arm

minlang112, 2021-06-26:
我记得不是某些型号默认把温度监控去掉了么……提升首页显示的流畅性

Boos4721, 2021-06-26:
> 我记得不是某些型号默认把温度监控去掉了么……提升首页显示的流畅性

https://github.com/coolsnowwolf/lede/blob/master/package/lean/autocore/files/arm/sbin/cpuinfo

AmadeusGhost, 2021-06-26:
> 我记得不是某些型号默认把温度监控去掉了么……提升首页显示的流畅性

那个是换一种方式读取温度了，不是去掉了。

closed 2021-06-27 by coolsnowwolf