coolsnowwolf/lede #7195: 寻找适合Lean openwrt 当前版本的luci-app-fileassistant文件助手

gztingting, 2021-06-26:
之前版本还可以用，不知道怎么适配。

coolsnowwolf, 2021-06-26:
具体是什么问题

gztingting, 2021-06-27:
进入文件助手界面后无法正确显示文件列表和其他操作等。
见下面截图
[image]

281677160, 2021-06-27:
> 进入文件助手界面后无法正确显示文件列表和其他操作等。
> 见下面截图
> [image]

文件助手是用来干嘛的？上传文件到路由器吗？修改文件内容？如果就这的话，干嘛不直接用luci-app-filebrowser

gztingting, 2021-06-27:
> > 进入文件助手界面后无法正确显示文件列表和其他操作等。
> > 见下面截图
> > [image]
> 
> 文件助手是用来干嘛的？上传文件到路由器吗？修改文件内容？如果就这的话，干嘛不直接用luci-app-filebrowser

和luci-app-filebrowser功能类似，没有后者那么复杂，fileassistant可以直接查看文件、下载、上传、安装等，界面简单，filebrowser类似于云盘，在外网房屋时还要多映射一个端口。

我猜应该是路径问题，移植自lienol的luci-package，两者的路径，我看原仓库的luasrc/view/下的fileassistant.htm文件两行
<link rel="stylesheet" href=”/luci-static/resources/fileassistant/fb.css?v=@ver">

实际上fb.css和fb.js文件是在/www/luci-static/resources/fileassistant路径下。
我改了这个路径，依然没有正确显示。

closed 2021-06-27 by gztingting