coolsnowwolf/lede #7199: 今天编译的固件“SoftEther VPN 服务器”连不上

mitsukileung, 2021-06-27:
X86固件
之前编译出来的固件没有问题，设置都是和之前的设置一样。
防火墙 500 和 4500的端口默认开启

Beginner-Go, 2021-06-27:
https://github.com/coolsnowwolf/packages/commit/724201e3fa5166904411bed26521289a465cfd48

mitsukileung, 2021-06-27:
> 
> 
> That's what caused it, right? [coolsnowwolf/packages@724201e](https://github.com/coolsnowwolf/packages/commit/724201e3fa5166904411bed26521289a465cfd48)

原来更新了版本，等大佬修复

Beginner-Go, 2021-06-27:
can you test ? 
https://github.com/coolsnowwolf/lede/pull/7201

mitsukileung, 2021-06-27:
> 
> 
> can you test ?
> #7201

好的，我试试

mitsukileung, 2021-06-27:
> 
> 
> can you test ?
> #7201

好像也不行，我把 /etc/init.d 里面的 softethervpn 文件按你的PR改了也不行，固件还没有编译完
SSH进去看了下目录，的确是/usr/libexec/softethervpn，换了你的PR启动不成功
我的问题是 softethervpn 能启动，设置好SERVER后，手机VPN设置了还是不能连上
前天编译的固件是可以的，就是昨天更新后再编译出来就不行了

Beginner-Go, 2021-06-27:
改了后需要重启这个进程

mitsukileung, 2021-06-27:
> 
> 
> can you test ?
> #7201

合并PR编译的固件还是不行，这次是直接启动不了

dlj390, 2021-06-28:
同样的问题，之前是可以的。

dlj390, 2021-06-28:
出现问题的固件发生在昨天，之前都好。

closed 2021-06-28 by coolsnowwolf