coolsnowwolf/lede #7203: 新版本固件 不保留配置升级后的系统日志错误

XinSSS, 2021-06-27:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
新编译的固件， 无配置保留升级后， 启动时能看到系统日志报错

Q：是否搜索了issue (使用 "x" 选择)
* [x] 有类似的issue
已搜索issue， 关闭了automount解决了文件系统的报错问题，但很明显，这个不是解决问题的方法。
extra_command 报错也查看了base-file, 是最新的没错

## 2. 详细叙述

### (1) 具体问题
新编译的固件， 无配置保留升级后， 启动时能看到系统日志报错

### (2) 路由器型号和固件版本

A：x86  uefi版本

### (3) 详细日志

kern.err kernel: [   39.805382] squashfs: Unknown parameter 'discard'
kern.info kernel: [   39.811285] F2FS-fs (sda128): Magic Mismatch, valid(0xf2f52010) - read(0xd44d89e4)
kern.err kernel: [   39.818986] F2FS-fs (sda128): Can't find valid F2FS filesystem in 1th superblock
kern.info kernel: [   39.826884] F2FS-fs (sda128): Magic Mismatch, valid(0xf2f52010) - read(0x80a345bd)
kern.err kernel: [   39.834820] F2FS-fs (sda128): Can't find valid F2FS filesystem in 2th superblock
kern.err kernel: [   39.843685] exFAT-fs (sda128): invalid boot record signature
kern.err kernel: [   39.849459] exFAT-fs (sda128): failed to read boot sector
kern.err kernel: [   39.854998] exFAT-fs (sda128): failed to recognize exfat type

以及

daemon.err uhttpd[3871]: sh: /etc/init.d/dropbear: line 13: extra_command: not found
daemon.err uhttpd[3871]: sh: /etc/init.d/haproxy: line 21: extra_command: not found

shikkoku, 2021-06-28:
同样复现，用Actions编译的应该不存在旧缓存了吧。

#7133

shikkoku, 2021-06-30:
复现错误
daemon.err uhttpd[3871]: sh: /etc/init.d/dropbear: line 13: extra_command: not found
daemon.err uhttpd[3871]: sh: /etc/init.d/haproxy: line 21: extra_command: not found

方法，进去启动项里重启计划任务（crond）的进程。日志会记录这一段错误

Wed Jun 30 13:58:00 2021 cron.err crond[9267]: crond (busybox 1.33.1) started, log level 5
Wed Jun 30 13:58:00 2021 daemon.err uhttpd[2681]: sh: write error: Broken pipe
Wed Jun 30 13:58:30 2021 daemon.err uhttpd[2681]: sh: /etc/init.d/dropbear: line 13: extra_command: not found
Wed Jun 30 13:58:38 2021 daemon.err uhttpd[2681]: sh: /etc/init.d/dropbear: line 13: extra_command: not found
Wed Jun 30 13:58:38 2021 cron.err crond[10661]: crond (busybox 1.33.1) started, log level 5
Wed Jun 30 14:14:29 2021 daemon.err uhttpd[2681]: sh: /etc/init.d/dropbear: line 13: extra_command: not found
Wed Jun 30 14:14:34 2021 daemon.err uhttpd[2681]: sh: /etc/init.d/dropbear: line 13: extra_command: not found
Wed Jun 30 14:14:35 2021 cron.err crond[14695]: crond (busybox 1.33.1) started, log level 5

closed 2021-06-30 by coolsnowwolf

caliban511, 2021-12-24:
同样问题，，，

zjwconan, 2022-11-20:
做了对比验证
lede的R22.10.1 Release : x64 returns
openwrt-x64-R22.10.1-squashfs-combined-efi.img.gz

_[   39.988572] squashfs: Unknown parameter 'discard'
[   39.994314] F2FS-fs (sda128): Magic Mismatch, valid(0xf2f52010) - read(0x89e45589)
[   40.002366] F2FS-fs (sda128): Can't find valid F2FS filesystem in 1th superblock
[   40.010322] F2FS-fs (sda128): Magic Mismatch, valid(0xf2f52010) - read(0x2e48937b)
[   40.018377] F2FS-fs (sda128): Can't find valid F2FS filesystem in 2th superblock
[   40.027602] exFAT-fs (sda128): invalid boot record signature
[   40.033602] exFAT-fs (sda128): failed to read boot sector
[   40.041430] exFAT-fs (sda128): failed to recognize exfat type_

openwrt原版22.03.2无该问题
[image]