coolsnowwolf/lede #7209: ddns只有启动的第一次可以成功。其他时候都不行了。

13810169, 2021-06-28:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
ddns只有创建的第一次成功，后续一直不行。
Q：是否搜索了issue (使用 "x" 选择)
* [X] 没有类似的issue
* 
### (2) 路由器型号和固件版本
x86_64 5.10.43内核版本
A：

### (3) 详细日志
 223848  info : Update successful - IP '2409:xxxx:xxxx:xxxx:xxxx:xxxx:xxxx:xxxx' send
 223849  info : Forced update successful - IP: '2409:xxxx:xxxx:xxxx:xxxx:xxxx:xxxx:xxxx' send
 223849       : Waiting 600 seconds (Check Interval)
 224849       : Detect registered/public IP
 224849       : #> /usr/bin/nslookup 马赛克.xyz  >/var/run/ddns/OpenWrt_WEB.dat 2>/var/run/ddns/OpenWrt_WEB.err
 224849  WARN : NO valid IP found
 224849  WARN : Get registered/public IP for '马赛克.xyz' failed - retry 1/0 in 60 seconds
 224949       : #> /usr/bin/nslookup 马赛克.xyz  >/var/run/ddns/OpenWrt_WEB.dat 2>/var/run/ddns/OpenWrt_WEB.err
 224949  WARN : NO valid IP found
 224949  WARN : Get registered/public IP for '马赛克.xyz' failed - retry 2/0 in 60 seconds
 225049       : #> /usr/bin/nslookup 马赛克.xyz  >/var/run/ddns/OpenWrt_WEB.dat 2>/var/run/ddns/OpenWrt_WEB.err

只有第一次成功，后续一直是这样的，已经跑到50/0了。都是重复的内容

yjd, 2021-06-30:
新版不清楚，4月份以前的我这边用得好好的。日志有时候会有很多超时错误等。但不影响使用。

lonelysoul, 2021-07-01:
阿里云的ddns怎么申请呀，不成功。现在用dynu的……

missorforget, 2022-02-11:
DNS-服务器 填入一个ipv6 dns服务器

closed 2024-11-19 by coolsnowwolf