coolsnowwolf/lede #7210: 编译docker需要选哪些？

wxs-dev, 2021-06-28:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：选了luci-app-docker之后编译失败了，还需要选哪些吗？第一次编译doker

### (2) 路由器型号和固件版本

A：N1

gztingting, 2021-06-28:
默认已经把依赖勾选了吧，或者在utilities里选择docker，dockerd，dockerd里面加上一些内核、网络、存储选型。

birdmanwings, 2021-06-29:
请问下编译的时候勾选了luci-app-docker，使用/etc/docker-web的时候还是显示docker: not found，是还需要勾选配置什么呢

> 默认已经把依赖勾选了吧，或者在utilities里选择docker，dockerd，dockerd里面加上一些内核、网络、存储选型。

Dimitri2020007, 2021-06-29:
docker需要luci-app-docker、luci-lib-docker、utilities里选择docker，dockerd，dockerd里面加上一些内核、网络、存储选型。

wxs-dev, 2021-06-29:
> docker需要luci-app-docker、luci-lib-docker、utilities里选择docker，dockerd，dockerd里面加上一些内核、网络、存储选型。

[image]
我只有这几个，没见到内核、网络、存储选型，这些要在哪选？

acooler15, 2021-07-01:
> > docker需要luci-app-docker、luci-lib-docker、utilities里选择docker，dockerd，dockerd里面加上一些内核、网络、存储选型。
> 
> [image]
> 我只有这几个，没见到内核、网络、存储选型，这些要在哪选？

同问

wxs-dev, 2021-07-03:
已解决，删掉了lede重新拉项目，选择luci-app-docker即可

closed 2021-07-03 by wxs-dev